openwrt/packages #13728: 19.07: net-snmp: snmpd not listening after reboot

karlp, 2020-10-20:
Maintainer: @stintel 
Environment: ath79, openwrt-19.07 branch (works fine in master!)

Description: snmpd responds perfectly well on first boot, but after reboot, it is non-responsive until it is restarted.  snmp requests time out, no logging.  Process is running, conf file is generated in /var/run/snmpd.conf as normal

Only hint is that the agentXSocket file is _NOT_ created!  (This is how I first came to this, my subagent wasn't working)

```
root@OpenWrt:/# pgrep -lfa snmpd
1248 /usr/sbin/snmpd -Lf /dev/null -f
root@OpenWrt:/# cat /var/run/snmpd.conf 
agentaddress UDP:161,UDP6:161
master agentx
agentXSocket /var/run/agentx.sock
sysLocation office
sysContact bofh@example.com
sysName HeartOfGold
com2sec ro default public
com2sec rw localhost private
com2sec6 ro default public
com2sec6 rw localhost private
group public v1 ro
group public v2c ro
group public usm ro
group private v1 rw
group private v2c rw
group private usm rw
view all included .1 
access public "" any noauth exact all none none
access private "" any noauth exact all all all
exec  filedescriptors /bin/cat /proc/sys/fs/file-nr
engineIDType 3
engineIDNic eth0
root@OpenWrt:/# ls -l /var/run/ag*
ls: /var/run/agent*: No such file or directory
```

After restart, works just fine, and agentx file is created and snmp requests are answered
```
root@OpenWrt:/# /etc/init.d/snmpd restart
root@OpenWrt:/# ls -l /var/run/ag*
srwxr-xr-x    1 root     root             0 Oct 20 10:27 /var/run/agentx.sock
root@OpenWrt:/# 
```

Openwrt build info.  (Tried various points to very early on 19.07 branch, but that was still with my own extra apps, this is a perfectly clean one)
```
root@OpenWrt:/# cat /etc/openwrt_release 
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='19.07-SNAPSHOT'
DISTRIB_REVISION='r11221-7dd822983b'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 19.07-SNAPSHOT r11221-7dd822983b'
DISTRIB_TAINTS='no-all'
root@OpenWrt:/# 
```

Diffconfig on clean 19.07 branch
```
$ ./scripts/diffconfig.sh 
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_etactica_eg200=y
CONFIG_DEVEL=y
CONFIG_DOWNLOAD_FOLDER="/home/karlp/src/dl"
CONFIG_PACKAGE_libnetsnmp=y
CONFIG_PACKAGE_libpci=y
CONFIG_PACKAGE_snmpd=y
```

karlp, 2020-10-22:
"Someone" didn't entirely trust my builds, so I retested with https://downloads.openwrt.org/releases/19.07.4/targets/ath79/generic/openwrt-19.07.4-ath79-generic-etactica_eg200-squashfs-sysupgrade.bin which matches my hardware, and then
```
opkg update
opkg install snmpd
```

and it still fails to respond after reboot, with the same symptoms.  process is running, config file references an agentx socket that hasn't been created.

karlp, 2020-10-22:
strace output of the when stuck.  (lengthy!)
it is happily "running" and reading system metrics, and it does get a line when I make a snmp request, but it doesn't answer.
Annotates prefixed with >>>>

```
_newselect(8, [4 6 7], [], [], {tv_sec=1, tv_usec=971968}

>>>>>periodic sleep here,

) = 0 (Timeout)
_newselect(8, [4 6 7], [], [], {tv_sec=0, tv_usec=19057}) = 0 (Timeout)
open("/proc/stat", O_RDONLY|O_LARGEFILE) = 8
read(8, "cpu  4545 0 3713 76699 0 0 32 0 "..., 4095) = 317
close(8)                                = 0
open("/proc/vmstat", O_RDONLY|O_LARGEFILE) = 8
read(8, "nr_free_pages 8051\nnr_zone_inact"..., 4095) = 842
close(8)                                = 0
_newselect(8, [4 6 7], [], [], {tv_sec=0, tv_usec=110056}) = 0 (Timeout)
open("/proc/net/dev", O_RDONLY|O_LARGEFILE) = 8
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 9
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 10
ioctl(10, SIOCGIFCONF, {ifc_len=8 * sizeof(struct ifreq) => 2 * sizeof(struct ifreq), ifc_buf=[{ifr_name="lo", ifr_addr={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("127.0.0.1")}}, {ifr_name="br-lan", ifr_addr={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.88.192")}}]}) = 0
ioctl(10, SIOCGIFCONF, {ifc_len=16 * sizeof(struct ifreq) => 2 * sizeof(struct ifreq), ifc_buf=[{ifr_name="lo", ifr_addr={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("127.0.0.1")}}, {ifr_name="br-lan", ifr_addr={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.88.192")}}]}) = 0
ioctl(10, SIOCGIFINDEX, {ifr_name="lo", }) = 0
socket(AF_NETLINK, SOCK_DGRAM, NETLINK_ROUTE) = 11
sendto(11, {{len=24, type=0x16 /* NLMSG_??? */, flags=NLM_F_REQUEST|0x100, seq=0, pid=0}, "\x02\x00\x00\x00\x00\x00\x00\x00"}, 24, 0, NULL, 0) = 24
recvfrom(11, [{{len=76, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=0, pid=3509681667}, "\x02\x08\x80\xfe\x00\x00\x00\x01\x00\x08\x00\x01\x7f\x00\x00\x01\x00\x08\x00\x02\x7f\x00\x00\x01\x00\x07\x00\x03\x6c\x6f\x00\x00"...}, {{len=88, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=0, pid=3509681667}, "\x02\x18\x80\x00\x00\x00\x00\x04\x00\x08\x00\x01\xc0\xa8\x58\xc0\x00\x08\x00\x02\xc0\xa8\x58\xc0\x00\x08\x00\x04\xc0\xa8\x58\xff"...}], 16384, 0, NULL, NULL) = 164
close(11)                               = 0
ioctl(10, SIOCGIFNETMASK, {ifr_name="lo", ifr_netmask={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("255.0.0.0")}}) = 0
ioctl(10, SIOCGIFFLAGS, {ifr_name="lo", ifr_flags=IFF_UP|IFF_LOOPBACK|IFF_RUNNING}) = 0
ioctl(10, SIOCGIFINDEX, {ifr_name="br-lan", }) = 0
socket(AF_NETLINK, SOCK_DGRAM, NETLINK_ROUTE) = 11
sendto(11, {{len=24, type=0x16 /* NLMSG_??? */, flags=NLM_F_REQUEST|0x100, seq=0, pid=0}, "\x02\x00\x00\x00\x00\x00\x00\x00"}, 24, 0, NULL, 0) = 24
recvfrom(11, [{{len=76, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=0, pid=3345413570}, "\x02\x08\x80\xfe\x00\x00\x00\x01\x00\x08\x00\x01\x7f\x00\x00\x01\x00\x08\x00\x02\x7f\x00\x00\x01\x00\x07\x00\x03\x6c\x6f\x00\x00"...}, {{len=88, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=0, pid=3345413570}, "\x02\x18\x80\x00\x00\x00\x00\x04\x00\x08\x00\x01\xc0\xa8\x58\xc0\x00\x08\x00\x02\xc0\xa8\x58\xc0\x00\x08\x00\x04\xc0\xa8\x58\xff"...}], 16384, 0, NULL, NULL) = 164
close(11)                               = 0
ioctl(10, SIOCGIFNETMASK, {ifr_name="br-lan", ifr_netmask={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("255.255.255.0")}}) = 0
ioctl(10, SIOCGIFFLAGS, {ifr_name="br-lan", ifr_flags=IFF_UP|IFF_BROADCAST|IFF_RUNNING|IFF_MULTICAST}) = 0
close(10)                               = 0
open("/proc/net/if_inet6", O_RDONLY|O_LARGEFILE) = 10
read(10, "fe80000000000000c69300fffe0366bd"..., 1024) = 108
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 11
ioctl(11, SIOCGIFINDEX, {ifr_name="br-lan", }) = 0
close(11)                               = 0
socket(AF_NETLINK, SOCK_DGRAM, NETLINK_ROUTE) = 11
sendto(11, {{len=24, type=0x16 /* NLMSG_??? */, flags=NLM_F_REQUEST|0x100, seq=0, pid=0}, "\x0a\x00\x00\x00\x00\x00\x00\x00"}, 24, 0, NULL, 0) = 24
recvfrom(11, [{{len=72, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=0, pid=2685241717}, "\x0a\x80\x80\xfe\x00\x00\x00\x01\x00\x14\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x14\x00\x06"...}, {{len=72, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=0, pid=2685241717}, "\x0a\x40\x80\xfd\x00\x00\x00\x04\x00\x14\x00\x01\xfe\x80\x00\x00\x00\x00\x00\x00\xc6\x93\x00\xff\xfe\x03\x66\xbd\x00\x14\x00\x06"...}], 16384, 0, NULL, NULL) = 144
close(11)                               = 0
socket(AF_NETLINK, SOCK_DGRAM, NETLINK_ROUTE) = 11
sendto(11, {{len=24, type=0x16 /* NLMSG_??? */, flags=NLM_F_REQUEST|0x100, seq=0, pid=0}, "\x0a\x00\x00\x00\x00\x00\x00\x00"}, 24, 0, NULL, 0) = 24
recvfrom(11, [{{len=72, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=0, pid=3368625778}, "\x0a\x80\x80\xfe\x00\x00\x00\x01\x00\x14\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x14\x00\x06"...}, {{len=72, type=0x14 /* NLMSG_??? */, flags=NLM_F_MULTI, seq=0, pid=3368625778}, "\x0a\x40\x80\xfd\x00\x00\x00\x04\x00\x14\x00\x01\xfe\x80\x00\x00\x00\x00\x00\x00\xc6\x93\x00\xff\xfe\x03\x66\xbd\x00\x14\x00\x06"...}], 16384, 0, NULL, NULL) = 144
close(11)                               = 0
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 11
ioctl(11, SIOCGIFINDEX, {ifr_name="lo", }) = 0
close(11)                               = 0
socket(AF_NETLINK, SOCK_DGRAM, NETLINK_ROUTE) = 11
[... cut ...]
ioctl(9, SIOCGIFCONF, {ifc_len=16 * sizeof(struct ifreq) => 2 * sizeof(struct ifreq), ifc_buf=[{ifr_name="lo", ifr_addr={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("127.0.0.1")}}, {ifr_name="br-lan", ifr_addr={sa_family=AF_INET, sin_port=htons(0), sin_addr=inet_addr("192.168.88.192")}}]}) = 0
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 10
ioctl(10, SIOCGIFINDEX, {ifr_name="eth0", }) = 0
close(10)                               = 0
open("/sys/class/net/eth0/devic^C|O_LARGEFILEstrace: Process 1427 detached
 <detached ...>
```

karlp, 2020-10-22:
And tcpdump requested, matches expectations, udp packets arrive, udp packets never leave.  this is when it's non-responsive
```
root@OpenWrt:/# tcpdump port 161 and udp
[ 1347.387495] device br-lan entered promiscuous mode
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on br-lan, link-type EN10MB (Ethernet), capture size 262144 bytes
10:50:07.667406 IP 192.168.88.210.57295 > 192.168.88.192.161:  GetNextRequest(25)  
10:50:08.668428 IP 192.168.88.210.57295 > 192.168.88.192.161:  GetNextRequest(25)  
10:50:09.669425 IP 192.168.88.210.57295 > 192.168.88.192.161:  GetNextRequest(25)  
10:50:10.670415 IP 192.168.88.210.57295 > 192.168.88.192.161:  GetNextRequest(25)  
10:50:11.671391 IP 192.168.88.210.57295 > 192.168.88.192.161:  GetNextRequest(25)  
10:50:12.672417 IP 192.168.88.210.57295 > 192.168.88.192.161:  GetNextRequest(25)  
```

arrmo, 2021-02-22:
FYI, having the same issue - fix is pretty easy, move snmpd down in the boot order. And in reality, it's not critical, agreed?

Similar thread here,
https://forum.openwrt.org/t/snmpd-not-starting-on-boot/87882/23

Submit a PR, to change from S50 to S90? This seems to work for me, avoid the race condition (as it seems to be).

Thanks!

stintel, 2021-03-17:
Looks like I might be hitting this on master now, on realtek (Netgear GS108Tv3). Can't reliably reproduce it though.

arrmo, 2021-03-17:
Yes, it seems to happen to me ... most of the time. Timing is on the edge, so sometimes OK, sometimes not. At S90 - very reliably starts up.

Thanks!

stintel, 2021-03-17:
So after writing that I was unable to reproduce it again. I think the previous two times were after sysupgrade. I vaguely recall running into it on some of my APs a while ago. This leads me to believe it is easier to trigger on slow, single-core devices.

Ideally we report this upstream, but since it's difficult to reproduce ... I'll do some more testing locally, if I don't find anything in reasonable time will go with START=90 - thanks for the suggestion!

stintel, 2021-03-18:
Would you mind trying this:

```diff
diff --git a/net/net-snmp/files/snmpd.init b/net/net-snmp/files/snmpd.init
index fbae9c286..4d6c5533e 100644
--- a/net/net-snmp/files/snmpd.init
+++ b/net/net-snmp/files/snmpd.init
@@ -347,7 +347,7 @@ service_triggers(){
        local name=$(basename ${script:-$initscript})

        procd_open_trigger
-       procd_add_raw_trigger "interface.*" 2000 /etc/init.d/$name reload
+       procd_add_raw_trigger "interface.*" 5000 /etc/init.d/$name reload
        procd_close_trigger

        procd_add_reload_trigger 'snmpd'
```

arrmo, 2021-03-18:
Sure, NP! Actually, will just edit `/etc/init.d/snmpd` on the router, reboot a few times. Will let you know.

arrmo, 2021-03-18:
Seems to work! Not sure I quite understand it ... LOL. At least for me, 5 reboots, no issues.

Thanks!

arrmo, 2021-03-26:
If this is working (for other as well), get it in the OpenWrt code? Thanks!

luizluca, 2021-03-26:
I also have this issue. I have dozens of equal conf but different hw routers (79) with this issue that happens randomly both on boot and sometimes when wan is restarted. My routers use VPN that might slow down the boot process.
I just tested the patch:

> ```diff
> -       procd_add_raw_trigger "interface.*" 2000 /etc/init.d/$name reload
> +       procd_add_raw_trigger "interface.*" 5000 /etc/init.d/$name reload
> ```

It didn't work. After a reboot, 31% (25/79) didn't had snmpd working.
I also tested with 9000 instead of 5000. “If brute force fails, you're not using enough.”
None failed this time.

> ```diff
> -       procd_add_raw_trigger "interface.*" 2000 /etc/init.d/$name reload
> +       procd_add_raw_trigger "interface.*" 9000 /etc/init.d/$name reload
> ```

@stintel , just increasing the timeout might not solve the issue but just make it less likely. Anyway, 5000 is too small, at least for me. Do you have any idea of what is happening? I noticed that this started to happen with me after I added a second
VPN connection (now I have three). Would it be related to the issue this [patch](http://lists.openwrt.org/pipermail/openwrt-devel/2021-March/034427.html) is trying to fix ?

As a workaround, I've been running this oneliner script every hour for months:

```
pgrep snmpd >/dev/null && ! [ -e /var/run/agentx.sock ] && /etc/init.d/snmpd restart && echo snmp restarted || :
```

It is dirty but do the job

stintel, 2021-11-26:
I am hitting this again, again on GS108Tv3. I just spotted something that might lead us closer to finding the root cause.

```
root@sw3:~# netstat -anput | grep snmpd
udp        0      0 0.0.0.0:161             0.0.0.0:*                           2745/snmpd
```
vs
```
root@sw3:~# netstat -anput | grep snmpd
udp        0      0 0.0.0.0:161             0.0.0.0:*                           3560/snmpd
udp        0      0 :::161                  :::*                                3560/snmpd
```
When snmpd is not listening on IPv6, it does not respond. When it does, it does.

hnyman, 2023-01-14:
@stintel 
Is this related to just the EoL 19.07?
or also newer versions?

stintel, 2023-01-15:
This happened on master before 22.03 also, iirc.