openwrt/packages #13750: Unbound: remove 'interface: 0.0.0.0 / ::0' from generated config - it is redundant, limiting and plain insecure

kernins, 2020-10-22:
Maintainer: @EricLuehrsen
Environment: 19.07

Description:
Please remove ```interface: 0.0.0.0``` (and similar ip6 wildcard) directives added by UCI's ```option protocol``` handler
1. this prevents one to configure unbound to listen only on ```lo``` (which is the default upstream behavior and doesn't require any cfg options) or some particular ifaces via ```unbound_srv.conf```
2. this is redundant - ```interface-automatic``` (UCI ```interface_auto```, enabled by-default) achieves similar outcome for those who want/need it
3. It is just nonsense to have non-overridable listen-on-wildcard and thus force users to patch security with acl and/or firewall rules or doing ```manual_conf 1```

I also see no reason for ```outgoing-interface: 0.0.0.0```. Default upstream behavior with no ```outgoing-interface``` specified is to use all available ifaces. So it is also redundant and limiting in terms of configuration flexibility/freedom.

EricLuehrsen, 2020-10-25:
The first goal behind UCI for Unbound was replicate UCI for dnsmasq as much as possible. This allows the package to drop in and replace. This is what most people receive with OpenWrt or through OEM basing their firmware on OpenWrt. 

The interface behavior is common. They both bind to all interfaces. dnsmasq has an option which only respects queries from the local subnet. Identical UCI option for Unbound does the same expanding into ACL. If you believe this is a security risk, then you should have a dialog on dnsmasq mailing list and openwrt-dev mailing list. You should attempt to close this alleged hole on installations that affect the most people. Whatever is common to the base OpenWrt, I will attempt replicate to Unbound.

The "interface-automatic" option is related, but it does not control initial binding. When Unbound is listening on multiple subnets, then it replies back on any of them in round robin. It uses a common  method to upstream recursion. This option tells Unbound to reply back on the exact route the query came in on. Consider a common configuration for owners WiFi and guest WiFi. Unbound will reply round robin from 192.168.0.1 and 192.168.1.1. However some clients reject answers from a server that doesn't have the IP they sent to.

Think about this and get back to me. Also please dial down the tone.

kernins, 2020-10-26:
>The interface behavior is common. They both bind to all interfaces

Yes, but still dnsmasq have the ```list interface``` in UCI to control this, unbound currently doesn't

>The "interface-automatic" option is related, but it does not control initial binding

It actually does. From man unbound.conf(5)
`Listen on all addresses on all (current and future) interfaces, detect the source interface on UDP  queries  and  copy  them  to replies...`
```on all (current and future) interfaces``` - exactly a wildcard-bind.
I've also tested it specifically on OpenWRT by commenting-out ```interface``` options in ```unbound.sh```, so generated "raw" daemon config will contain no ```interface``` entries. Then, with ```interface_auto``` enabled in UCI unbound binds to wildcard, without it - to localhost only, as expected. And user gains a clean way to specify ifaces he wants unbound to listen on via ```unbound_srv.conf```. UCI option to specify logical ifaces like one dnsmasq have will be even better

>If you believe this is a security risk, then you should have a dialog on dnsmasq mailing list and openwrt-dev mailing list

Sorry, I should have phrased it more specifically. What I meant is that leaving something wide-open, relying to upper security layers (i.e. netfilter) is generally insecure. OpenWRT in particular may have REJECT by default for wan, but in case of potential misconfig by user it may become wide open. That wouldn't happen if daemon, unbound in particular, would listen on lan/lo only, so impact of such potential misconfig would be limited. 
I'm not going to argue whether or not the current wildcard-bind behavior is acceptable as **default/initial** config, neither I'm asking/proposing to change it. There just definitely MUST be a way to reconfig it in a proper secure way without having to resort to plain daemon config and giving up UCI/fancy web UI.

BTW, there is also a bug in that code area
`ip6_local)
   echo "  interface: ::0"`
I think it should be ```::1``` for ip6_**local**

>Also please dial down the tone

I'm sorry if you felt my tone as offending, meant nothing like this

EricLuehrsen, 2020-10-28:
First the easy one, `ip6_local` is for ip6 ULA on LAN but your ISP provides no GLA for you on WAN. It is not loop back. Unbound will listen on IPv6. Unbound will not recurse on IPv6 and blowup the log with errors. Unbound finally has interface by name rather than subnet. Script complications like this might be dropped.

Again Unbound 1.1.12 can now bind to interfaces by name. However it remains to be seen if it tracks IPv6 DHCP-PD side effects well. Binding to `::0#53` simply works without problems. The protection is that Unbound can be asked to only answer queries that are from interfaces known subnets. Netfilter only protects from the ISP neighborhood subnet. Unbound probably will be most stable if the binding is still `::0#53` but perhaps UCI-to-ACL can be optionally more custom to block WAN.

The ACL is internal to Unbound and not tied to the OS, so they don't cause crashes. The interface bindings could fail or crash (at least at one time) if an interface address changed depending on the nature of the change.

kernins, 2020-10-30:
Hmm, I don't get it, why you're so reluctant to make such a simple change?

I'm not talking about changing default behavior or forcing some particular config, on the contrary - I'm talking about *not to force* it.

As shown above, we already have an option to tell unbound to listen on any iface - `interface_auto`. If you think there could be issues with that option alone, you can keep your current `interace`-adding UCI options, just add an option to *not add* any `interface` directives. This would give a user freedom in configuring his instance. This is my only point - *avoid forcing* any particular config *unless absolutely* necessary. Everything else is just implementation details.

ACL is nice on its own. But not a solution to "avoid forcing" issue, just workaround.

>The interface bindings could fail or crash

I think you're trying to "protect" unexperienced users too much, to the point where it becomes limiting. Sane and robust defaults are enough, and for not so robust options a simple warning about possible corner cases should be fine.

Anyway, ACL and iface selector in UCI are both new features, somewhat relevant but not required. It's another story whether and how to implement them.

EricLuehrsen, 2020-10-30:
It is not reluctance to a simple script change. It is not protecting inexperienced users. It is wrestling with defects or non-robustness in OpenWrt and Unbound. 

Someone thought is was a good idea to have netifd raise (IFUP for trivial IPv6](https://bugs.openwrt.org/index.php?do=details&task_id=713&order=status&sort=desc) updates on WAN. procd has poor filtering choices so then caching servers which use an `ineterface.*` style restart would flush and restart every 2 minutes or so. The fix was to give users the ability to set that trigger to something else, only LAN. This had issues with crashes. A fixed address which may go up or down, such as mesh or repeater WiFi configuration. Procd reloads crashing programs, but it does not trigger a full script restart without a trigger. So repeat crash. Solution was bind wildcard and Unbound just worked.

An annoying defect on Unbounds part is that it floods the log file with broken IPv6 connectivity, even when its told not to use IPv6. Explicitly setting interfaces helped squelch some of that. They way that can be done was limited by the wall of text above. Their is still log noise and no good way to snub all of it.

I am not yet sure `interface-automatic:` works exactly the way we would hope it will. If it does, then fantastic, but I need to test some first to avoid regression respective of above.

Unbound 1.1.12 finally allows selecting upstream and downstream interfaces by name (like dnsmasq or anything else). It should automatically use that interfaces subnet(s). This would be an easy UCI addition and in the scheme of working through above, provide a tool to work with unforeseen side effects.

Whether Unbound binds to an interface or binds to all interfaces and uses ACL, the reality is the same. Either the OS drops the packet because no server is listening, or Unbound drops the packet because it came from the wrong place. The UDP packet is just data until an application acts on it. I agree it is philosophically cleaner to not bind to an interface you don't use, but in this case I don't see a material consequence to the practical solution. As I attempt to work through all the above, what remainder falls out does at it does.

EricLuehrsen, 2020-11-01:
(I will try to summarize my comment which I just stupidly deleted)
The reluctance is not against a simple script change. Nor, was this to protect inexperienced users. The UCI translations were emperical as different defects and non-robustness issues were discovered in OpenWrt and Unbound. OpenWrt introduced some poor design choices in [netifd IFUP](https://bugs.openwrt.org/index.php?do=details&task_id=713&string=netifd&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=) and also didn't put any respective filtering in procd. The solutions to flushing DNS cache every two minutes had potential crashes in some use cases. Whether the ACL or OS drops packets, practically its the same thing. Philosophically I agree its better to not bind to unused addresses, but whatever works comes first.

kernins, 2020-11-02:
I still got your deleted comment on email though

>Nor, was this to protect inexperienced users
>It is wrestling with defects or non-robustness in OpenWrt and Unbound

I meant exactly that. And I completely support you in this, it just shouldn't be limiting.

>Whether the ACL or OS drops packets, practically its the same thing

Yes, but only while everything is configured and working as expected. Even with properly configured internal ACL there still may be potential attack surface due to possible bugs in implementation (ofc, Linux IP stack may also have bugs, but it is much more mature and widely tested, and nothing stops one to use ACL, netfilter, etc as additional measures). Also, as you've already said, proper binding is just cleaner, not only philosophically, but also practically - a brief look on `netstat -ln` output is enough to see a service is protected, as opposed to further digging into config or netfilter rules.

What do you think about my latest proposal - to introduce another option for UCI's `protocol`, lets call it "Manual", that won't be adding any hardcoded values to unbound's config. With current script, taking `default` as a base, it would be only `edns-buffer-size` and `port` which are UCI-configurable. With that option selected other related directives still may be configured in `unbound_srv.conf` according to user's requirements or just left to unbound's defaults. This would also provide a way to publicly test `interface-automatic` in relation to binding and collect feedback. 

The defaults and existing options remain the same, so no regression could possibly occur

PR #13841 (2020-11-05): unbound: update to 1.1.12
Maintainer: me
Tested: Linksys WRT3200ACM
Description: resolves #13178, resolves #13623, resolves #13750, resolves #13852
If DoH is enabled, then a few glitches in the upstream tuning options that are being worked out. Default compile option is disabled, but it is available if someone wants to try it.
closed 2020-11-05 by neheb