openwrt/packages #13754: nginx-util: compilation failure

jammmet, 2020-10-23:
Maintainer: @peter-stadler 
Environment: ramips, zbt-we826-16MB, master

Description:

Building nginx-util fails

.config:

```
CONFIG_PACKAGE_nginx=y
# CONFIG_PACKAGE_nginx-all-module is not set
CONFIG_PACKAGE_nginx-mod-luci=y
# CONFIG_PACKAGE_nginx-ssl is not set
# CONFIG_PACKAGE_nginx-ssl-util is not set
CONFIG_PACKAGE_nginx-ssl-util-nopcre=y
```

output

```
make[3]: Entering directory 'feeds/packages/net/nginx-util'
rm -f build_dir/target-mipsel_24kc_musl/nginx-util-1.4/.built
touch build_dir/target-mipsel_24kc_musl/nginx-util-1.4/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremapbuild_dir/target-mipsel_24kc_musl/nginx-util-1.4:nginx-util-1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -Istaging_dir/target-mipsel_24kc_musl/usr/include -Istaging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/usr/include -Istaging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include/fortify -Istaging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremapbuild_dir/target-mipsel_24kc_musl/nginx-util-1.4:nginx-util-1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -Istaging_dir/target-mipsel_24kc_musl/usr/include -Istaging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/usr/include -Istaging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include/fortify -Istaging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/include " LDFLAGS="-Lstaging_dir/target-mipsel_24kc_musl/usr/lib -Lstaging_dir/target-mipsel_24kc_musl/lib -Lstaging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/usr/lib -Lstaging_dir/toolchain-mipsel_24kc_gcc-7.5.0_musl/lib -znow -zrelro " make -j1 -C build_dir/target-mipsel_24kc_musl/nginx-util-1.4/. AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremapbuild_dir/target-mipsel_24kc_musl/nginx-util-1.4:nginx-util-1.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CROSS="mipsel-openwrt-linux-musl-" ARCH="mipsel" CMAKE_COMMAND='staging_dir/host/bin/cmake' CMAKE_DISABLE_cmake_check_build_system=1 ;
make[4]: Entering directory 'build_dir/target-mipsel_24kc_musl/nginx-util-1.4'
make[5]: Entering directory 'build_dir/target-mipsel_24kc_musl/nginx-util-1.4'
make[6]: Entering directory 'build_dir/target-mipsel_24kc_musl/nginx-util-1.4'
make[6]: Leaving directory 'build_dir/target-mipsel_24kc_musl/nginx-util-1.4'
make[6]: Entering directory 'build_dir/target-mipsel_24kc_musl/nginx-util-1.4'
[ 25%] Building CXX object CMakeFiles/nginx-ssl-util-nopcre.dir/nginx-util.cpp.o
In file included from build_dir/target-mipsel_24kc_musl/nginx-util-1.4/nginx-util.hpp:17:0,
                 from build_dir/target-mipsel_24kc_musl/nginx-util-1.4/nginx-util.cpp:3:
build_dir/target-mipsel_24kc_musl/nginx-util-1.4/ubus-cxx.hpp: In constructor 'ubus::message::message(ubus::msg_ptr, ubus::strings)':
build_dir/target-mipsel_24kc_musl/nginx-util-1.4/ubus-cxx.hpp:139:5: error: declaration of 'filter' shadows a member of 'ubus::message' [-Werror=shadow]
     : msg{std::move(message_ptr)}, keys{std::move(filter)} {}
     ^
build_dir/target-mipsel_24kc_musl/nginx-util-1.4/ubus-cxx.hpp:166:10: note: shadowed declaration is here
     auto filter(Strings ...filter)
          ^~~~~~
build_dir/target-mipsel_24kc_musl/nginx-util-1.4/ubus-cxx.hpp: In member function 'auto ubus::message::filter(Strings ...)':
build_dir/target-mipsel_24kc_musl/nginx-util-1.4/ubus-cxx.hpp:167:5: error: declaration of 'filter' shadows a member of 'ubus::message' [-Werror=shadow]
     {
     ^
build_dir/target-mipsel_24kc_musl/nginx-util-1.4/ubus-cxx.hpp:166:10: note: shadowed declaration is here
     auto filter(Strings ...filter)
          ^~~~~~
cc1plus: all warnings being treated as errors
CMakeFiles/nginx-ssl-util-nopcre.dir/build.make:81: recipe for target 'CMakeFiles/nginx-ssl-util-nopcre.dir/nginx-util.cpp.o' failed
make[6]: *** [CMakeFiles/nginx-ssl-util-nopcre.dir/nginx-util.cpp.o] Error 1
make[6]: Leaving directory 'build_dir/target-mipsel_24kc_musl/nginx-util-1.4'
CMakeFiles/Makefile2:96: recipe for target 'CMakeFiles/nginx-ssl-util-nopcre.dir/all' failed
make[5]: *** [CMakeFiles/nginx-ssl-util-nopcre.dir/all] Error 2
make[5]: Leaving directory 'build_dir/target-mipsel_24kc_musl/nginx-util-1.4'
Makefile:148: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory 'build_dir/target-mipsel_24kc_musl/nginx-util-1.4'
Makefile:74: recipe for target 'build_dir/target-mipsel_24kc_musl/nginx-util-1.4/.built' failed
make[3]: *** [build_dir/target-mipsel_24kc_musl/nginx-util-1.4/.built] Error 2
make[3]: Leaving directory 'feeds/packages/net/nginx-util'
time: package/feeds/packages/nginx-util/compile#5.78#0.51#6.08
package/Makefile:112: recipe for target 'package/feeds/packages/nginx-util/compile' failed
make[2]: *** [package/feeds/packages/nginx-util/compile] Error 2
make[2]: Leaving directory '/home/builder/oneboxx-2020-master'
package/Makefile:108: recipe for target 'staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/oneboxx-2020-master'
include/toplevel.mk:234: recipe for target 'world' failed
make: *** [world] Error 2
```

peter-stadler, 2020-10-23:
Thank you, I will look into it.

peter-stadler, 2020-10-23:
I renamed `filter` to `key_filter` for the [new version](https://github.com/openwrt/packages/pull/13405). It should solve this errror, although I could not reproduce it; do you know, which compiler version is used in your case?

jammmet, 2020-10-27:
Thanks Peter - using `key_filter` fixed the issue. I had tried with both GCC 7 and 10 (same issue).

closed 2020-10-27 by jammmet