openwrt/packages #13757: <Domoticz>: Build failure with python 3.9

kofec, 2020-10-24:
@neheb I also raised issue here as this is openwrt specific:
https://github.com/domoticz/domoticz/issues/4415

Now we have a issue with python 3.9
https://github.com/domoticz/domoticz/issues/3703

/* _Py_DEC_REFTOTAL macro has been removed from Python 3.9 by:
    https://github.com/python/cpython/commit/49932fec62c616ec88da52642339d83ae719e924 */

So we need to modify:
https://github.com/domoticz/domoticz/pull/3713

The solution is here. Prepared by Fedora team:
https://src.fedoraproject.org/rpms/domoticz/c/0cb4a4b4734638b8b07177713455611746b74185

But it still didn't work my case:
```
/home/openwrt/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /home/openwrt/openwrt/tmp/ccHRxNhU.ltrans4.ltrans.o: in function `py3__Py_DECREF(char const*, int, _object*) [clone .isra.13] [clone .lto_priv.3954] [clone .lto_priv.4005]':
<artificial>:(.text+0x1698): undefined reference to `_Py_Dealloc'
collect2: error: ld returned 1 exit status
make[5]: *** [domoticz] Error 1
make[5]: Leaving directory `/home/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/domoticz-805fee10fd609a02182eaa01e99bcb4b1b560426'
make[4]: *** [CMakeFiles/domoticz.dir/all] Error 2
make[4]: Leaving directory `/home/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/domoticz-805fee10fd609a02182eaa01e99bcb4b1b560426'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/domoticz-805fee10fd609a02182eaa01e99bcb4b1b560426'
make[2]: *** [/home/openwrt/openwrt/build_dir/target-mipsel_24kc_musl/domoticz-805fee10fd609a02182eaa01e99bcb4b1b560426/.built] Error 2
make[2]: Leaving directory `/home/openwrt/openwrt/feeds/packages/utils/domoticz'
time: package/feeds/packages/domoticz/compile#171.69#11.61#178.54
make[1]: *** [package/feeds/packages/domoticz/compile] Error 2
make[1]: Leaving directory `/home/amk082/openwrt'
make: *** [package/domoticz/compile] Error 2
```

neheb, 2020-10-24:
Worked in my VM.

kofec, 2020-10-24:
By VM you mean Virtual Machine? Are you using master branch with python3.9 ? If so IT is strange as _Py_DEC_REFTOTAL macro has been removed from Python 3.9

neheb, 2020-10-24:
I'll post a patch when I can.

kofec, 2020-10-24:
I wilk try with gcc 10

kofec, 2020-10-24:
Maybe this is something with my environment:
```
[  3%] Linking CXX executable domoticz
/home/openwrt/openwrt//staging_dir/toolchain-mipsel_24kc_gcc-10.2.0_musl/lib/gcc/mipsel-openwrt-linux-musl/10.2.0/../../../../mipsel-openwrt-linux-musl/bin/ld: /home/openwrt/openwrt//tmp/domoticz.oZBfUT.ltrans76.ltrans.o: in function `py3__Py_DECREF(char const*, int, _object*) [clone .constprop.0]':
<artificial>:(.text+0x19c): undefined reference to `_Py_Dealloc'
collect2: error: ld returned 1 exit status
make[6]: *** [domoticz] Error 1
make[6]: Leaving directory `/home/openwrt/openwrt//build_dir/target-mipsel_24kc_musl/domoticz-805fee10fd609a02182eaa01e99bcb4b1b560426'
make[5]: *** [CMakeFiles/domoticz.dir/all] Error 2
make[5]: Leaving directory `/home/openwrt/openwrt//build_dir/target-mipsel_24kc_musl/domoticz-805fee10fd609a02182eaa01e99bcb4b1b560426'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/openwrt/openwrt//build_dir/target-mipsel_24kc_musl/domoticz-805fee10fd609a02182eaa01e99bcb4b1b560426'
make[3]: *** [/home/openwrt/openwrt//build_dir/target-mipsel_24kc_musl/domoticz-805fee10fd609a02182eaa01e99bcb4b1b560426/.built] Error 2
make[3]: Leaving directory `/home/openwrt/openwrt//feeds/packages/utils/domoticz'
time: package/feeds/packages/domoticz/compile#170.40#12.66#177.66
make[2]: *** [package/feeds/packages/domoticz/compile] Error 2
make[2]: Leaving directory `/home/openwrt/openwrt/'
make[1]: *** [/home/openwrt/openwrt//staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/openwrt/openwrt/'
make: *** [world] Error 2
```
I will experiment more

steelrat2000, 2020-10-25:
Hello. Thanks to this amendment I was able to compile Domoicz for openwrt.
I don't know if it's correct but it works.

`diff -ruN b/DelayedLink.h a/DelayedLink.h
--- b/hardware/plugins/DelayedLink.h	2020-10-20 08:53:17.000000000 +0200
+++ a/hardware/plugins/DelayedLink.h	2020-10-25 19:39:50.197422300 +0100
@@ -13,6 +13,17 @@
 #include <structmember.h>
 #include <frameobject.h>
 #include "../../main/Helper.h"
+#ifndef _Py_DEC_REFTOTAL            
+ /* _Py_DEC_REFTOTAL macro has been removed from Python 3.9 by:            
+https://github.com/python/cpython/commit/49932fec62c616ec88da52642339d83ae719e924 */            
+#ifdef Py_REF_DEBUG
+#define _Py_DEC_REFTOTAL _Py_RefTotal--
+#else
+#define _Py_DEC_REFTOTAL
+#define _Py_Dealloc
+#endif
+#endif
+
 
 #if PY_VERSION_HEX >= 0x030800f0
 static inline void
`

closed 2020-11-20 by neheb