openwrt/packages #13758: Samba privilege escalation exploit (low severity)

hauke, 2020-10-24:
Maintainer: @Andy2244
Environment: OpenWrt master and OpenWrt 19.07

Description:
There is a privilege escalation problem in the samba init script:
https://github.com/full-disclosure/FDEU-CVE-2020-1FC5

The author of this problem contacted us some time ago, sorry for the delay:
{quote}
Hi,

we would like to report a vulnerability in /etc/init.d/samba script which may allow to make arbitrary changes in smb.conf, such as enable symlink and set root user for guest

Vulnerable code is:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/network/services/samba36/files/samba.init;h=1c5bb3b3c43eacc6ee3a181a16b63c906365b81b;hb=refs/heads/openwrt-18.06#l32

  32         sed -e "s#|NAME|#$name#g" \
  33             -e "s#|WORKGROUP|#$workgroup#g" \
  34             -e "s#|DESCRIPTION|#$description#g" \
  35             -e "s#|INTERFACES|#$interfaces#g" \
  36             -e "s#|CHARSET|#$charset#g" \
  37             /etc/samba/smb.conf.template > /var/etc/smb.conf

The variables $name, $workgroup and others passed into sed must be sanitized and all symbols such "#" replaced

An example of an exploit is this value for "Description":

pwned#g ; s#follow symlinks = no#follow symlinks = yes#g ; s#wide links = no#wide links = yes#g ; s#security = share#security = user#g ; s#guest account = nobody#guest account = root

This will enable R/W access on the symlink pointed to "/"

The severity for OpenWRT is probably low, as only admin can change smb.conf template. But in other systems, like Technicolor this allows local privilege escalation

As soon as Luci releases restricted users functionality - this would also become an issue for privilege escalation

A PoC is available that attacks Technicolor routers:

https://github.com/full-disclosure/FDEU-CVE-2020-1FC5

Hope this helps,

Full Disclosure team
{quote}

I agree that the severity for OpenWrt is low, but this should get fixed anyway.
This was reported for the old samba3 package which is already removed in OpenWrt master, but the same code is also in the samba4 package. 

If someone wants to develop a fixes, I can take care of backporting it to samba3 package in OpenWrt 19.07 and 18.06.

Andy2244, 2020-10-24:
ok, can try to find a fix for this, but atm i'm a little time restricted. Will probably just try to get the next releases ready and after this may look more into this issues.

Thanks for reporting this again.

Andy2244, 2020-12-11:
> I agree that the severity for OpenWrt is low, but this should get fixed anyway.

While we can sanitize/filter the simple inputs like "DESCRIPTION", i see no actual way to make the template edit field secure, the samba config is just too big and has to many parameters that we cant sanitize.

So without removing/restricting the template edit functionality there is no real "fix" for this issue in openwrt.
I will add a fix for the simple external parameters, but downstream distros would have to remove the template edit field option from there UI's to actually fix the issue.

commit 4c373ec7 (2021-01-26): ksmbd-tools: fix #13758

* fix for possible exploit #13758
* sanetize all external template/config inputs
* fix some shellcheck warnings
commit d2e8582b (2021-01-26): Merge pull request #14576 from Andy2244/ksmbd-fix-#13758

ksmbd-tools: fix #13758
closed 2021-01-26 by neheb