openwrt/packages #13759: mwan3: $DEVICE environment variable not populated on ifdown event

jamesmacwhite, 2020-10-24:
Maintainer: @feckert
Environment: OpenWrt 19.07.3 r11063-85e04e9f46 / LuCI openwrt-19.07 branch git-20.234.16293-a82f67e

Description:

The $DEVICE environment variable does not seem to be populated on an ifdown event. I implemented some custom email notifications functionality within `/etc/mwan3.user` recently, to test I restarted a network interface being tracked by mwan3. This triggered four email notifications, each for the disconnected, ifdown, ifup and connected actions. On the ifdown action, the $DEVICE value was blank, for all others it was shown.

ptpt52, 2020-10-24:
ping @aaronjg

aaronjg, 2020-10-24:
@jamesmacwhite I believe this is an issue with OpenWRT. Can you add a hotplug script outside of mwan3, disable mwan3 and check if the DEVICE is set on ifdown?

jamesmacwhite, 2020-10-24:
@aaronjg You seem to be correct here, it's not a mwan3 specific problem

This is the custom hotplug I made for testing:

```
[ "${ACTION}" = "ifup" ] && {
    logger -t hotplug "Device: ${DEVICE} / Action: ${ACTION}"
} 
```

The output was the following, when reloading the network interface

```
Sat Oct 24 17:47:57 2020 user.notice hotplug: CUSTOM HOTPLUG: Device:  / Action: ifdown
```

`$DEVICE` was indeed blank.

aaronjg, 2020-11-01:
I'm not sure why it is not set on ifdown. It could be that in some cases, such as tunnel devices, the device will have been destroyed by the time the ifdown hotplug event happens. Maybe file a bug report/feature request with a use case for having the DEVICE set in the ifdown event call.

jamesmacwhite, 2020-11-02:
@aaronjg Interesting theory. Although I don't think it is specifically related to tunnel related interfaces. I tested it on an independent WAN connection through Ethernet, the only unique element would be using a LAN port as a WAN port via a VLAN config, but otherwise it's pretty much a standard network interface. It isn't really an issue, just maybe worth highlighting in the docs for `/etc/mwan3.user` that $DEVICE isn't available on ifdown.

aaronjg, 2020-11-02:
> It isn't really an issue, just maybe worth highlighting in the docs for `/etc/mwan3.user` that $DEVICE isn't available on ifdown.

Yeah, I think that makes sense. Want to make a P/R to update the header comments for mwan3.user to include this and the ifupdate event?

jamesmacwhite, 2020-11-06:
Staged a PR, @feckert wasn't keen on ifupdate because it has no direct usage in mwan3track, but I guess my point is if you configure something like custom notifications and aren't specifically targeting any $ACTION value, you may see ifupdate, but it's not mentioned specifically in the mwan3.user, it is covered in the main hotplug docs though: https://openwrt.org/docs/guide-user/base-system/hotplug

feckert, 2020-11-09:
I'm not sure if I want that in mwan3.
Then I will have to intercept all other [events](https://git.openwrt.org/?p=project/netifd.git;a=blob;f=interface-event.c;hb=ed11f0c0ffe4fdacfe3f8223049ef8a61d9c53e9#l33) as well.
In mwan3 I currently only use ifup and ifdown actions.
The mwan3 itself creates connected and disconnected actions.

aaronjg, 2020-11-09:
> Then I will have to intercept all other [events](https://git.openwrt.org/?p=project/netifd.git;a=blob;f=interface-event.c;hb=ed11f0c0ffe4fdacfe3f8223049ef8a61d9c53e9#l33) as well.

Interesting. `ifup-failed`, `free`, `reload`, `iflink`, and `create` aren't listed in the hotplug doc that James mentioned, do those in fact generate hotplug actions?. I think it is nice to have the list of values that `$ACTION` can take in the hotplug script so the user can write their own handler for them and not get unexpected values. 

At the very least, mwan3.user should have a comment that those four actions are not an exhaustive list of possible values for `$ACTION`.

The other question is do we actually need to respond to the ifupdate events. I think yes, if an interface changes ip address or device name, mwan3track needs to be made aware. However, I have not yet run into any issue where this happens so I can't test a fix. Nevertheless, I have a draft P/R that should fix the issue if it comes up:
https://github.com/openwrt/packages/pull/13889

jamesmacwhite, 2020-11-09:
I have an interface that regularly triggers an ifupdate event (approx hourly), it is an IPv6 alias interface to an L2TP connection, it is using DHCPv6 to get a static delegated /48 prefix over a tunnel. I'm assuming the ifupdate events are DHCPv6 related, perhaps a RS packet or similar. In my case the interface properties i.e. IPv6 address don't change as it's truly static IPv6 addresses with a fixed /48 prefix. What led me to realise this was hooking into the mwan3.user file with no direct condition for the $ACTION value and saw the event occurring regularly, which wasn't listed in default file. It is the only interface that seems to do it, so it might even be L2TP specific too.

I don't think mwan3 not handling ifupdate is breaking anything and I wasn't aware there were even more hotplug type events not listed on the OpenWrt docs, so maybe that needs to be amended first. Perhaps, a reference to $ACTION values not being exhaustive and then linking to an updated source is best to cover off that issue, so keeping the main mwan3 events listed and then clarifying, non mwan3 events may occur that the mwan3.user file can pick up?

Maybe I have opened up a larger talk, but I think the key one originally was $DEVICE isn't available on ifdown, so if anything that should be mentioned in the /etc/mwan3.user, in case someone wanted to write some custom logic around that, only $INTERFACE is usable for that event. Although to get that fixed sounds like it's needed in core OpenWrt.

aaronjg, 2020-11-10:
>  It is the only interface that seems to do it, so it might even be L2TP specific too. 

I don't think it is L2TP specific. I see them on my device too, and it is likely after getting a router advertisement message.

> I don't think mwan3 not handling ifupdate is breaking anything 

Yeah, I haven't seen it break anything yet, but I changed mwan3track to get the source ip address after an ifup, rather than in every tracking loop, so there is the possibility that things could be broken on DHCPv4 when an IP address changes. I need to create a testing set up so that I can recreate the issue though.

> Perhaps, a reference to $ACTION values not being exhaustive and then linking to an updated source is best to cover off that issue

Makes sense to me. I had run into a similar issue to you early on where I thought I was handling all the events, but then ended up sending way to many monitoring emails.

> Maybe I have opened up a larger talk, but I think the key one originally was $DEVICE isn't available on ifdown, so if anything that should be mentioned in the /etc/mwan3.user

Yeah, sorry for taking things off topic. Seems like it could be helpful to have that in a comment in `mwan3.user` so that authors of their own hotplug scripts will know not to rely on $DEVICE on `ifdown` events.

jamesmacwhite, 2020-11-13:
I noticed that the official OpenWrt docs were lacking much about the `/etc/mwan3.user` file, so I've expanded the section with some information for now:

https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#alertsnotifications

jamesmacwhite, 2020-11-16:
Documentation updated in `/etc/mwan3.user` file with reference to the OpenWrt Wiki information recently added.

closed 2020-11-16 by jamesmacwhite