openwrt/packages #13760: mwan3: ifupdate is not referenced in example /etc/mwan3.user file

jamesmacwhite, 2020-10-24:
Maintainer: @feckert 

Description:

The action `ifupdate` exists, but is not referenced in the example `/etc/mwan3.user`. I don't think this is used by mwan3, but more general to OpenWrt. Should it be referenced?

aaronjg, 2020-10-25:
Yes, and it is possible that mwan3 is not handling this correctly. What sort of thing are you seeing trigger an ifupdate event? What parameters on the network are being updated?

jamesmacwhite, 2020-10-25:
I've noticed it since implementing custom email notifications, however I've now wrapped ithis with a conditional to ignore ifupdate, because it seems to be happening regularly on a specific network device and I assume it's just related to DHCPv6. In my case it is a L2TP network device regularly having an ifupdate action, which is specifically my wan6 interface which is an alias interface using DHCPv6. So I assume it's related to constant polling with DHCPv6, possible router solicitation or similar.

aaronjg, 2020-10-25:
Got it. Those routing updates should then be handled properly by mwan3rtmon, so likely won't cause problems. However, let me know if any issues come up.

We should update the comment on the example to advise users of this additional type of routing event.

jamesmacwhite, 2020-10-25:
Cool. Yeah it is more for documentation purposes than anything, there is no bug here, just an additional event not mentioned in the `/etc/mwan3.user` file.

ptpt52, 2020-10-26:
ifupdate event comes up when ip changes happen on the interface

jamesmacwhite, 2020-10-26:
In my case the interface is an IPv6 alias to a L2TP network device and the IP is static for both IPv4 and IPv6. I'm assuming it's DHCPv6 triggering it.

feckert, 2020-10-27:
The question is do we need the **ifupdate** event. This comes for netifd.
I think this has nothing to do with the mwan3.
The mwan3 currently reacts to ifup and ifdown events.
And itself it creates a connected and disconnected event.

This has not changed in the latest version.
This has been the case for several versions.

x-wrt-dev, 2020-10-27:
sometimes ifupdate change the interface ip address

jamesmacwhite, 2020-10-27:
Yes, I don't think is directly related to mwan3, but should it be mentioned in the comments on the `/etc/mwan3.user file? If you use it $ACTION could be ifupdate, but it isn't mentioned directly, so more of a documentation question more than anything.

feckert, 2020-10-29:
Lets close this because this is not related to mwan3.
Up to now nothing happens on ifupdate event in mwan3.

closed 2020-10-29 by feckert