openwrt/packages #13769: Merge PR #10428 into openwrt-19.07 branch

jgora, 2020-10-26:
Hi

Could anyone merge issue #10373 solution in PR #10428 by @Yonoesio into openwrt-19.07 branch?

Regards

neheb, 2020-10-29:
No.

I just tried.

```
++<<<<<<< HEAD
 +PKG_VERSION:=2.14
 +PKG_RELEASE:=8
 +PKG_HASH:=d215a9a8b8e66aae04b982fa2e1228e8a71e7dfe42320df99e34e5000cbdf152
++=======
+ PKG_VERSION:=3.7
+ PKG_RELEASE:=2
+ PKG_HASH:=045ee3fcfe5d6edb95483cf49e2446ff8cab4bb79411018b18354138b70c5eb5
++>>>>>>> 3aa525157... xtables-addons: Support IPTV Timeshift
```

jgora, 2020-10-29:
And could be possible a new PR for backport to openwrt-19.07 branch the changes of net/xtables-addons/patches/100-add-rtsp-conntrack.patch in this commit https://github.com/openwrt/packages/commit/3aa52515720b75ada15777ab8d14c1d253921638 ?

Commented in https://github.com/openwrt/packages/pull/10495#issuecomment-575415582

jgora, 2020-10-30:
Hi @neheb , I just created PR https://github.com/openwrt/packages/pull/13809

closed 2020-10-31 by jgora