openwrt/packages #13771: sshtunnel: Add support for "ProxyCommand"

lars18th, 2020-10-26:
Maintainer: @nunojpg
Environment: all

Description:

Current implementation of the `sshtunnel` package can only stablish TCP connections to the SSH server. However, the open-ssh client can connect using a thirdparty helper tool. The objective with this tools is to stablish a previous socket tunnel, for example over a HTTP Proxy (using the CONNECT command for HTTPS). So I suggest to update the init script with support for the ProxyCommand:

https://github.com/openwrt/packages/blob/master/net/sshtunnel/files/sshtunnel.init

I'm not requesting to add full HTTP CONNECT PROXY support. Only to add the option of `ProxyCommand` inside the script. Then the user will be the responsible of configuring this parameter with the correct data. For example, using the examples described here:

https://stackoverflow.com/questions/19161960/connect-with-ssh-through-a-proxy

With this _small_ enhancement, this package will can stablish _automatically_ a tunnel inside very restricted environments. And without using external VPN tools. So a huge improvement.

Please, consider this.
Regards.

nunojpg, 2020-10-26:
Yes, I've needed such in the past. Feel free to submit a pull request.

closed 2020-10-26 by nunojpg

lars18th, 2020-10-26:
Hi @nunojpg ,

> Yes, I've needed such in the past. Feel free to submit a pull request.

This is a request. At time I can't implement it and provide a PR.
So please, reopen this issue until someone can be assigned to it.
Regards.

lars18th, 2020-10-27:
Hi @nunojpg ,

> Yes, I've needed such in the past. Feel free to submit a pull request.

I really need it, so I implemented it.
Here it's it: https://github.com/openwrt/packages/pull/13787

Please, review it and merge.
Regards.

lars18th, 2020-10-28:
Hi @nunojpg ,

Please, can you review and merge my updated version? https://github.com/openwrt/packages/pull/13789
Thank you!