openwrt/packages #13773: miniupnpd: Port forwarding not work when the router is behind NAT

dianlujitao, 2020-10-26:
Maintainer: @neheb 
Environment: NETGEAR WNDR4300 / Atheros AR9344 rev 2 / OpenWrt 19.07.4 r11208-ce6496d796 / LuCI openwrt-19.07 branch git-20.287.57033-3d52019

Description:

Miniupnpd of OpenWrt 19.07.4 refuses to start port forwarding when the router is behind NAT, while the one from 18.06 works. There are two factors in my understanding:

## New miniupnpd requires STUN protocol to be explicitly configured in its .conf file when the router is behind NAT, or it'll refuse to start.

Here's the logging with default configurations:

```
Mon Oct 26 22:31:26 2020 daemon.info miniupnpd[12751]: version 2.1.20200510 starting NAT-PMP/PCP UPnP-IGD ext if pppoe-wan BOOTID=1603722686
Mon Oct 26 22:31:26 2020 daemon.info miniupnpd[12751]: specific IPv6 ext if pppoe-wan
Mon Oct 26 22:31:26 2020 daemon.info miniupnpd[12751]: Reserved / private IP address 10.194.xx.xx on ext interface pppoe-wan: Port forwarding is impossible
Mon Oct 26 22:31:26 2020 daemon.info miniupnpd[12751]: You are probably behind NAT, enable option ext_perform_stun=yes to detect public IP address
Mon Oct 26 22:31:26 2020 daemon.info miniupnpd[12751]: Or use ext_ip= / -o option to declare public IP address
Mon Oct 26 22:31:26 2020 daemon.notice miniupnpd[12751]: HTTP listening on port 5000
......
```
The restrictions on private IP was introduced with upstream commits https://github.com/miniupnp/miniupnp/commit/8e10a1aeab9b8cd4d3b2e964b02e9ad409cf3aaa https://github.com/miniupnp/miniupnp/commit/8c97654d70fdc46963dc817cd5b4fc5764b002a9 which were NOT present in 18.06's miniupnpd

Following the logs, I'm going to enable `ext_perform_stun=yes` and luckily there's already uci options, just not implemented in luci, I believe this is a luci issue and reported in https://github.com/openwrt/luci/issues/4544

## The router is behind restrictive NAT

I don't quite understand `restrictive NAT`, but that's what miniupnpd told me after enabling STUN:\

```
Mon Oct 26 22:32:27 2020 daemon.notice miniupnpd[12928]: perform_stun: 1 response out of 4 received
Mon Oct 26 22:32:27 2020 daemon.notice miniupnpd[12928]: perform_stun: #0 external address or port changed
Mon Oct 26 22:32:27 2020 daemon.warn miniupnpd[12928]: STUN: ext interface pppoe-wan with IP address 10.194.xx.xx is now behind restrictive NAT with public IP address 117.xx.xx.xx: Port forwarding is now impossible
Mon Oct 26 22:32:27 2020 daemon.notice miniupnpd[12928]: HTTP listening on port 5000
......
```

## Workaround

Since I didn't found any config options to disable the restrictive NAT check, I chose to downgrade to 18.06's miniupnpd. Specifically, I downloaded the old build from https://downloads.openwrt.org/releases/18.06.8/packages/mips_24kc/packages/miniupnpd_2.1-1_mips_24kc.ipk , extracted `/usr/sbin/miniupnpd` and replaced its dependencies `libip4tc.so.0`, `libip6tc.so.0` with `libip4tc.so.2`, `libip6tc.so.2`, respectively. Then pushed to the router and upnp started working.

I believe my issue is similar to https://github.com/openwrt/luci/issues/3471 and expect proper fixes from upstream.

ptpt52, 2020-10-26:
I have a workaround on this issue

https://github.com/x-wrt/packages/commit/8901950eef71a68ab407c39a826fc20ece8e07c8

dianlujitao, 2020-10-27:
> I have a workaround on this issue
>
> x-wrt/packages@8901950

Maybe upstream to miniupnpd?

neheb, 2020-10-27:
I don't maintain this.

Which means PRs welcome.

curtdept, 2020-11-02:
Luci support was merged and is waiting on this

ptpt52, 2020-11-18:
> […](#)

It just a hack workaround, won't be accepted by upstream

neheb, 2020-11-25:
Make a PR at least. It seems a fair amount of people are running into this issue.

ptpt52, 2020-11-26:
ok, PR on upstream https://github.com/miniupnp/miniupnp/pull/511

neheb, 2020-11-26:
I meant here.

closed 2020-11-26 by neheb

Neustradamus, 2021-09-16:
There is a new patch from @ptpt52:
- https://github.com/openwrt/packages/pull/16011#issuecomment-920538698

It replaces the old 301-ext_ip_reserved_ignore.patch.

Tested by @openips!

largeword, 2022-03-17:
For someone who encountered with this issue.

I use OpenWrt 21.02 and miniupnpd 2.2, and the log shows error
`Thu Mar 17 14:09:21 2022 daemon.notice miniupnpd[15353]: private/reserved address 10.161.153.189 is not suitable for external IP
`
After referring article [https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=4055380&extra=page%3D1&ordertype=1](url), I finally fix the issue of the upnp feature in my router.

alexdelli, 2022-04-06:
> For someone who encountered with this issue.
>
> I use OpenWrt 21.02 and miniupnpd 2.2, and the log shows error `Thu Mar 17 14:09:21 2022 daemon.notice miniupnpd[15353]: private/reserved address [...]

I can't find the solution. Is it closed now?

largeword, 2022-04-07:
> 

The [url](https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=4055380&extra=page%3D1&ordertype=1#:~:text=%E5%90%97%EF%BC%8C%E6%96%B0%E6%89%8B%E4%B8%8D%E5%A4%AA%E6%87%82-,%E6%88%91%E4%B8%8D%E7%9F%A5%E9%81%93%E4%BD%A0%E7%9A%84%E6%9E%B6%E6%9E%84(%E8%BF%99%E9%87%8C%E4%BB%A5x86%E4%B8%BA%E4%BE%8B),-%E8%BF%9B%E5%85%A5https%3A//archive) should point to a Chinese forum instead of a issue of github.

In short, you need to upload libip4tc.so.0 and libip6tc.so.0 to /usr/lib/, then replace the original miniupnpd in your system with version 2.1

largeword, 2022-04-07:
> I can't find the solution. Is it closed now?

The [url](https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=4055380&extra=page%3D1&ordertype=1#:~:text=%E5%90%97%EF%BC%8C%E6%96%B0%E6%89%8B%E4%B8%8D%E5%A4%AA%E6%87%82-,%E6%88%91%E4%B8%8D%E7%9F%A5%E9%81%93%E4%BD%A0%E7%9A%84%E6%9E%B6%E6%9E%84(%E8%BF%99%E9%87%8C%E4%BB%A5x86%E4%B8%BA%E4%BE%8B),-%E8%BF%9B%E5%85%A5https%3A//archive) should point to a Chinese forum instead of a issue of github.

In short, you need to upload libip4tc.so.0 and libip6tc.so.0 to /usr/lib/, then replace the original miniupnpd in your system with version 2.1

rmilecki, 2025-04-28:
Our downstream solution https://github.com/openwrt/packages/pull/16639 (`miniupnpd: add force_forwarding option support`) didn't fly but a similar option (`ignore_private_ip`) was added in the upstream by https://github.com/miniupnp/miniupnp/pull/813 and is present in the 2.3.9 release. It may help I think.

PR #14031 (2020-11-26): net/miniupnpd: ext_ip_reserved_ignore support
This add an option to ignore ext_ip reserved
Enable this option in config by default.

Fixes: #13773 
Patche is upstream pending https://github.com/miniupnp/miniupnp/pull/511
commit b76aa991 (2020-11-26): net/miniupnpd: ext_ip_reserved_ignore support

This add an option to ignore ext_ip reserved
Enable this option in config by default.

Fixes: #13773
commit 808a67b3 (2020-11-30): net/miniupnpd: ext_ip_reserved_ignore support

This add an option to ignore ext_ip reserved
Enable this option in config by default.

Fixes: #13773

(cherry picked from commit b76aa9919489f49b472a8f939f6d46ca33d05f64)