openwrt/packages #13778: unbound issues

doverride, 2020-10-26:
Maintainer: @EricLuehrsen
Environment: all, snapshot

Description:

In trying to configure unbound i identified two issues:

1. bad permissions on /tmp/lib/unbound cause hotplugcall to require `CAP_DAC_OVERRIDE` to create /tmp/lib/unbound/hotplug.time

```
20:03 < grift> aparcar[m]: https://github.com/openwrt/packages/blob/master/net/unbound/files/unbound.sh#L263 and line 264 are problematic because of this https://github.com/openwrt/packages/blob/master/net/unbound/files/unbound.ntpd#L21
20:04 < grift> i would argue that it should probably be unbound.root 0775
20:04 < grift> so that hotplugcall doesnt need cap_dac_override to create that time stamp in there
```
2. config host block is not processed by uci

the `config host` block in /etc/config/unbound is not being processed by uci.
I had to hack around this:
```
cat >/etc/unbound/unbound_ext.conf<<EOF
##############################################################################
# Extended user clauses added to the end of the UCI generated 'unbound.conf'
#
# Put your own forward:, view:, stub:, or remote-control: clauses here. This
# file is appended to the end of 'unbound.conf' with an include: statement.
# Notice that it is not part of the server: clause. Use 'unbound_srv.conf' to
# place custom option statements in the server: clause.
##############################################################################

forward-zone:
	name: "."
	forward-host: cloudlflare-dns.com
	forward-first: no
	forward-tls-upstream: yes
	forward-addr: 2606:4700:4700::1111@853
	forward-addr: 1.1.1.1@853
	forward-addr: 2606:4700:4700::1001@853
	forward-addr: 1.0.0.1@853
```
3. the package seems to add `/etc/unbound/unbound.conf` to the list of `sysupgrade` files. This seems to not make sense as that file is not used. It would instead, IMHO, make more sense to add `/etc/unbound/unbound_ext.conf` and `/etc/unbound/unbound_srv.conf` to the list of `sysupgrade` files

EricLuehrsen, 2020-10-27:
1) Where is this issue generated? It has been this way for years without any problems. Hotplug shell should be running as root (initd) and has permission to whatever.
2) Unbound doesn't have a `config host` block. That is a confounded DHCP/DNS configuration in `/etc/config/dhcp`. This tells odhcpd or dnsmasq to assign such MAC/DUID to a fixed address. In turn and if option enabled, Unbound will read the lease file(s) and enter them into DNS.
3) `/etc/unbound/unbound.conf` is used if you set the UCI to manual. The UCI expansion files should probably be added though.

doverride, 2020-10-27:
I believe the issue is generated here: 
https://github.com/openwrt/packages/blob/master/net/unbound/files/unbound.sh#L263
```
- chown -R unbound:unbound $UB_VARDIR
+chown -R unbound:root $UB_VARDIR
```
and here:
https://github.com/openwrt/packages/blob/master/net/unbound/files/unbound.sh#L264
```
-chmod 755 $UB_VARDIR
+chmod 775 $UB_VARDIR
```
This issue (and others) may have been here for a long time but SELinux recently exposed it. With SELinux one can block access to capabilities. If you grant root access to /tmp/lib/unbound then hotplug-call will not need CAP_DAC_OVERRIDE and we wont have to allow it that broad access. Thereby improving security quite a bit because CAP_DAC_OVERRIDE is a powerful capability

Maybe I am misunderstanding but i followed this: https://openwrt.org/docs/guide-user/services/dns/dot_unbound 

The above adds a config zone block to /etc/config/unbound and it does not end up in /tmp/lib/unbound

doverride, 2020-10-27:
>     2. Unbound doesn't have a `config host` block. 

Oops I meant `config zone`

>        _update:_ All the files are marked as configuration, so this issue is should be closed.

Strange, unless I was overlooking something, only /etc/unbound/unbound.conf got backed up. I had to add the other two manually to /etc/sysupgrade.conf

EricLuehrsen, 2020-10-28:
(1) Okay then. The status quo is that OpenWrt is not installed with SELinux. A change to SELinux causes this. It seems like OpenWrt core has growing pains including procd and hotplug are not complex like systemd. It would seem the directory permissions are correct. procd and hotplug should have a means to drop a spawn script from root to user (unbound).
(2) Try adding this configuration in `/etc/config/unbound`
```
config zone 'fwd_cloudflare'
	option fallback '0'
	option tls_index 'cloudflare-dns.com'
	option tls_upstream '1'
	option zone_type 'forward_zone'
	list server '1.1.1.1'
	list server '1.0.0.1'
	list zone_name '.'
	option enabled '1'
```
(3) I don't know what sysupgrade is doing. They are listed in the Makefile and that's all the control we have from a standard packaging function.

doverride, 2020-10-28:
1. I rest my case. Not sure what is so hard to understand about this

2. now it seems to work. Not sure why this works for me and why this did not work for me: https://openwrt.org/docs/guide-user/services/dns/dot_unbound

3. ok

closed 2020-10-28 by doverride

EricLuehrsen, 2020-10-30:
(1) It is clearly understood. SELinux is new and still being refined as to how it will work within resource constraints seen in OpenWrt. The right way is to have applications jailed or voluntarily self jail (Unbound method) to a working directory only writable by them. I would expect, long term, that SELinux, procd, and hotplug as a system work the right way. It just isn't there yet, and the Unbound package will be adapted when it is.