openwrt/packages #13781: <uwsgi>: Python plugin not updated?

MassiPi, 2020-10-26:
Maintainer: @Ansuel 
Environment: R7800 - OpenWrt SNAPSHOT r14776+73-8087499ac5

Description: i have uwsgi serving some python apps through the Python plugin.
It used to work, but recently (since 5 days) it does no longer work because it can't find the correct version.
It seems uwsgi is still looking for python 3.8 where the system is now on python 3.9

This is the error i get when the vassal is run
```
 !!! UNABLE to load uWSGI plugin: Error loading shared library libpython3.8.so.1.0: No such file or directory (needed by /usr/lib/uwsgi/python_plugin.so) !!!
```

My system has this libpython files
```
root@RUTTO:~# find / -name "libpython*"
/rom/usr/lib/libpython3.9.so
/rom/usr/lib/libpython3.9.so.1.0
/usr/lib/libpython3.9.so
/usr/lib/libpython3.9.so.1.0
```

am i missing something?
thanks

jefferyto, 2020-10-27:
Did you try reinstalling the uwsgi packages?

MassiPi, 2020-10-27:
i first noticed the issue with r14740
Because of the issue, i re-compiled the image yesterday evening, before posting, with r14776.
same issue
i don't think i can do anything more than re-compile the package, am i wrong?
thanks

jefferyto, 2020-10-27:
I'm not sure, from the error message it looks like the Python plugin was compiled in the environment with Python 3.8 (Python packages were updated to 3.9 a few days ago). I suggest doing a clean checkout and build.

MassiPi, 2020-10-27:
help :) is this something different than my simple 
```
git pull
./scripts/feeds update -a
./scripts/feeds install -a
make defconfig
make download
make
```

?
i can try with a make clean, but i'm not a github master :)
thanks

MassiPi, 2020-10-27:
well, i recompiled everything after a make clean and now it works again.
good catch jeffer!
thanks

closed 2020-10-27 by MassiPi