openwrt/packages #13794: docker-ce compile error with gcc5.5 and glibc

houaq, 2020-10-28:
Maintainer: @G.M0N3Y.2503
Environment: (put here arch, model, OpenWrt version)
OpenWrt 18.06.2 on ARM cortex-a7 (imx6ul)

Description:

My toolchain is gcc 5.5 with glibc, I got error when compiling docker-ce, the log output is:

```
touch /home/houaq/openwrt/trunk/build_dir/target-arm_cortex-a7+neon_glibc_eabi/docker-ce-19.03.13/.built_check
( export GOOS="linux" GOARCH="arm" GO386="" GOARM="7" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="arm-openwrt-linux-gnueabi-gcc" CXX="arm-openwrt-linux-gnueabi-g++" CGO_CFLAGS="-Os -pipe -mcpu=cortex-a7 -mfpu=neon -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/houaq/openwrt/trunk/build_dir/target-arm_cortex-a7+neon_glibc_eabi/docker-ce-19.03.13:docker-ce-19.03.13 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/houaq/openwrt/trunk/staging_dir/target-arm_cortex-a7+neon_glibc_eabi/usr/include -I/home/houaq/openwrt/trunk/staging_dir/target-arm_cortex-a7+neon_glibc_eabi/include -I/home/houaq/openwrt/trunk/staging_dir/toolchain-arm_cortex-a7+neon_gcc-5.5.0_glibc_eabi/usr/include -I/home/houaq/openwrt/trunk/staging_dir/toolchain-arm_cortex-a7+neon_gcc-5.5.0_glibc_eabi/include" CGO_CXXFLAGS="-Os -pipe -mcpu=cortex-a7 -mfpu=neon -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/houaq/openwrt/trunk/build_dir/target-arm_cortex-a7+neon_glibc_eabi/docker-ce-19.03.13:docker-ce-19.03.13 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/houaq/openwrt/trunk/staging_dir/target-arm_cortex-a7+neon_glibc_eabi/usr/lib -L/home/houaq/openwrt/trunk/staging_dir/target-arm_cortex-a7+neon_glibc_eabi/lib -L/home/houaq/openwrt/trunk/staging_dir/toolchain-arm_cortex-a7+neon_gcc-5.5.0_glibc_eabi/usr/lib -L/home/houaq/openwrt/trunk/staging_dir/toolchain-arm_cortex-a7+neon_gcc-5.5.0_glibc_eabi/lib -znow -zrelro -static -lpthread -lc -lgcc_eh" GOPATH="/home/houaq/openwrt/trunk/build_dir/target-arm_cortex-a7+neon_glibc_eabi/docker-ce-19.03.13/.go_work/build" GOCACHE="/home/houaq/openwrt/trunk/tmp/go-build" GOMODCACHE="/home/houaq/openwrt/trunk/dl/go-mod-cache" GOENV=off GITCOMMIT=4484c46d9d  DOCKER_GITCOMMIT=4484c46d9d  DOCKER_BUILDTAGS='' VERSION=19.03.13 && echo "Compiling CLI..." && cd /home/houaq/openwrt/trunk/build_dir/target-arm_cortex-a7+neon_glibc_eabi/docker-ce-19.03.13/cli && ./scripts/build/binary && echo "Compiling Engine..." && cd /home/houaq/openwrt/trunk/build_dir/target-arm_cortex-a7+neon_glibc_eabi/docker-ce-19.03.13/engine && ./hack/make.sh binary )
Compiling CLI...
Building statically linked build/docker-linux-arm-v7
Compiling Engine...

Removing bundles/

---> Making bundle: binary (in bundles/binary)
Building: bundles/binary-daemon/dockerd-19.03.13
GOOS="linux" GOARCH="arm" GOARM="7"
# github.com/docker/docker/cmd/dockerd
/home/houaq/openwrt/trunk/staging_dir/hostpkg/lib/go-cross/pkg/tool/linux_amd64/link: running arm-openwrt-linux-gnueabi-gcc failed: exit status 1
/opt/staging/toolchain-arm_cortex-a7+neon_gcc-5.5.0_glibc_eabi/bin/../lib/gcc/arm-openwrt-linux-gnueabi/5.5.0/../../../../arm-openwrt-linux-gnueabi/bin/ld: /home/houaq/openwrt/trunk/staging_dir/toolchain-arm_cortex-a7+neon_gcc-5.5.0_glibc_eabi/lib/libpthread.a(pthread_create.o): relocation R_ARM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
/home/houaq/openwrt/trunk/staging_dir/toolchain-arm_cortex-a7+neon_gcc-5.5.0_glibc_eabi/lib/libpthread.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status

Makefile:155: recipe for target '/home/houaq/openwrt/trunk/build_dir/target-arm_cortex-a7+neon_glibc_eabi/docker-ce-19.03.13/.built' failed
make[2]: *** [/home/houaq/openwrt/trunk/build_dir/target-arm_cortex-a7+neon_glibc_eabi/docker-ce-19.03.13/.built] Error 1
make[2]: Leaving directory '/home/houaq/openwrt/trunk/feeds/packages/utils/docker-ce'
time: package/feeds/packages/docker-ce/compile#6.84#1.62#8.14
package/Makefile:107: recipe for target 'package/feeds/packages/docker-ce/compile' failed
make[1]: *** [package/feeds/packages/docker-ce/compile] Error 2
make[1]: Leaving directory '/home/houaq/openwrt/trunk'
/home/houaq/openwrt/trunk/include/toplevel.mk:216: recipe for target 'package/feeds/packages/docker-ce/compile' failed
make: *** [package/feeds/packages/docker-ce/compile] Error 2

```

neheb, 2020-10-28:
I was going to say duplicate of https://github.com/openwrt/packages/issues/12353

But this looks like a PIC problem.

houaq, 2020-10-29:
> I was going to say duplicate of #12353
> 
> But this looks like a PIC problem.

Yes, I have tried that solution, the tini works, but docker-ce not.
Now, I changed to use dynamic link, by change the bundle to dynbinary in docker-ce Makefile.

closed 2025-11-14 by GeorgeSapkin