openwrt/packages #13803: libftdi1: currently not installable on 19.07.3?

spacelama, 2020-10-29:
Maintainer: @neheb ?
Environment: 
model=GL-B1300 / Convexa-B - GL.iNet
OPENWRT_BOARD="ipq40xx/generic"
OPENWRT_ARCH="arm_cortex-a7_neon-vfpv4"
VERSION_ID="19.07.3"

Description:

libftdi1 is not able to be installed on my box right now.  I notice libftdi1 was updated just a day ago.  Tried refreshing over several hours to see if the repository was still being synced, but still no go:

```
root@b1300:~# opkg update ; opkg install crelay
...
Signature check passed.
Installing crelay (0.13-1) to root...
Downloading http://downloads.openwrt.org/releases/19.07.3/packages/arm_cortex-a7_neon-vfpv4/packages/crelay_0.13-1_arm_cortex-a7_neon-vfpv4.ipk
Installing libftdi1 (1.4-3) to root...
Downloading http://downloads.openwrt.org/releases/19.07.3/packages/arm_cortex-a7_neon-vfpv4/packages/libftdi1_1.4-3_arm_cortex-a7_neon-vfpv4.ipk
Collected errors:
 * opkg_install_pkg: Package size mismatch: libftdi1 is 21247 bytes, expecting 21260 bytes
 * opkg_install_cmd: Cannot install package crelay.
```

neheb, 2020-10-29:
Sounds weird. I just pushed a bunch of backported updates. Wait a day and see if your problem is fixed.

You could also try installing with one of opkg's --force options.

spacelama, 2020-10-29:
Confirmed, went with:

```opkg --force-checksum install crelay```

and that worked.  I might try uninstall and reinstall fresh tomorrow.  Neat package BTW - I'm not used to software just working the moment you install it.

neheb, 2020-10-31:
I assume this can be closed?

spacelama, 2020-10-31:
Yep, reinstalling libftdi1 today picked up 1.4-6 instead of 1.4-3

closed 2020-10-31 by spacelama