openwrt/packages #13808: can't compile transmission-daemon, missing dependencies

bobafetthotmail, 2020-10-29:
Maintainer: @neheb 
Environment: x86_64, snapshot

Description:

can't seem to compile transmission-daemon, it complains about missing libcrypto.so.1.1 and libssl.so.1.1 no matter what I choose as other packages.

here the part of the log where it's actually compiling it and throws the error

```
make[3]: Entering directory '/home/alby/autocompile_owrt_images/openwrt/feeds/packages/net/transmission'
mkdir -p /home/alby/autocompile_owrt_images/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/alby/autocompile_owrt_images/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/.pkgdir/transmission-daemon/. /home/alby/autocompile_owrt_images/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/alby/autocompile_owrt_images/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.transmission-daemon_installed
mkdir -p /home/alby/source_code/openwrt_compiled_images/binary_folder/x86-vm-transmission/targets/x86/64/packages /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/CONTROL /home/alby/autocompile_owrt_images/openwrt/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/usr/bin
install -m0755 /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-install/usr/bin/transmission-daemon /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/usr/bin/
install -d -m0755 /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/etc/init.d/
install -m0755 files/transmission.init /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/etc/init.d/transmission
install -d -m0755 /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/etc/config
install -m0600 files/transmission.config /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/etc/config/transmission
install -d -m0755 /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/etc/sysctl.d/
install -m0600 files/transmission.sysctl /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/etc/sysctl.d/20-transmission.conf
install -d -m0755 /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/etc/seccomp
install -m0644 ./files/transmission-daemon.json /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon/etc/seccomp/
find /home/alby/autocompile_owrt_images/openwrt/build_dir/target-x86_64_musl/transmission-3.00/ipkg-x86_64/transmission-daemon -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package transmission-daemon is missing dependencies for the following libraries:
libcrypto.so.1.1
libssl.so.1.1
make[3]: *** [Makefile:157: /home/alby/source_code/openwrt_compiled_images/binary_folder/x86-vm-transmission/packages/x86_64/packages/transmission-daemon_3.00-7_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/alby/autocompile_owrt_images/openwrt/feeds/packages/net/transmission'
time: package/feeds/packages/transmission/compile#0.19#0.08#0.24
make[2]: *** [package/Makefile:114: package/feeds/packages/transmission/compile] Error 2
make[2]: Leaving directory '/home/alby/autocompile_owrt_images/openwrt'
make[1]: *** [package/Makefile:108: /home/alby/autocompile_owrt_images/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/alby/autocompile_owrt_images/openwrt'
make: *** [/home/alby/autocompile_owrt_images/openwrt/include/toplevel.mk:236: world] Error 2

```

neheb, 2020-10-29:
I would run make clean and make menuconfig.

I tested this locally and see no problems.

bobafetthotmail, 2020-10-30:
hmm, I can't reproduce on a new git pull of the repo either, there is something wrong on my side, never mind

closed 2020-10-30 by bobafetthotmail