openwrt/packages #13810: Unbound: Should be build with ipset support

diabl0w, 2020-10-30:
Maintainer: @EricLuehrsen
Environment: WRT3200acm Openwrt 19.07.4

Description:
According to the Makefile, unbound-daemon-heavy version 1.11.0-2 should have ipset set support, but:

> Package unbound-daemon-heavy (1.11.0-2) installed in root is up to date
> unbound -v
> ...
> Linked modules: dns64 respip validator iterator

EricLuehrsen, 2020-10-30:
ipset with libmnl support was added to master but not stable 19.07. Master also removed the light/heavy prebuilt packages due to trickiness with opkg. Daemon and library need to match but often they wouldn't. On master, extra options or tiny options are check boxes for build-your-self similar to so many other packages.

diabl0w, 2020-10-30:
> ipset with libmnl support was added to master but not stable 19.07. Master also removed the light/heavy prebuilt packages due to trickiness with opkg. Daemon and library need to match but often they [...]

WHOOPS! i was browsing 19.07 branch when i searched for "unbound ipset" and it brought me to the committ https://github.com/openwrt/packages/commit/839c2e3caef684d71120f8ff7e7c7fb1e6ed1e40
but i forgot github redirects back to master when you use the search function.

thanks anyway.... FWIW I vote to keeping ipset included by default in future versions. Thanks!

closed 2020-10-30 by diabl0w

diabl0w, 2020-10-30:
@EricLuehrsen sorry to bother you, quick question if you don't mind... I have the new binary up and running on my device, but I need it's init.d script to start the unbound service as root user instead of as unbound user... do you know how to do this? I looked through the init.d script but can't find anything useful

EricLuehrsen, 2020-10-30:
Unbound drops user and changes its root directory view on its own, voluntary jail. The options that control that are written by `/usr/lib/unbound/unbound.sh` Like many packages in many distros, this is not a user option and we see more applications running as user by application name for security. The need to run as `root` is compelling reason to **not** compile ipset by default.

diabl0w, 2020-10-30:
> Unbound drops user and changes its root directory view on its own, voluntary jail. The options that control that are written by `/usr/lib/unbound/unbound.sh` Like many packages in many distros, this [...]

interesting, I wonder how dnsmasq manages to use ipset without root

EricLuehrsen, 2020-11-02:
I intend to stick to the current direction of best practices. These are the standard three: chroot or an external launcher jail, auto-config into /var/lib/[app] or /var/run/[app], and restrictive application user name. These are seen in most distributions now. These are not options for convenient user front ends. If you want to get crazy with the configuration, then you can use the manual override to UCI.

diabl0w, 2020-11-02:
thanks for the update. I wasn't requesting to go away from standards, but
rather to just make it possible for users to override them which it looks
like maybe the UCI override might do. Either way, I have resorted just
using the ipset-dns package and assigning it as a dns server for unbound to
forward certain requests to. ipset-dns then properly sets the ipset rules.
I'm not sure why exactly it is able to set ipset rules while unbound cannot
without root, but it works... albeit with some extra overhead.

EricLuehrsen, 2020-11-02:
see also #12084