openwrt/packages #13811: openssh-client (7.7p1-1) missing libcrypto.so.1.0.0

modemmike, 2020-10-30:
Environment: OpenWrt 18.06.1 r7258-5eb055306f / LuCI openwrt-18.06 branch (git-18.196.56128-9112198)
GL.iNet GL-AR300M

Description:

Seems the reference to crypto is wrong:

Error loading shared library libcrypto.so.1.0.0: No such file or directory (needed by /usr/bin/ssh)

I see: /usr/lib/libcrypto.so.1.1 is installed, maybe I need a different image?

Continuing with the errors:

Error relocating /usr/bin/ssh: X509_new: symbol not found
Error relocating /usr/bin/ssh: EVP_MD_block_size: symbol not found
Error relocating /usr/bin/ssh: DSA_do_sign: symbol not found
Error relocating /usr/bin/ssh: EC_KEY_new_by_curve_name: symbol not found
Error relocating /usr/bin/ssh: BIO_s_mem: symbol not found
Error relocating /usr/bin/ssh: EVP_MD_CTX_cleanup: symbol not found
Error relocating /usr/bin/ssh: ECDH_compute_key: symbol not found
Error relocating /usr/bin/ssh: EC_POINT_new: symbol not found
Error relocating /usr/bin/ssh: EC_POINT_free: symbol not found
Error relocating /usr/bin/ssh: BN_clear_free: symbol not found
Error relocating /usr/bin/ssh: BIO_write: symbol not found
Error relocating /usr/bin/ssh: EVP_CIPHER_CTX_key_length: symbol not found
Error relocating /usr/bin/ssh: BN_div: symbol not found
Error relocating /usr/bin/ssh: EC_GROUP_cmp: symbol not found
Error relocating /usr/bin/ssh: EC_KEY_set_group: symbol not found
Error relocating /usr/bin/ssh: ENGINE_load_builtin_engines: symbol not found
Error relocating /usr/bin/ssh: PEM_read_bio_PrivateKey: symbol not found
Error relocating /usr/bin/ssh: ERR_load_crypto_strings: symbol not found
Error relocating /usr/bin/ssh: EC_GROUP_get_degree: symbol not found
Error relocating /usr/bin/ssh: RSA_sign: symbol not found
Error relocating /usr/bin/ssh: ERR_get_error: symbol not found
Error relocating /usr/bin/ssh: EVP_CIPHER_CTX_set_key_length: symbol not found
Error relocating /usr/bin/ssh: PEM_write_bio_ECPrivateKey: symbol not found
Error relocating /usr/bin/ssh: EVP_MD_CTX_md: symbol not found
Error relocating /usr/bin/ssh: SSLeay_version: symbol not found
Error relocating /usr/bin/ssh: EC_POINT_is_at_infinity: symbol not found
Error relocating /usr/bin/ssh: d2i_X509: symbol not found
Error relocating /usr/bin/ssh: DSA_generate_key: symbol not found
Error relocating /usr/bin/ssh: BN_cmp: symbol not found
Error relocating /usr/bin/ssh: X509_free: symbol not found
Error relocating /usr/bin/ssh: EC_KEY_free: symbol not found
Error relocating /usr/bin/ssh: EVP_CIPHER_CTX_new: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_128_cbc: symbol not found
Error relocating /usr/bin/ssh: DH_size: symbol not found
Error relocating /usr/bin/ssh: EC_GROUP_get_curve_name: symbol not found
Error relocating /usr/bin/ssh: EVP_CIPHER_CTX_ctrl: symbol not found
Error relocating /usr/bin/ssh: PEM_write_bio_RSAPrivateKey: symbol not found
Error relocating /usr/bin/ssh: EC_POINT_point2oct: symbol not found
Error relocating /usr/bin/ssh: RSA_new: symbol not found
Error relocating /usr/bin/ssh: BN_bn2bin: symbol not found
Error relocating /usr/bin/ssh: EVP_Digest: symbol not found
Error relocating /usr/bin/ssh: AES_encrypt: symbol not found
Error relocating /usr/bin/ssh: DH_generate_key: symbol not found
Error relocating /usr/bin/ssh: DH_compute_key: symbol not found
Error relocating /usr/bin/ssh: PEM_write_bio_DSAPrivateKey: symbol not found
Error relocating /usr/bin/ssh: EVP_MD_CTX_init: symbol not found
Error relocating /usr/bin/ssh: EC_METHOD_get_field_type: symbol not found
Error relocating /usr/bin/ssh: EC_GROUP_method_of: symbol not found
Error relocating /usr/bin/ssh: BIO_free: symbol not found
Error relocating /usr/bin/ssh: BIO_ctrl: symbol not found
Error relocating /usr/bin/ssh: BIO_new: symbol not found
Error relocating /usr/bin/ssh: DSA_generate_parameters_ex: symbol not found
Error relocating /usr/bin/ssh: EC_KEY_set_asn1_flag: symbol not found
Error relocating /usr/bin/ssh: EVP_PKEY_get1_RSA: symbol not found
Error relocating /usr/bin/ssh: RSA_public_decrypt: symbol not found
Error relocating /usr/bin/ssh: EC_KEY_get0_group: symbol not found
Error relocating /usr/bin/ssh: EVP_DigestUpdate: symbol not found
[... 75 lines cut ...]
Error relocating /usr/bin/ssh: EVP_sha512: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_192_cbc: symbol not found
Error relocating /usr/bin/ssh: EVP_sha1: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_192_ctr: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_128_ctr: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_256_ctr: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_256_gcm: symbol not found
Error relocating /usr/bin/ssh: EVP_des_ede3_cbc: symbol not found
Error relocating /usr/bin/ssh: EVP_sha384: symbol not found
Error relocating /usr/bin/ssh: EVP_sha256: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_128_cbc: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_128_gcm: symbol not found
Error relocating /usr/bin/ssh: EVP_md5: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_256_cbc: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_256_cbc: symbol not found
Error relocating /usr/bin/ssh: EVP_sha512: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_192_cbc: symbol not found
Error relocating /usr/bin/ssh: EVP_sha1: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_192_ctr: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_128_ctr: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_256_ctr: symbol not found
Error relocating /usr/bin/ssh: EVP_aes_256_gcm: symbol not found
Error relocating /usr/bin/ssh: EVP_des_ede3_cbc: symbol not found
Error relocating /usr/bin/ssh: EVP_sha384: symbol not found
Error relocating /usr/bin/ssh: EVP_sha256: symbol not found

neheb, 2020-10-31:
18.06 uses OpenSSL 1.0.2. Why do you have 1.1 installed?

modemmike, 2020-10-31:
@neheb thanks for the response, the answer is, I dunno, thats what installs when I install libopenssl as referenced here: https://openwrt.org/packages/pkgdata_owrt18_6/openssh-client  opkg apparently does not have method for specifying a version and I'm def a newb with Linux. I even tried symlinking the old to the new but that resulted in more errors.

modemmike, 2020-10-31:
Package libopenssl (1.1.1d-2) installed in root is up to date.

neheb, 2020-10-31:
That's...not right. 1.1.1 is not supposed to be installed in 18.06. I assume you're using custom feeds?

modemmike, 2020-11-02:
Nope, OEM firmware from https://dl.gl-inet.com/firmware/ar300m/ and I haven't changed anything, this is a fresh install. I will try some other images.

neheb, 2020-11-02:
And there we go. This issue should be taken with them.

closed 2020-11-03 by BKPepe