openwrt/packages #13820: banip: dependency on logd breaks syslog-ng installation

boecko, 2020-11-01:
Hi,

Building an image fails, when syslog-ng and banip is selected, because banip depends on logd.

logd and syslog-ng both want to provide /sbin/logread

### Solution
depend on logd OR syslog-ng

dibdot, 2020-11-01:
If you've tested banIP with both logger variants feel free to open a PR.

boecko, 2020-11-02:
I did. logread from syslog-ng is compatible.
Should i remove the dependency ```logd```?
Is there way to make an 'OR' dependency on that in the Makefile?

dibdot, 2020-11-03:
Thanks for testing & confirmation. Currently I'm working on a bigger update. If you don't mind I'll remove the dependency with this update, too. Of course, this will take some time ... if it's urgent for you, open a PR with the dependency removal.

ja-pa, 2020-11-03:
> I did. logread from syslog-ng is compatible.
> Should i remove the dependency `logd`?
> Is there way to make an 'OR' dependency on that in the Makefile?

Offtopic: It is possible to use OR, etc see **Dependency Types** section https://openwrt.org/docs/guide-developer/packages

boecko, 2020-11-03:
> Offtopic: It is possible to use OR, etc see **Dependency Types** section https://openwrt.org/docs/guide-developer/packages

I tried different lines .. all but one ended in a syntax error 
This one managee to compile 
``` @(PACKAGE_logd||PACKAGE_syslog-ng)```
But it seems wrong for me.

Something like dpkg Provides would solve that in a more elegant way.
```Provides: logread```

dibdot, 2020-11-03:
@boecko I've removed the logd dependency. Thanks for bringing this up.

PR #13880 (2020-11-11): banip: removed logd check since logd dep has been dropped
Maintainer: Dirk Brenken <dev@brenken.org>
Compile tested: x86_64 master
Run tested: x86_64 master

Description:
#13848 dropped the logd dependency in commit 14a69715899accff66009d45a4a13ab7cb5a38b7 trying to fix #13820 

I say trying because **banip.sh** checks whether **/etc/init.d/log** is running at runtime, failing when syslog-ng is used.
closed 2020-11-03 by dibdot