openwrt/packages #13823: MWAN3: can't determine gateway of openvpn tap interface

kghost, 2020-11-01:
OpenWrt 19.07.4, r11208-ce6496d796

interface information is empty for openvpn tap device

```
root@OpenWrt:~# ubus call network.interface dump

                {
                        "interface": "vpn",
                        "up": true,
                        "pending": false,
                        "available": true,
                        "autostart": true,
                        "dynamic": false,
                        "uptime": 503,
                        "l3_device": "tap0",
                        "proto": "none",
                        "device": "tap0",
                        "metric": 0,
                        "dns_metric": 0,
                        "delegation": true,
                        "ipv4-address": [

                        ],
                        "ipv6-address": [

                        ],
                        "ipv6-prefix": [

                        ],
                        "ipv6-prefix-assignment": [

                        ],
                        "route": [

                        ],
                        "dns-server": [

                        ],
                        "dns-search": [

                        ],
                        "neighbors": [

                        ],
                        "inactive": {
                                "ipv4-address": [

                                ],
                                "ipv6-address": [

                                ],
                                "route": [

                                ],
                                "dns-server": [

                                ],
                                "dns-search": [

                                ],
                                "neighbors": [

                                ]
                        },
                        "data": {

                        }
                },
```

Here is my openvpn configure to set the gateway:

```
route-nopull
route 0.0.0.0 0.0.0.0 vpn_gateway 30
```

It maybe hard to get the gateway of the tap device, is there any alternative way to manually set or override the information for mwan3 ?

aaronjg, 2020-11-01:
The issue is that openvpn does not use the `netifd` subsystem, so the gateway and IP address are not available in ubus. 

Can you try the latest version of mwan3 from snapshot? I recently changed the route tables to get all of the routes from the main routing table, rather than trying to figure out the default route from the netifd data.

The latest version is not available in 19.07, so please install it manually for your device from here:
https://downloads.openwrt.org/snapshots/packages/

ptpt52, 2020-11-02:
This should not work with openvpn, mwan3 rely on the hotplug iface event, where depends on the netifd.

maybe we could change, let the iface event generates from mwan3rtmon, where we could deteminate the events from the route table changes

kghost, 2020-11-02:
Is there any workaround to let me manually set the gateway ? Either through netif or mwan3

aaronjg, 2020-11-02:
> This should not work with openvpn, mwan3 rely on the hotplug iface event, where depends on the netifd.

Mwan3 will start the tracker when it starts. It may not work as smoothly as it does with a supported netifd protocol, but I have used mwan3 with openvpn, and it works well enough.

> Is there any workaround to let me manually set the gateway ? Either through netif or mwan3

Please try with the snapshot version, and report any issues.

kghost, 2020-11-07:
Tried snapshot version `mwan3 - 2.10.1-1`, works like a charm.

kghost, 2020-11-08:
The snapshot version report all my interfaces are offline

```
config interface 'wan'              
        option enabled '1'          
        option initial_state 'online'
        option family 'ipv4'  
```

```
root@OpenWrt:~# mwan3 interfaces
Interface status:
 interface wan is offline 00h:00m:00s, uptime 00h:17m:15s and tracking is not enabled
```

But the `wan` connection is active

kghost, 2020-11-08:
It looks like there are problems with tracking disabled. I manually deleted the while loop in `mwan3track` then it works

aaronjg, 2020-11-08:
Yes, there was a bug in mwan3track. Could you please try these commits rather than deleting the while loop?

https://github.com/openwrt/packages/commit/94a25dbd92adcf070e10a508f5103d6914449f2a
https://github.com/openwrt/packages/commit/81e6a8fd3be57cba3b5bcbeb40fbbef52779f70b

PR #13853 (2020-11-08): mwan3: code cleanup and fixes
Maintainer: me
Compile tested: x86_64, APU3, OpenWrt master
Run tested: x86_64, APU3, OpenWrt master

Description:
This is a maintenance P/R with the following goals.
- Clean up the source code
- Fix pending errors through community tests which are found through different test scenarios

Fixes #13823 
Fixes #13865
closed 2020-11-08 by feckert