openwrt/packages #13835: [chrony - RFC] NTS & RTC support not compiled in?

n8v8R, 2020-11-02:
Maintainer: @mlichvar 
Environment: chrony_4.0-2_arm_cortex-a9_vfpv3-d16.ipk

Description:

```
chronyd version 4.0 starting (+CMDMON +NTP +REFCLOCK -RTC +PRIVDROP -SCFILTER -SIGND +ASYNCDNS -NTS -SECHASH +IPV6 -DEBUG)
```

Does `-RTC` and `-NTS` mean that the support for either has not been compiled in? If so what would be the reasoning for removing those from that target (mvebu)?

ja-pa, 2020-11-02:
Default should be without NTS https://github.com/openwrt/packages/blob/master/net/chrony/Makefile#L44

n8v8R, 2020-11-02:
What is the point of removing NTS feature from Chrony? And then adding options for NTS in this commit https://github.com/openwrt/packages/commit/21c0f580f1c3701a96cf0621f3ec431fa57e034d?

And it does not clarify why RTS is being removed as well, and then again adding options in commit https://github.com/openwrt/packages/commit/21c0f580f1c3701a96cf0621f3ec431fa57e034d?

mlichvar, 2020-11-03:
It's about storage space. Enabling NTS in chrony pulls in as dependencies some large packages: libgnutls, libnettle, libgmp, ca-bundle. Together, they are about 20x larger than chrony itself. I have chrony installed on some routers that have only 4MB flash. With NTS, the minimum is 8MB flash.

Is it common for packages to have different defaults on different targets? Maybe there could be a separate chrony-nts package, like tcpdump vs tcpdump-mini, for instance?

n8v8R, 2020-11-03:
The space/mem computation for severely restricted hardware is unfortunately a bit of an Achilles' heel of OpenWrt and this package now being another example of better self-compiling it.

Since the distro provides alternative ntp server/client userland - one that suits such severely restricted hardware -  I would have expected that owners of such are hardware content with what their hardware is capable of and not penalising owners of more advanced hardware instead. 

On the other hand recent efforts are made in the recent package commits aiming at NTS and RTC and even stating in the commit message

>should be better than no NTS at all

If I have to self-compile the package in the first place I would do so with the development source and not pulling it from OpenWrt.

closed 2020-11-03 by n8v8R