openwrt/packages #13843: service dockerd start fail!

peterwillcn, 2020-11-03:
OpenWrt SNAPSHOT r14851-08d90a75f91 / LuCI Master git-20.305.72436-6e4e132
                                                 
root@OpenWrt:~# service dockerd start
Syntax: /etc/init.d/dockerd [command]

Available commands:
	uciadd  Add default bridge configuration to network and firewall uci config
	ucidel  Delete default bridge configuration from network and firewall uci config	running         Check if service is running
	status          Service status
	trace           Start with syscall trace

feckert, 2020-11-03:
This has to do with the following change https://github.com/openwrt/openwrt/commit/1a69f50dc627f6fc5860f871edec357cc0a187e6 and will get fixed with this P/R
https://github.com/openwrt/packages/pull/13798

Thanks for reporting

feckert, 2020-11-04:
Pullrequest https://github.com/openwrt/packages/pull/13798 was merged.
So this could be closed.

closed 2020-11-04 by feckert