openwrt/packages #13846: mwan3: active user rules missing after reboot router

ptpt52, 2020-11-03:
Latest master code in mwan3

add rules:
```
config rule 'test_rule'
	option dest_ip 'www.google.com'
	option proto 'all'
	option sticky '0'
	option use_policy 'balanced'
```
mwan3 status:
```
Interface status:
 interface wan is online 00h:10m:48s, uptime 00h:11m:07s and tracking is active
 interface xwan01 is online 00h:10m:41s, uptime 00h:11m:06s and tracking is active
 interface xwan02 is online 00h:10m:35s, uptime 00h:11m:07s and tracking is active
 interface wg is online 00h:10m:56s, uptime 00h:11m:09s and tracking is active
 interface owan is offline and tracking is down

Current ipv4 policies:
balanced:
 wg (25%)
 xwan02 (25%)
 xwan01 (25%)
 wan (25%)

Current ipv6 policies:
balanced:
 unreachable

Directly connected ipv4 networks:
10.8.11.0/24
192.168.18.146
224.0.0.0/3
192.168.18.1
192.168.16.1
192.168.18.147
192.168.17.0/24
169.254.0.0/16
192.168.15.0/24
127.0.0.0/8

Directly connected ipv6 networks:
fd8a:9567:893::/64
fe80::/64

Active ipv4 user rules:
## nothing here
Active ipv6 user rules:

```

This failed because the iptables rule setting with domain name resolution failed during boot
PS: during boot, network not ready, dns not work. so it fail.

give an example
```
iptables -t mangle -I PREROUTING -d www.google.com -j DROP
iptables v1.8.4 (legacy): host/network `www.google.com' not found
Try `iptables -h' or 'iptables --help' for more information.
```

ptpt52, 2020-11-03:
Also I found that the `ip rule` missing two lines after reboot
I have to do mwan3 restart to restore them.
```
2061:	from all fwmark 0x3d00/0x3f00 blackhole
2062:	from all fwmark 0x3e00/0x3f00 unreachable
```

feckert, 2020-11-03:
> This failed because the iptables rule setting with domain name resolution failed during boot
PS: during boot, network not ready, dns not work. so it fail.

As far as I know It has always been like this. So no regression since last change

ptpt52, 2020-11-03:
> As far as I know It has always been like this. So no regression since last change.

it should work before this commit https://github.com/openwrt/packages/commit/c07f5230be128669f7b6731415de26f8176fbf5b#diff-1501c6d736d400a33d21d08234c123d98c65a8fd0e6e785b28e45a2268aff725

I notice that after this commit hotplug missing reload some rules,
if we reload the rules or something, the iptables rules should work, in that case network is ready, dns works.

ptpt52, 2020-11-04:
@feckert 
hi, I notice the iptables rules
```
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j CONNMARK --restore-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_ifaces_in
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_connected
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_rules
-A mwan3_hook -j CONNMARK --save-mark --nfmask 0x3f00 --ctmask 0x3f00
-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected

```

why 2 lines match -j mwan3_connected in mwan3_hook chain?
```
-A mwan3_hook -m mark --mark 0x0/0x3f00 -j mwan3_connected
-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected
```

it seems `-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected` is unepected?

feckert, 2020-11-04:
> it seems `-A mwan3_hook -m mark ! --mark 0x3f00/0x3f00 -j mwan3_connected` is unepected?

That has always been the case.
I have looked at the history when this was included, and I can say that it was already there until version 2.0.0.
This is from the original developer Jeroen Louwes.
I haven't changed anything about that.

If then I have to take a closer look at it.
But as I understand it.
If the default mark is not match, then look again in the connected table to see if you can find something there.

aaronjg, 2020-11-05:
> config rule 'test_rule'
> 	option dest_ip 'www.google.com'
> 	option proto 'all'
> 	option sticky '0'
> 	option use_policy 'balanced'

This is a suboptimal way to do accomplish rules based on domain name. In addition to the issue you raised there are two other problems:
1) Iptables attempts to resolve the domain to the ip address, and then adds a rule for each IP. This could result in many rules for domains with multiple A/AAAA records, and having more rules will slow down the processing of iptables.
2) The DNS resolution happens at the time the route is added, and it will not always resolve to the same IP address. For example DNS based load balancing, or DNS targeting based on source IP.

This is why mwan3 supports ipsets in the rules. You can add an ipset for www.google.com, and then add to it when domains are resolved by adding a line like so to your `/etc/config/dhcp`

```
config dnsmasq
  ...
  list ipset '/www.google.com/vpnbypass_6,vpnbypass_4'
```

and in your `/etc/config/firewall`
```
config  ipset
        option  name            'vpnbypass_4'
        option  match           'src_net'
        option  storage         'hash'
        option  enabled         '1'

config  ipset
        option  name            'vpnbypass_6'
        option  match           'src_net'
        option  family          'ipv6'
        option  storage         'hash'
        option  enabled         '1'

config  ipset
        option  name            'vpnbypass'
        option  match           'set'
        option  storage         'list'
        option  enabled         '1'
        list    entry           'vpnbypass_4'
        list    entry           'vpnbypass_6'
```

and change your mwan3 config to:
```
 config rule 'test_rule'
 	option ipset 'vpnbypass'
 	option proto 'all'
 	option sticky '0'
 	option use_policy 'balanced'
```

aaronjg, 2020-11-05:
> Also I found that the `ip rule` missing two lines after reboot
> I have to do mwan3 restart to restore them.

This is a result of some of the speed improvements where I moved the common set up out of the hotplug scripts. It looks like when network restarts, it flushes the ip rules and reloads them. 

I have a P/R (#13857) to fix this here to add that bit of setup back to the hotplug scripts. It might be a bit cleaner to handle this on an interface reload trigger with procd, but this should be fine for now.