openwrt/packages #13855: CI:  arm_cortex-a9_vfpv3-d16 fakeroot segfaults in SDK

aparcar, 2020-11-04:
The CI uses SDK snapshots to build changed packages for multiple architectures. It currently fails for `arm_cortex-a9_vfpv3-d16`
 with the following error message:

```
/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/build/openwrt/bin/targets/mvebu/cortexa9/packages
/home/build/openwrt/staging_dir/host/bin/fakeroot /home/build/openwrt/scripts/ipkg-build -m "" /home/build/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/toolchain/ipkg-arm_cortex-a9_vfpv3-d16/libgcc /home/build/openwrt/bin/targets/mvebu/cortexa9/packages
/home/build/openwrt/staging_dir/host/bin/fakeroot: line 185:  8682 Segmentation fault      (core dumped) FAKEROOTKEY=$FAKEROOTKEY LD_LIBRARY_PATH="$PATHS" LD_PRELOAD="$LIB" "$@"
make[2]: Leaving directory '/home/build/openwrt/package/toolchain'
make[2]: *** [Makefile:764: /home/build/openwrt/bin/targets/mvebu/cortexa9/packages/libgcc1_8.4.0-2_arm_cortex-a9_vfpv3-d16.ipk] Error 139
make[1]: *** [package/Makefile:114: package/toolchain/compile] Error 2
```

It used to work for a while but now crashes. As it only fails for the single architecture I'm not sure if it's an issue with the two recent `fakeroot` commits over at `openwrt.org`:

https://github.com/openwrt/openwrt/commit/0fdbd2e61f742d9fbd6263bd86da3013ebcb1b9c
https://github.com/openwrt/openwrt/commit/f93cb5c2c8dcffae6f7bc80b9c7807de30d67371

EricLuehrsen, 2020-11-05:
cc: @jow- @dangowrt

neheb, 2020-11-06:
Does this happen with only the SDK?

danvd, 2020-11-07:
Minor addition: Upon submitting any PR the test "Test Build / Test arm_cortex-a9_vfpv3-d16 (pull_request)" is also failing due to this segfault.
```
/home/build/openwrt/staging_dir/host/bin/fakeroot: line 185: 31187 Segmentation fault      (core dumped) FAKEROOTKEY=$FAKEROOTKEY LD_LIBRARY_PATH="$PATHS" LD_PRELOAD="$LIB" "$@"
```

aparcar, 2020-11-08:
I couldn't reproduce the issue locally...

aparcar, 2020-11-09:
Looking at the docker CI builds, it's failing since the October 30. Which happens to be the same date as the above mentioned patches were added. In the Docker CI the x86/64 target fails, so it is not a problem of mvebu/cortex-a9 but all targets.

aparcar, 2020-11-09:
Failing Docker CI: https://gitlab.com/openwrt/docker/-/pipelines
Failing Docker CI test: https://gitlab.com/openwrt/docker/-/jobs/837338425

jefferyto, 2020-11-09:
This is not just a CI issue - I am seeing this error locally with the armvirt-32, armvirt-64, malta-be, x86-generic and x86-64 SDKs (8 Nov) on Ubuntu 20.10.

I can't update any packages without a working SDK :slightly_frowning_face:

jefferyto, 2020-11-09:
Passing the `-l $(FAKEROOT_SO)` option with the absolute path to libfakeroot.so (removed in openwrt/openwrt@0fdbd2e61f742d9fbd6263bd86da3013ebcb1b9c) works around the segfault.

It appears it isn't finding the right shared library for some reason.

dangowrt, 2020-11-09:
@jow- Any idea why this could be happening? It's quite weird because it seems to depend on _which_ of the buildhosts made the SDK for it to have that bug or not...

ynezz, 2020-11-10:
Should be fixed via https://git.openwrt.org/d4521fb132266fd6c94796e19d3d1e925847338d

closed 2020-11-10 by ynezz