openwrt/packages #13860: Installing OpenSSL headers

noplanman, 2020-11-05:
Environment: Turris Omnia, armv7l, Turris OS 5.1.2 (based on OpenWrt 19.07.4)

Description:
I'm trying to install borgbackup on my Turris Omnia router, but it fails with:
```
Exception: Unable to find OpenSSL >= 1.0 headers.
```

Does anybody know where I can find and how to install these?

BKPepe, 2020-11-06:
Hello @noplanman,

I am not able to find borgbackup in OpenWrt feeds neither in Turris feeds (which are using OpenWrt feeds most of the time as it is). Could you please tell us if you are using any other custom feed or are you trying to package borgbackup to OpenWrt?

noplanman, 2020-11-06:
I initially tried packaging but got lost with how it works.

What I was looking at is simply installing borgbackup directly with `pip` on the Omnia, which requires the SSL headers. I was hoping there is some OpenWrt package to provide them.

Regarding the packaging, I've given it another try but am a bit overwhelmed.

I've used the `youtube-dl` Makefile as a template and adjusted it for borgbackup, this is what I have:
```
#
# Copyright (C) 2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=borgbackup
PKG_VERSION:=1.1.14
PKG_RELEASE:=1

PYPI_NAME:=borgbackup
PKG_HASH:=7dbb0747cc948673f695cd6de284af215f810fed2eb2a615ef26ddc7c691edba

PKG_MAINTAINER:=Me <me@me.me>
PKG_LICENSE:=Unlicense
PKG_LICENSE_FILES:=LICENSE

include ../../lang/python/pypi.mk
include $(INCLUDE_DIR)/package.mk
include ../../lang/python/python3-package.mk

define Package/borgbackup
  SECTION:=utils
  CATEGORY:=Utils
  TITLE:=Utility to back up stuff
  URL:=https://borgbackup.org
  DEPENDS:= \
    +libfuse \
    +libopenssl \
    +liblz4 \
    +libzstd \
    +libacl \
    +pkg-config
  VARIANT:=python3
endef

define Package/borgbackup/description
  borgbackup is a small command-line program to back stuff up.
endef

$(eval $(call Py3Package,borgbackup))
$(eval $(call BuildPackage,borgbackup))
$(eval $(call BuildPackage,borgbackup-src))
```

Is there a way to execute the build of just a single package, or is `make -j2 V=99` the correct way?

Thanks for any input 🙏

The-BB, 2020-11-06:
```
PYTHON3_PKG_SETUP_VARS+= \
	BORG_OPENSSL_PREFIX='$(STAGING_DIR)/usr' \
	BORG_LIBZSTD_PREFIX='$(STAGING_DIR)/usr'
```

BKPepe, 2020-11-06:
> Is there a way to execute the build of just a single package, or is `make -j2 V=99` the correct way?

For Turris, you might want to check this [README](https://gitlab.nic.cz/turris/turris-build/#user-content-building-packages). If you find there something misunderstanding, please let me know or anyone else in Turris-build repository. It is also mirrored on GitHub. Also, this is documented in OpenWrt [docs](https://openwrt.org/docs/guide-developer/single.package).

What you need is to do: `make package/borgbackup/compile -jX V=s`. While at it, please remove OpenWrt copyright according to [CONTRIBUTING.md ](https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md). If you have any other questions, please don't hesitate to ask and we will try to help you!

It would be good if someone can benefit from your packaged borgbackup. So hopefully, you won't give up and we will see pull request soon! Keep going.

noplanman, 2020-11-06:
@BKPepe Thanks for that 👍 

I've tried a few things but other errors are popping up.

Will give it another go when I have time and then open a new issue more specific to building the borgbackup package if need be.

closed 2020-11-06 by noplanman