openwrt/packages #13865: Master branch load balancing displayed offline in multi-dial state!!

zxlhhyccc, 2020-11-06:
[image]

feckert, 2020-11-06:
You'll have to give us a little more information!
I can't do anything with that.

zxlhhyccc, 2020-11-06:
> You'll have to give us a little more information!
> I can't do anything with that.

I set up a single line multi-dial, multi-dial all successful online, but the load balancing display interface is: offline

zxlhhyccc, 2020-11-06:
Instead of the 19.07 branch mWAN3, it will display normally: online.
[image]

feckert, 2020-11-06:
Did test the latest master branch?
If so could try this https://github.com/openwrt/packages/pull/13853

aaronjg, 2020-11-06:
Can you report the output of 

`mwan3 interfaces`

This will help determine if it is a `luci-app-mwan3` bug or an issue in `mwan3`

zxlhhyccc, 2020-11-07:
> mwan3

Error caused by MWAN3

aaronjg, 2020-11-07:
Please ssh to your router post the output of the command I asked for. I can not reproduce and need additional information to determine which package needs to be updated.

zxlhhyccc, 2020-11-07:
[image]

aaronjg, 2020-11-07:
Thanks. Looks like mwan3 did not start. Can you run ‘mwan3 start’ and then ‘logread -e mwan3’ and share the output?

zxlhhyccc, 2020-11-07:
> Thanks. Looks like mwan3 did not start. Can you run ‘mwan3 start’ and then ‘logread -e mwan3’ and share the output?

After running MWAN3 Start, the routing will be down and the routing will have to be restarted before the load balancing can work properly!
[image]

zxlhhyccc, 2020-11-07:
> Did test the latest master branch?
> If so could try this #13853
Use #13853 for the same problem

zxlhhyccc, 2020-11-07:
[image]

aaronjg, 2020-11-07:
What is `mwan3helper`? I don't see it in any of the official feeds. It looks like that may be causing some problems.

Can you share the output of `ipset list`

`opkg list-installed` and `ip route list table all`

Also, if you do a fixed-width block with "\`\`\` \n \<command output\> \n  \`\`\` , rather than pasting images, it will be easier to see what is going on.

zxlhhyccc, 2020-11-08:
> What is `mwan3helper`? I don't see it in any of the official feeds. It looks like that may be causing some problems.
>
> Can you share the output of `ipset list`
>
> `opkg list-installed` and `ip route [...]
ipset list:
`````
Name: mwan3_connected
Type: list:set
Revision: 3
Header: size 8
Size in memory: 184
References: 0
Number of entries: 6
Members:
mwan3_dynamic_v4
mwan3_dynamic_v6
mwan3_custom_v4
mwan3_custom_v6
mwan3_connected_v4
mwan3_connected_v6

Name: mwan3_sticky_https
Type: list:set
Revision: 3
Header: size 8
Size in memory: 88
References: 0
Number of entries: 2
Members:
mwan3_sticky_v4_https
mwan3_sticky_v6_https
````
opkg list-installed
```````
6in4 - 26
AdGuardHome - 0.104.0-1
adbyby-ram_edition - 2.7-20200315
amule - 2.3.2-5
antileech - 2.3.2-1
appfilter - 3.0-1
autocore-arm - 1-28
automount - 2-30
base-files - 235-r14869-9867d08e07
bash - 5.0-4
bc - 1.06.95-1
bind-host - 9.16.8-1
bind-libs - 9.16.8-1
block-mount - 2020-10-16-f25ab8a8-1
boost - 1.74.0-6
boost-chrono - 1.74.0-6
boost-date_time - 1.74.0-6
boost-program_options - 1.74.0-6
boost-random - 1.74.0-6
boost-system - 1.74.0-6
brcmfmac-firmware-usb - 20201022-1
brook - 20200909-1
busybox - 1.31.1-5
ca-bundle - 20200601-1
ca-certificates - 20200601-1
cgi-io - 2020-10-11-d2907bd1-19
chinadns-ng - 1.0-beta.22-4
coremark - 2020-09-17-41537ea3-15
coreutils - 8.32-5
coreutils-base64 - 8.32-5
coreutils-nohup - 8.32-5
coreutils-od - 8.32-5
curl - 7.72.0-2
ddns-scripts - 2.8.2-3
ddns-scripts-services - 2.8.2-3
[... 450 lines cut ...]
anycast fe80:: dev devguest table local proto kernel metric 0 pref medium
anycast fe80:: dev eth0.2 table local proto kernel metric 0 pref medium
anycast fe80:: dev eth0 table local proto kernel metric 0 pref medium
anycast fe80:: dev br-lan table local proto kernel metric 0 pref medium
anycast fe80:: dev wlan1 table local proto kernel metric 0 pref medium
anycast fe80:: dev wlan0 table local proto kernel metric 0 pref medium
anycast fe80:: dev macvlan1 table local proto kernel metric 0 pref medium
anycast fe80:: dev macvlan2 table local proto kernel metric 0 pref medium
local fe80::22cd:59ff:feab:bb8e dev macvlan1 table local proto kernel metric 0 pref medium
local fe80::6a01:6bff:fe96:2e36 dev macvlan2 table local proto kernel metric 0 pref medium
local fe80::767d:24ff:fed3:6e5a dev eth0 table local proto kernel metric 0 pref medium
local fe80::767d:24ff:fed3:6e5a dev br-lan table local proto kernel metric 0 pref medium
local fe80::767d:24ff:fed3:6e5b dev eth0.2 table local proto kernel metric 0 pref medium
local fe80::767d:24ff:fed3:6e5b dev wlan0 table local proto kernel metric 0 pref medium
local fe80::767d:24ff:fed3:6e5c dev wlan1 table local proto kernel metric 0 pref medium
local fe80::f05a:45ff:fec2:2cd0 dev devguest table local proto kernel metric 0 pref medium
ff00::/8 dev devguest table local metric 256 pref medium
ff00::/8 dev br-lan table local metric 256 pref medium
ff00::/8 dev eth0.2 table local metric 256 pref medium
ff00::/8 dev eth0 table local metric 256 pref medium
ff00::/8 dev wlan0 table local metric 256 pref medium
ff00::/8 dev wlan1 table local metric 256 pref medium
ff00::/8 dev macvlan1 table local metric 256 pref medium
ff00::/8 dev macvlan2 table local metric 256 pref medium
````````

aaronjg, 2020-11-08:
1) What is mwan3helper? I don't see it in any of the official feeds. It looks like that may be causing some problems.

2) You seem to be missing a default route for pppoe-vwan1 - please set a metric for all interfaces, according to the setup instructions here: https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3.

3) Could you try the branch here: https://github.com/openwrt/packages/pull/13853 - it looks like you are still on mwan3 2.10.1. Even if you tried it before, please try again, since some new commits have been added (https://github.com/openwrt/packages/pull/13853/commits/94a25dbd92adcf070e10a508f5103d6914449f2a)

closed 2020-11-08 by feckert

zxlhhyccc, 2020-11-08:
With MWAN3Helper removed and updated to 2.10.2, the result of the junction load balancing remains offline. But load balancing is displayed online without overdialing!
At the same time, using MWAN3 Start can start and run, but the network has been connected but cannot access the Internet, you must manually retain and apply the relevant interface in load balancing to access the Internet normally!!
[image]
[image]
[image]
[image]

aaronjg, 2020-11-08:
> With MWAN3Helper removed and updated to 2.10.2, the result of the junction load balancing remains offline. But load balancing is displayed online without overdialing!
At the same time, using MWAN3 Start can start and run, but the network has been connected but cannot access the Internet, you must manually retain and apply the relevant interface in load balancing to access the Internet normally!!

I’m not following some of the terms you are using. Perhaps an issue with machine translation?

What do you mean by “junction” load balancing, “overdialing”, and “manually retain and apply?”

One issue I see is that you do not have a metric associated with the WANs, which will cause some problems. Please see point 2 from before.

PR #13853 (2020-11-08): mwan3: code cleanup and fixes
Maintainer: me
Compile tested: x86_64, APU3, OpenWrt master
Run tested: x86_64, APU3, OpenWrt master

Description:
This is a maintenance P/R with the following goals.
- Clean up the source code
- Fix pending errors through community tests which are found through different test scenarios

Fixes #13823 
Fixes #13865