openwrt/packages #13869: golang: requires golang-src even when binary packages installed

flyn-org, 2020-11-07:
Maintainer: @jefferyto 
Environment: OpenWrt master on x86_64

Description:

I just updated an x86_64 device to the current OpenWrt master, and now I find that golang cannot build programs that rely on standard packages unless golang-src is installed.

Here is an example of trying to build a "Hello, world!" example:
```
# go build hello.go 
hello.go:4:2: cannot find package "fmt" in any of:
	/usr/lib/go-1.15/src/fmt (from $GOROOT)
	/root/go/src/fmt (from $GOPATH)
package command-line-arguments: cannot find package "runtime" in any of:
	/usr/lib/go-1.15/src/runtime (from $GOROOT)
	/root/go/src/runtime (from $GOPATH)
```
I can confirm that fmt.a is installed:
```
# ls /usr/lib/go-1.15/pkg/linux_amd64/fmt.a 
/usr/lib/go-1.15/pkg/linux_amd64/fmt.a
```
Installing golang-src "fixes" this:
```
# opkg install golang-src
#  go build hello.go 
# ./hello
Hello, world!
```
I am not sure why golang-src is required when the binary package is present. Is it possible that a directory name changed because of the version bump, but the package Makefile was not updated to reflect this?

flyn-org, 2020-11-07:
Quick update: I just realized golang depends of golang-src. For some reason that dependency did not get pulled in during my install---not the golang package's fault, though. Thus this is not a regression, as I thought it might be. I guess I am still curious why the source is required when the binary packages appear available.

closed 2020-11-07 by flyn-org

jefferyto, 2020-11-07:
> I guess I am still curious why the source is required when the binary packages appear available.

IIRC this was something I copied from Debian but also tested briefly when I first worked on the Go compiler package.

golang/go#4635 is the reference issue I have. The "mtime hack" mentioned appears to have stopped working (https://github.com/golang/go/issues/12186#issuecomment-132434333) at around Go 1.5. Both issues were marked as duplicates of golang/go#2775, which was resolved by adding support for binary-only packages. (I don't believe the compiler makes a distinction between its own packages and third-party packages.) Binary-only packages were dropped in golang/go#28152.

I haven't looked into this lately - it would be nice to drop the src package dependency but I also doubt many people actually install the Go compiler on their device. I think you're the first person I have heard to do this :joy: