openwrt/packages #13896: mwan3 - Missing dependency

reinerotto, 2020-11-10:
On trunk, and on 19.07.x I get following error messages 
mwan3 status
Interface status:
interface wan is offline and tracking is paused
Command failed: Not found
Failed to parse message data
WARNING: Variable 'interfaces' does not exist or is not an array/object
WARNING: Variable 'wwan' does not exist or is not an array/object
interface wwan is online 00h:00m:00s, uptime 00h:00m:00s and tracking is active
Command failed: Not found
Failed to parse message data
WARNING: Variable 'interfaces' does not exist or is not an array/object
WARNING: Variable 'wifiwan' does not exist or is not an array/object
interface wifiwan is online 00h:00m:00s, uptime 00h:00m:00s and tracking is active

Current ipv4 policies:
standard:
wifiwan (100%)

The reason for this is a missing dependency to rpcd when building own/custom image, like I did.
_Explicitly_ including package rpcd (without options) gives

mwan3 status
Interface status:
interface wan is offline and tracking is down
interface wwan is online 03h:42m:10s, uptime 03h:42m:12s and tracking is active
interface wifiwan is online 03h:42m:43s, uptime 03h:42m:45s and tracking is active

Current ipv4 policies:
standard:
wifiwan (100%)

It looks like having LuCi in the image, it causes silently (also) to include package rpcd. 
However, custom built images do _not_ include LuCi by default, and will be hit byt this issue. Like me. 
For more details, refere to https://forum.openwrt.org/t/mwan3-2-10-1-cpu-hog/78443/15

aaronjg, 2020-11-11:
It looks like the dependency was introduced here: b0acbf057e and is just needed to get "online" and "uptime" statistics. Maybe these functions could be moved into common.sh and called directly, so we don't need to add the `rpcd` dependency.

If the rpcd package is not built, we can alter the Makefile so the mwan3 rpcd module is not installed.

feckert, 2020-11-11:
> It looks like the dependency was introduced here: [b0acbf0](https://github.com/openwrt/packages/commit/b0acbf057e05bbfb165fd83a181c0d1607da5558) and is just needed to get "online" and "uptime" statistics. Maybe these functions could be moved into common.sh and called directly, so we don't need to add the `rpcd` dependency.

Thats right this commit adds the dependency to rpcd. I thought that this rpcd is always include in the software.
And that there is no openwrt system where this rpcd is not included.

> If the rpcd package is not built, we can alter the Makefile so the mwan3 rpcd module is not installed.

This rpcd module is mainly intended to bundle the information. And the main advantiag I see is that the information are available as json. Also other modules, which need information from mwan3, can access the information more easily. The question is, does it make sense to declare the rpcd as optional?
@aaronjg if you want get ride of the use of the rpcd internal is ok for me. That was the easiest way to get the information I needed.

I see 2 possibilities.
* Add dependency to the makefile for rpcd (That would be the easiest way for openwrt-19.07, because the next point release is in the pipline http://lists.openwrt.org/pipermail/openwrt-devel/2020-November/032048.html)
* Remove rpcd usage internal and make rpcd optinal. Maby we could move the rpcd into a own subpackage mwan3-rpcd. (That could be done in the master branch)

reinerotto, 2020-11-11:
I think, you should simply add the dependency to rpcd. When LuCI is included in image, this is redundant, but does not harm. In case of no LuCi (default for custom built images), the dependency will fix this issue.

feckert, 2020-11-11:
Fixed in openwrt-19.07 with commit https://github.com/openwrt/packages/commit/68eb1943ccb35d81c98ab254460d1dd590ce7c04
We need also cherry-pick this to master or fix the dependency so I will leave this open.

aaronjg, 2020-11-11:
I think we should remove internal mwan3 rpcd calls. Since it will only be specialized builds that don’t have rpcd, we can just not install rpcd/mwan3 when that config option is turned off.

aaronjg, 2020-11-13:
@reinerotto Would you mind testing P/R #13908?  That should remove the rpcd dependency for `mwan3 interfaces` and skip installing the mwan3 rpcd script rpcd is not enabled on the image.

reinerotto, 2020-11-13:
Willing to test, but dunno the "How to".

reinerotto, 2020-11-13:
COrrection: Willing to test, but dunno the "How to do" in the build process. Pls, advise.

aaronjg, 2020-11-13:
You can add the repository to your `feeds.conf`
```
src-git packages_ajg https://github.com/aaronjg/openwrt-packages.git;mwan3-rpcd
```
Then run:
```
./scripts/feeds uninstall mwan3
./scripts/feeds install -f -p packages_ajg mwan3
```
to reinstall the package from the repository.

Then a `make menuconfig` to turn of rpcd, and make sure mwan3 is still turned on.

reinerotto, 2020-11-13:
Very sorry, but obviously I can be of no help, because of lack of knowledge regarding the build procedure. Which I can only use the "standard" way. Thus, I do not understand, how/where to use your recipe as written.
Anyway, thank you very much for your efforts.
Cheers,
Reiner

aaronjg, 2020-11-13:
>  Which I can only use the "standard" way. 

That _is_ the standard build procedure for including alternative package feeds:
https://openwrt.org/docs/guide-developer/feeds

commit 68eb1943 (2020-11-11): mwan3: add missing dependency

Add missing rpcd dependency.

Fixes #13896
PR #13908 (2020-11-18): mwan3: remove dependency on rpcd & fix iputils-ping check
Maintainer @feckert 

Allow `mwan3 interfaces` to get uptime via an internal function and
thus remove the dependency on rpcd for `mwan3 interface` calls.

Do not install the rpcd script if an image is built without rpcd. rpcd
is included in the standard images, so this will only impact custom images.

Fixes: #13896
closed 2020-11-18 by feckert