openwrt/packages #13905: arp-scan-database: does not exist

almereyda, 2020-11-12:
Maintainer: @urusha @defanator 
Environment: mips_24kc, TP-Link TL-1043ND ,OpenWrt 19.07.4 r11208-ce6496d796 

Description:

When installing and using `arp-scan`, it will output a warning about missing files

```
Interface: br-lan, datalink type: EN10MB (Ethernet)
WARNING: Cannot open MAC/Vendor file /usr/share/arp-scan/ieee-oui.txt: No such file or directory
WARNING: Cannot open MAC/Vendor file /usr/share/arp-scan/ieee-iab.txt: No such file or directory
WARNING: Cannot open MAC/Vendor file /usr/share/arp-scan/mac-vendor.txt: No such file or directory
Starting arp-scan 1.9.5 with 256 hosts (https://github.com/royhills/arp-scan)

10 packets received by filter, 0 packets dropped by kernel
Ending arp-scan 1.9.5: 256 hosts scanned in 1.804 seconds (141.91 hosts/sec). 0 responded
```

but not output the results of the packets received by the filter.

Last year the `arp-scan-database` had been moved into its separate package 41ff8a50b218eb54906f6e1788eb2def3faba356, but that does not seem to be packaged well, as it does not show up in https://openwrt.org/packages/index/network and cannot be installed, after `opkg update`:

```
# opkg install arp-scan-database
Unknown package 'arp-scan-database'.
Collected errors:
 * opkg_install_cmd: Cannot install package arp-scan-database.
# opkg info arp-scan-database
#
```

From looking at the Makefile for `arp-scan`, we would eventually do good in removing

https://github.com/openwrt/packages/blob/569e06c82ff642f31b2b1231095bafb57dc1ad7d/net/arp-scan/Makefile#L62-L82

and providing it with a similar header to

https://github.com/openwrt/packages/blob/569e06c82ff642f31b2b1231095bafb57dc1ad7d/net/arp-scan/Makefile#L1-L33

in some place like a `packages/net/arp-scan-database/Makefile`.

skaox, 2020-12-18:
Environment : Netgear R6220 (ramips/mt7621) - OpenWrt 19.07.5 r11257-5090152ae3 - Packages (mips_24kc)

Hi,

maybe there is another solution but I had the same problem and solved it with arp-scan_1.9.7 in snapshot :

opkg remove arp-scan
cd /tmp
wget https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/arp-scan_1.9.7-1_mipsel_24kc.ipk
wget https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/arp-scan-database_1.9.7-1_mipsel_24kc.ipk
opkg install arp-scan_1.9.7-1_mipsel_24kc.ipk
opkg install arp-scan-database_1.9.7-1_mipsel_24kc.ipk

I don't know why those packages are still in snapshot [#10694](https://github.com/openwrt/packages/issues/10694).

Have fun ;

feckert, 2020-12-21:
I master branch the package is built and could be installed with the command `opkg update && opkg install arp-scan-database`
https://downloads.openwrt.org/snapshots/packages/x86_64/packages/

closed 2020-12-21 by feckert

skaox, 2024-03-19:
OpenWrt 19.07 ?
Opening a new issue with more information would be a good idea.