openwrt/packages #13906: openwrt netdata and ttyd in https webui is no display

Dimitri2020007, 2020-11-12:
Please make sure that the issue subject starts with `luci-ttyd luci-netdata: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

``` this problem on 19.07 and higher
Formating code blocks by wrapping them with pairs of ```
```

build platform as x86_64

build branch as 19.07

tty terminal no display netdata no display (as https webui)

http webui is normal

but why?

why to this happend?

please tell me what to do

erdoukki, 2021-03-02:
+1 same

diizzyy, 2022-03-12:
Closing this as report covery multiple issues and I'm not even sure what the actually issue is.

closed 2022-03-12 by diizzyy