openwrt/packages #13907: mwan3: cannot access from wanX

ptpt52, 2020-11-12:
wan01 is pppoe ip=192.168.18.14
wan02 is pppoe ip=192.168.18.13
```
/etc/config/network
config interface 'xwan01'
	option proto 'pppoe'
	option username 'user'
	option password 'password'
	option keepalive '5 5'
	option mtu '1492'
	option ipv6 '1'
	option peerdns '0'
	option dns '114.114.114.114'
	option ifname 'mvlan01'
	option metric '41'

config interface 'xwan02'
	option proto 'pppoe'
	option username 'user'
	option password 'password'
	option keepalive '5 5'
	option mtu '1492'
	option ipv6 '1'
	option peerdns '0'
	option dns '114.114.114.114'
	option ifname 'mvlan02'
	option metric '42'
```

main route table:
```
default via 192.168.17.1 dev eth0.1 proto static metric 40 
default via 192.168.18.1 dev pppoe-xwan01 proto static metric 41 
default via 192.168.18.1 dev pppoe-xwan02 proto static metric 42 
1.2.3.4 via 192.168.18.1 dev pppoe-xwan02 
169.254.0.0/16 dev br-lan proto kernel scope link src 169.254.254.254 
192.168.15.0/24 dev br-lan proto kernel scope link src 192.168.15.1 
192.168.17.0/24 dev eth0.1 proto static scope link metric 40 
192.168.18.1 dev pppoe-xwan02 proto kernel scope link src 192.168.18.13 
192.168.18.1 dev pppoe-xwan01 proto kernel scope link src 192.168.18.14 
```

ping from pppoe-server
```
ping 192.168.18.13 ok
ping 192.168.18.14 fail  -- icmp reply goes out via pppoe-xwan02 but it should goes out via pppoe-xwan01
```

top:
````
                     / (pppoe-xwan01)------------------\
client------router-(pppoe-xwan02)---------------------pppoe-server
````

ptpt52, 2020-11-12:
how to setup:
use Ubuntu as pppoe-server, please install pppoe-server pkg
and run:
```
sudo pppoe-server -I <iface> -L 192.168.18.1 -R 192.168.18.10 -N 200
```

router:
```
create macvlan 'mvlan01' for wan(eth0.1)
create macvlan 'mvlan02' for wan(eth0.1)

connect  wan to pppoe-server

create xwan01 for mvlan01 and dial pppoe to server
create xwan02 for mvlan01 and dial pppoe to server
```

wulfy23, 2020-11-12:
how is this related to mwan3?

without further routing table manipulation... what you have asked for is exactly what you have got...

ptpt52, 2020-11-12:
> how is this related to mwan3?
> 
> without further routing table manipulation... what you have asked for is exactly what you have got...

It may/should, if..
if it setup the routing tables and rules to handle this situation.

packet comes in from wanX match the rules to lookup table-of-wanX, and packet reply also lookup this table,
so that would be possible to route back correctly.

But now mwan3 has been changed beyond recognition, based on some unusual understandings

feckert, 2020-12-22:
The issue is old and there were a few changes and fixes because of the routing in mwan3.
I assume that your issue is also fixed.
Please test the new master version.

closed 2020-12-22 by feckert

ptpt52, 2020-12-22:
not really.
this cannot be fixed

feckert, 2020-12-22:
@ptpt52 My problem is here again I do not understand your context!
You have to give me there already more information!
What your exact problem is and how does we could reproduce them.

ptpt52, 2020-12-22:
In fact, I know how to fix this issue, but it cannot be implemented in the current mwan3 framework. I implemented it on my fork of mwan3.
https://github.com/x-wrt/com.x-wrt/tree/master/net/mwan3plus

to reproduce this issue, just setup the same ip on your router
```
wan1  192.168.18.xxx
wan2 192.168.18.yyy
```