openwrt/packages #13909: mwan3: Ping echo reply going over the wrong WAN interface from external hosts

jamesmacwhite, 2020-11-13:
Maintainer: @feckert 
Environment: 19.07.3 (mwan3 2.8.14-2)

Description:

I originally posted about an interesting issue I observed with ICMP and replies going over the wrong interface on the forums a while ago. As a bit of background there are services that can track your connection by sending ICMP data every second or at regular intervals as a way of monitoring uptime. In my case, due to having multiple WANs I have multiple monitors pinging different IPs but they are terminating on the same router still. I don't block ICMP inbound, contrary to all the sites that say ICMP should be stealth (it's total BS). 

https://forum.openwrt.org/t/wan-ping-echo-reply-going-over-the-wrong-interface/75317

The behaviour observed with tcpdump I'm assuming is because mwan3 and the default rule will be to use a primary WAN and fallback to the next one in the policy if not available, however this will mess up the replies for a WAN interface that isn't the primary. I tend to find sometimes the ICMP traffic is correctly routed then it might just suddenly show request timeouts, because the echo reply is then going over the wrong interface as seen by tcpdump example.

Is the solution to define a specific mwan3 rule before the default rule that targets ICMP specifically and the dest_ip being the external IP address being pinged from whatever host?

```
config rule 'isp2_icmp'
	option proto 'icmp'
	option dest_ip '87.xx.xx.xx'
	option family 'ipv4'
	option use_policy 'isp2_only'

config rule 'isp3_icmp'
	option proto 'icmp'
	option dest_ip '81.xx.xxx.xxxx'
	option family 'ipv4'
	option use_policy 'isp3_only'
```

Or is there a better way? Right now I'm looking at a way to try and lock the ICMP traffic to properly go over the right interface, as without handling is seems to be OK sometimes, then will start to go wrong, doing mwan3 restart sometimes will help, but not all the time, so forcing with a specific rule seems like the best way, unless it can be specifically handled in mwan3 itself?

aaronjg, 2020-11-13:
The mwan3 conntrack rules should be handling that use case. (this is a slightly different issue than another bug report regarding to pings originating at the gateway).

What do you get with:

```
conntrack -F
conntrack -L|grep <ip of tracking server>
conntrack -E|grep <ip of tracking server>
```

The last command will show updates to the table, so you may need to wait for some failed pings before anything shows up.

jamesmacwhite, 2020-11-13:
It's an interesting one, it tends to work initially but can eventually start to send echo replies over the wrong interface. I'm assuming when interface statuses start changing the then packets might go through the wrong WAN as mwan3 adjusts to the set policies.

I'm happy with configuring the above rules as a workaround as they seem to be hit according to `mwan3 status` and so far seems to make ICMP go through the right interfaces.

aaronjg, 2020-11-13:
Hmm. It should be caught with the mwan3_ifaces_in and conntrack rules, which precede the policy based rules. 

Can you force it to happen if you do `mwan3 ifdown <iface>` on either of the interfaces?

jamesmacwhite, 2020-11-13:
I'll see if I can trigger the behaviour, and report back if I get any further logs.

ptpt52, 2020-11-13:
@jamesmacwhite  may be you are talking about the same issue as this https://github.com/openwrt/packages/issues/13907

aaronjg, 2020-11-13:
> @jamesmacwhite may be you are talking about the same issue as this #13907

As I said before, it’s a different issue. Things are handled differently when the packet originates from a directly connected network.

jamesmacwhite, 2020-11-15:
I'm inclined to agree with @aaronjg, I'm still using the stable branch build so 2.8.x, so I don't think it is related. I've had a bit of nightmare with my Linksys WRT3200ACM lately with CPU stalls and it dropping all routing after about 3 hours or so powered on time, not related mwan3 I don't think, for some reason 19.07.4 seems to be a problem for me but even 19.07.3 was doing the same thing but either way it really doesn't like something kernel related or something I've configured. Because I've got A LOT of WAN interfaces varying from different types PPPoE, L2TP, DOCSIS Cable, Wireguard etc, I'm not really wanting to mess with it too much, as I need things to work reliably for WFH currently. It's now the first time it has been stable over 4 hours the past couple of days, so it seems to have stopped the cycle of being up for about 4 hours and then eventually rebooting itself.

The rules workaround the issue for me for now, but I'll note the issue, maybe when I look at moving to 2.10 I'll revisit!

aaronjg, 2020-11-15:
I just backported https://github.com/torvalds/linux/commit/46d6c5ae953cc0be38efd0e469284df7c4328cf8 to 19.07, which fixed a longstanding bug in routing for tunnel interfaces (related to the L2TP bug we discussed earlier). I'm not sure what is going on with your tunnel interfaces, but if the issues persist, you might want to give that a try.

Backport is available here: https://github.com/openwrt/openwrt/pull/3587/commits/f078091c13ef2bd90c2e19592d2566bab8a7f3eb

jamesmacwhite, 2020-11-15:
@aaronjg Nice! Thank you for that. I'll note this for the future. I'm staying on 19.07.3 until I figure out why my router is having CPU stalls.

aaronjg, 2020-11-15:
I had some issues on 19.07.4 where sometimes packets from one tunnel were routed over a second tunnel rather than the main wan. Everything looked fine with respect to firewall marks, but when I took down a tunnel interface, the other tunnel interfaces would mysteriously also come down. I confirmed via tcpdump that the packets for the second tunnel were going out of the first.

Once I did the backport of the patch and applied it to 19.07.4, the issue went away. So perhaps your stability issues are related. Though it would require building the entire image from source after applying that patch.

jamesmacwhite, 2020-12-07:
Looks like that patch made it into 19 07.5, so I'll retest this soon.

jamesmacwhite, 2020-12-09:
So I upgraded to 19.07.5 via sysupgrade and on reboot I could see the ICMP echo reply for WANB going over the WAN interface after the firmware flash. However, it did seem to resolve itself after a bit of time. So maybe that kernel patch has at least changed something for the better.

This is the graph for my main WAN, the red is the brief drop when the firmware flash occurred. Very small amount of time:

[image]

Then if we look at the monitoring graph for WANB, it is longer but there's a couple of reasons for this:

[image]

* WANB is PPPoE so it takes longer to re-establish the connection again after a reboot, compared to WAN which is DOCSIS Cable via DHCP.

I confirmed with tcpdump that I saw ICMP replies for WANB going over WAN, but the graph shows the situation resolved itself and I didn't do anything to influence it.

jamesmacwhite, 2020-12-09:
Actually, I think it is a conntrack issue, because doing `conntrack -F`, which flushes conntrack seems to clear the problem.

feckert, 2020-12-09:
@jamesmacwhite mwan3 can flash the conntrack  table when an interface event happens see
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L1151

jamesmacwhite, 2020-12-09:
Interesting. I've definitely narrowed it down to conntrack.

Before flushing conntrack. Note only echo request packets on WANB, no replies.

```
root@linksys-wrt3200acm:~# tcpdump -i pppoe-wanb -p icmp
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on pppoe-wanb, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
19:51:28.273877 IP pingbox1.thinkbroadband.com > static-87-74-29-11.vodafonexdsl.co.uk: ICMP echo request, id 32865, seq 35949, length 8
19:51:29.273715 IP pingbox1.thinkbroadband.com > static-87-74-29-11.vodafonexdsl.co.uk: ICMP echo request, id 32865, seq 35950, length 8
```

After flushing conntrack:

```
root@linksys-wrt3200acm:~# conntrack -F
conntrack v1.4.5 (conntrack-tools): connection tracking table has been emptied.
root@linksys-wrt3200acm:~# tcpdump -i pppoe-wanb -p icmp
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on pppoe-wanb, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
19:51:36.273915 IP pingbox1.thinkbroadband.com > static-87-74-29-11.vodafonexdsl.co.uk: ICMP echo request, id 32865, seq 35957, length 8
19:51:36.273960 IP static-87-74-29-11.vodafonexdsl.co.uk > pingbox1.thinkbroadband.com: ICMP echo reply, id 32865, seq 35957, length 8
19:51:37.273611 IP pingbox1.thinkbroadband.com > static-87-74-29-11.vodafonexdsl.co.uk: ICMP echo request, id 32865, seq 35958, length 8
19:51:37.273665 IP static-87-74-29-11.vodafonexdsl.co.uk > pingbox1.thinkbroadband.com: ICMP echo reply, id 32865, seq 35958, length 8
19:51:38.273509 IP pingbox1.thinkbroadband.com > static-87-74-29-11.vodafonexdsl.co.uk: ICMP echo request, id 32865, seq 35959, length 8
19:51:38.273552 IP static-87-74-29-11.vodafonexdsl.co.uk > pingbox1.thinkbroadband.com: ICMP echo reply, id 32865, seq 35959, length 8
19:51:39.273648 IP pingbox1.thinkbroadband.com > static-87-74-29-11.vodafonexdsl.co.uk: ICMP echo request, id 32865, seq 35960, length 8
19:51:39.273705 IP static-87-74-29-11.vodafonexdsl.co.uk > pingbox1.thinkbroadband.com: ICMP echo reply, id 32865, seq 35960, length 8
```

feckert, 2020-12-16:
@jamesmacwhite Maybe we should include this in the documentation that flashing the conntrack table makes sense if an interface event happened.
Can you also verify again if the same problem exists with the newest master branch version?
Since we use the new c wrapper.

jamesmacwhite, 2020-12-17:
I can potentially workaround this by using the `/etc/mwan3.user` with something like:

```
if [ "${ACTION}" = "ifup" ] ||  [ "${ACTION}" = "ifupdate" ] ; then
  conntrack -F
fi

This would then flush conntrack, although it will probably excessively flush it when not always needed, given I seem to only need to do it once and everything is fine, but this would likely work.

I haven't tested the latest version because I had been watching the progress and development of it and it seems it had some issues, in addition when I tested the 2.10.x branch, my IPv6 was strangely broke and unfortunately, I didn't have time to debug it, so switched back, at some point I'll look at retesting.

feckert, 2020-12-17:
> I can potentially workaround this by using the `/etc/mwan3.user` with something like:
> 
> ```shell
> if [ "${ACTION}" = "ifup" ] ||  [ "${ACTION}" = "ifupdate" ] ; then
>   conntrack -F
> fi

I don't understand about the `/etc/mwan3.user` use for this use case.
This is already implemented in mwan3.
```
config interface wan
    list flush_conntrack 'ifup'
    list flush_conntrack 'ifupdate'

```

master:
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L1141

openwrt-19.07:
https://github.com/openwrt/packages/blob/openwrt-19.07/net/mwan3/files/lib/mwan3/mwan3.sh#L1170

jamesmacwhite, 2020-12-17:
@feckert Ah! I understand your comment earlier about the docs, I had no idea this existed and was configurable on `/etc/config/mwan3`.

I will add it to the docs and configure conntrack being flushed this way!

aaronjg, 2020-12-17:
> Can you also verify again if the same problem exists with the newest master branch version?
> Since we use the new c wrapper.

Since these are inbound pings, I suspect the issue will persist but would be good to know.

> I haven't tested the latest version because I had been watching the progress and development of it and it seems it had some issues, in addition when I tested the 2.10.x branch, my IPv6 was strangely [...]

2.10.x branch has had many bug fixes since first merging, so hopefully should be working well. If you have any IPv6 issues with it, it would be really helpful to identify them so I can work on fixing it before this is included in the next major OpenWRT release.
20.x

> although it will probably excessively flush it when not always needed,

Have you tried to identify the problematic conntrack entry? I wonder if there would be a way to fix this without flushing the entire conntrack table.

jamesmacwhite, 2020-12-17:
@aaronjg Thanks. Appreciate my comment around IPv6 "not working" isn't helpful. Unfortunately given work from home I can't break things at the moment, but hopefully over the Christmas period I'll have a chance to investigate more thoroughly as to what happens with my setup on 2.10.x

Regarding conntrack, I'm going to assume it's specific to the pingbox1.thinkbroadband.com host, but at least with the `list flush_conntrack method` it can be a little less excessive than just `conntrack -F` all the time and I'll add that to the mwan3 docs as I had no idea this was possible!

jamesmacwhite, 2020-12-17:
Added `flush_conntrack` option to the interface configuration on the docs. This has been missing for a while!

https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#interface_configuration

jamesmacwhite, 2020-12-22:
Closing as ultimately seems like conntrack issue. Having conntrack flushed on interface events is the solution for this behaviour.

closed 2020-12-22 by jamesmacwhite