openwrt/packages #13915: php7-mod-intl: seems unable to support NextCloud's use of Guzzle

flyn-org, 2020-11-13:
Maintainer: @mhei 
Environment: master on x86_64

Description:
The php7-mod-intl packages seems unable to support NextCloud's use of the Guzzle HTTP library. The presence of PHP's intl module activates Guzzle's IDN conversion. When the is active, I see the following in my logs when interacting with NextCloud:

```
{"reqId":"[ID]","level":3,"time":"2020-11-13T15:13:07+00:00","remoteAddr":"[A.B.C.D]","user":"[USER]","app":"internet_connection_check","method":"GET","url":"/share/index.php/settings/ajax/checksetup","message":{"Exception":"GuzzleHttp\\Exception\\InvalidArgumentException","Message":"IDN conversion failed","Code":0,"Trace":[...]}
```

flyn-org, 2020-11-13:
I just noticed this: https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/php7/compile.txt, which might be related. (Perhaps fixing/updating PHP in that context will help here.)

mhei, 2020-11-13:
Please check whether https://github.com/openwrt/packages/pull/13893 resolved your issue.

flyn-org, 2020-11-14:
I installed php7-mod-intl 7.4.12-2 and updated icu, but the "IDN conversion failed" error still occured. Installing icu-full-data seemed to fix this. I think the timing of a change within NextCloud coupled with the PHP package hiccup confused me. Things look good now. Thank you!

closed 2020-11-14 by flyn-org

mhei, 2020-11-14:
Thanks you for your testing and feedback.