openwrt/packages #13917: ttyd: Launches insecurely upon installation

Levitator1, 2020-11-14:
Just noticed that the moment you install ttyd, the service launches wide-open without any authentication by default, so any machine that has access to the router is instantly given unprotected root access. Going to suggest that maybe this service should be installed disabled, permitting the user to enable it once configured.

Jose

feckert, 2020-11-16:
From my point of view, we should become a little smarter and not only disable ttyd
I would start the ttyd as https  and also use the certificate from the uhttpd.
This has to be converted before it works into pem.
From my point of view this would be a good user experience.

Levitator1, 2020-11-17:
Well, yes, there are better options, but I was suggesting a minimal action
to remediate the security hole.

wulfy23, 2020-12-21:
who is the maintainer of this package?