openwrt/packages #13920: iputils: incorrect version string

aaronjg, 2020-11-15:
Maintainer: @nmeyerhans 

Ping from 20190709
```
# ping -V
ping from iputils s20190709
```

Ping from 20200821
Downloaded from here:
http://downloads.openwrt.org/snapshots/packages/arm_cortex-a15_neon-vfpv4/packages/iputils-ping_20200821-1_arm_cortex-a15_neon-vfpv4.ipk
```
# ping -V
ping from iputils v19.07.2-64-gefe837de84
```

Ping from 20200821 compiled on ubuntu
```
$ ./builddir/ping/ping -V
ping from iputils s20200821-3-g86ed089
```

I noticed this because mwan3 does some version checks on the ping version to assess its capabilities, and it was not detecting the version correctly.

nmeyerhans, 2021-02-03:
Fixed with #14638

closed 2021-02-03 by nmeyerhans

aaronjg, 2021-02-03:
Even if we can’t keep the iputils git tag, can we keep the ‘s’ prefix to make it consistent with previous iputils versions and the prefix of iputils built on other systems?

nmeyerhans, 2021-02-03:
> Even if we can’t keep the iputils git tag, can we keep the ‘s’ prefix to make it consistent with previous iputils versions and the prefix of iputils built on other systems?

No. Ping has changed its versioning scheme upstream. This will propagate out to other distros and environments in time.

aaronjg, 2021-02-03:
When building from source on Ubuntu I get: 
```
$ ./builddir/ping/ping -V
ping from iputils s20200821-3-g86ed089
```

Are you saying they changed the versioning again since then to remove the “s”?

nmeyerhans, 2021-02-03:
Yep! See [their changelog entry](https://github.com/iputils/iputils/blob/master/CHANGES#L4).

aaronjg, 2021-02-03:
> Yep! See [their changelog entry](https://github.com/iputils/iputils/blob/master/CHANGES#L4).

Got it. Thanks!