openwrt/packages #13925: boinc: update to version 7.16.11

BKPepe, 2020-11-16:
Maintainer: @N30dG @smoe
Environment: all

Description:

In OpenWrt, we have version 7.16.6, but I see there is the latest version 7.16.11.
Can you guys update it, please?

smoe, 2020-11-16:
Hello, the BOINC folks have introduced a mandatory file /etc/boinc-client/config.properties and the versions that follow that you do not want on OpenWrt. There is some light on the horizon https://github.com/BOINC/boinc/pull/4071 but before the BOINC devs do not show some better taste, there is IMHO no point in updating.

So, let me thank you for your interest in our package. 7.16.6. is fully functional from what I know about it. Let's live with it until 7.16.12 surfaces and has experienced some scrutiny on Debian+Ubuntu.

BKPepe, 2021-03-22:
Updated in https://github.com/openwrt/packages/pull/14862, closing.

closed 2021-03-22 by BKPepe