openwrt/packages #13936: "[package_install] kernel 5.4.75-1 Segmentation fault (core dumped)" using snapshot imagebuilder-bcm27xx for Raspberry Pi4

robcollins55, 2020-11-18:
For version r14955-a4d52522c7 snapshot published Nov 17 2020.

Running Openwrt imagebuilder as follows:
make image PROFILE=rpi-4 PACKAGES="luci-ssl-nginx luci-app-acme acme ca-certificates php7-fastcgi uwsgi luci-theme-openwrt-2020 luci-app-statistics collectd-mod-thermal luci-app-sqm nano bash kmod-usb-net-asix kmod-usb-net-asix-ax88179 kmod-usb-net-cdc-ether kmod-usb-net-ipheth kmod-usb-net-kaweth kmod-usb-net-mcs7830 kmod-usb-net-pegasus kmod-usb-net-rtl8150 kmod-usb-net-rtl8152 kmod-usb-net-sr9700 luci-app-wireguard wireguard-tools wireguard qrencode luci-app-samba4 luci-app-ddns luci-app-adblock arp-scan arp-scan-database netdiscover kmod-fs-exfat kmod-fs-ext4 kmod-fs-ntfs kmod-fuse kmod-usb-storage kmod-usb2 kmod-usb2-pci kmod-usb3 kmod-fs-autofs4 block-mount blockd" FILES=files/

Returns error and exits as follows:
Installing packages...
Installing libc (1.1.24-2) to root...
Installing libgcc1 (8.4.0-2) to root...
Downloading https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711/packages/libgcc1_8.4.0-2_aarch64_cortex-a72.ipk
Configuring libgcc1.
Configuring libc.
Installing kernel (5.4.75-1-e2202261d8d97069c53b903eb9986de9) to root...
Configuring kernel.
make[2]: *** [Makefile:161: package_install] Segmentation fault (core dumped)

Being a kernel install error I did not investigate further due to, well, being the kernel. Just letting all know that this is happening and likely needs looking into by those more knowledgeable than me.  Please post response if you would like any further info extracted and posted up - please take it easy on me with requests though and be very specific re how I would get the info to send as I am just an enthusiast not a coder or sysadmin.  Cheers!

aparcar, 2020-11-22:
While I was able to reproduce the error ([buildlog](https://paste2.org/G34PO4pE)) I couldn't make sense out of it, I tried again with todays snapshot and the build went fine.

robcollins55, 2020-11-22:
Thanks Paul. I do this building process on a weekly basis so I'll do again today or tomorrow and confirm if I'm still getting the error or it has disappeared for me too.  I'll report back shortly and hopefully we can close the issue ticket.

aparcar, 2020-11-22:
Excellent thank you for testing & reporting!

wulfy23, 2020-11-23:
in case @dangowrt picks this up again sometime... keep an eye out for
```
writev(4, [{iov_base="*** Error in `", iov_len=14}, {iov_base="/br/staging_dir/host/bin/opkg", iov_len=29}, {iov_base="': ", iov_len=3}, {iov_base="free(): invalid pointer", iov_len=23}, {iov_base=": 0x", iov_len=4}, {iov_base="00007fafa8e13278", iov_len=16}, {iov_base=" ***\n", iov_len=5}], 7*** Error in `/br/staging_dir/host/bin/opkg': free(): invalid pointer: 0x00007fafa8e13278 ***
```
from failed wget || missing package... so long as it finds what it's looking for it's likely dormant...

thanks for your work on this...

robcollins55, 2020-12-01:
I'm no longer getting the error so it indeed appears to have been resolved it has disappeared for me too. Closing the issue ticket, thanks for all your help :smile: .

closed 2020-12-01 by robcollins55