openwrt/packages #13947: xtables-addons: build breakes with kernel 5.4.78

adschm, 2020-11-19:
Bumping kernel to 5.4.78 (in preparation) breaks build of this package with the following error(s):

```
make[7]: Entering directory '/data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/linux-5.4.78'
  CC [M]  /data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/xtables-addons-3.10/extensions/xt_DELUDE.o
/data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/xtables-addons-3.10/extensions/xt_DELUDE.c: In function 'delude_send_reset':
/data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/xtables-addons-3.10/extensions/xt_DELUDE.c:125:30: error: passing argument 2 of 'ip_route_me_harder' from incompatible pointer type [-Werror=incompatible-pointer-types]
  if (ip_route_me_harder(net, nskb, addr_type))
                              ^~~~
In file included from ./include/linux/netfilter/x_tables.h:245,
                 from /data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/xtables-addons-3.10/extensions/xt_DELUDE.c:20:
./include/linux/netfilter_ipv4.h:19:54: note: expected 'struct sock *' but argument is of type 'struct sk_buff *'
 int ip_route_me_harder(struct net *net, struct sock *sk, struct sk_buff *skb, unsigned addr_type);
                                         ~~~~~~~~~~~~~^~
/data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/xtables-addons-3.10/extensions/xt_DELUDE.c:125:36: warning: passing argument 3 of 'ip_route_me_harder' makes pointer from integer without a cast [-Wint-conversion]
  if (ip_route_me_harder(net, nskb, addr_type))
                                    ^~~~~~~~~
In file included from ./include/linux/netfilter/x_tables.h:245,
                 from /data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/xtables-addons-3.10/extensions/xt_DELUDE.c:20:
./include/linux/netfilter_ipv4.h:19:74: note: expected 'struct sk_buff *' but argument is of type 'unsigned int'
 int ip_route_me_harder(struct net *net, struct sock *sk, struct sk_buff *skb, unsigned addr_type);
                                                          ~~~~~~~~~~~~~~~~^~~
/data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/xtables-addons-3.10/extensions/xt_DELUDE.c:125:6: error: too few arguments to function 'ip_route_me_harder'
  if (ip_route_me_harder(net, nskb, addr_type))
      ^~~~~~~~~~~~~~~~~~
In file included from ./include/linux/netfilter/x_tables.h:245,
                 from /data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/xtables-addons-3.10/extensions/xt_DELUDE.c:20:
./include/linux/netfilter_ipv4.h:19:5: note: declared here
 int ip_route_me_harder(struct net *net, struct sock *sk, struct sk_buff *skb, unsigned addr_type);
     ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[8]: *** [scripts/Makefile.build:266: /data/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_generic/xtables-addons-3.10/extensions/xt_DELUDE.o] Error 1
```

This appears to be caused by the following kernel commit, adding an argument to the ip_route_me_harder function:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable-rc.git/commit/?h=queue/5.4&id=b255b5cab13ac78c43b8c10c0e904b3e6f76b251

I'm not sure how to address this properly on the packages side (local patch, upstream patch, ...) and how to care about the synchronicity of main and packages repo in this context. Therefore, and since I have no specific know-how in the netfilter context, I'd be happy if somebody else could take over here.

@jow @pprindeville @dengqf6 @neheb

adschm, 2020-11-19:
Took the wrong jow again, @jow- is listed as maintainer.

adschm, 2020-11-19:
Appears to be fixed in (xtables-addons) upstream commit 0ab324790015a6396be5678b6dffeeaa1cd28299 "build: adjust for changed signature of ip_route_me_harder".

There also appears to be a 3.11 release (we have 3.10), but that does not include the relevant patch. However, one could update if touching this anyway.

ptpt52, 2020-11-19:
The upstream commit here:
```
commit 0ab324790015a6396be5678b6dffeeaa1cd28299
Author: Jan Engelhardt <jengelh@inai.de>
Date:   Thu Nov 19 12:28:55 2020 +0100

    build: adjust for changed signature of ip_route_me_harder
    
    (Cf. commit 46d6c5ae953cc0be38efd0e469284df7c4328cf8 in Linux.)

diff --git a/extensions/compat_xtables.h b/extensions/compat_xtables.h
index 36f3223..ee0caa0 100644
--- a/extensions/compat_xtables.h
+++ b/extensions/compat_xtables.h
@@ -41,6 +41,12 @@
 		((const unsigned char *)&addr)[3]
 #	define NIPQUAD_FMT "%hhu.%hhu.%hhu.%hhu"
 #endif
+#if LINUX_VERSION_CODE >= KERNEL_VERSION(5, 10, 0) || \
+    LINUX_VERSION_CODE >= KERNEL_VERSION(5, 9, 9) && LINUX_VERSION_CODE < KERNEL_VERSION(5, 10, 0)
+#else
+#	define ip_route_me_harder(xnet, xsk, xskb, xaddrtype) ip_route_me_harder((xnet), (xskb), (xaddrtype))
+#	define ip6_route_me_harder(xnet, xsk, xskb) ip_route_me_harder((xnet), (xskb))
+#endif
 
 static inline struct net *par_net(const struct xt_action_param *par)
 {
diff --git a/extensions/xt_DELUDE.c b/extensions/xt_DELUDE.c
index b384c8e..89cc41a 100644
--- a/extensions/xt_DELUDE.c
+++ b/extensions/xt_DELUDE.c
@@ -121,8 +121,7 @@ static void delude_send_reset(struct net *net, struct sk_buff *oldskb,
 
 	/* ip_route_me_harder expects skb->dst to be set */
 	skb_dst_set(nskb, dst_clone(skb_dst(oldskb)));
-
-	if (ip_route_me_harder(net, nskb, addr_type))
+	if (ip_route_me_harder(net, nskb->sk, nskb, addr_type))
 		goto free_nskb;
 	else
 		niph = ip_hdr(nskb);
diff --git a/extensions/xt_ECHO.c b/extensions/xt_ECHO.c
index e99312b..4376d2f 100644
--- a/extensions/xt_ECHO.c
+++ b/extensions/xt_ECHO.c
@@ -191,8 +191,7 @@ echo_tg4(struct sk_buff *oldskb, const struct xt_action_param *par)
 
 	/* ip_route_me_harder expects the skb's dst to be set */
 	skb_dst_set(newskb, dst_clone(skb_dst(oldskb)));
-
-	if (ip_route_me_harder(par_net(par), newskb, RTN_UNSPEC) != 0)
+	if (ip_route_me_harder(par_net(par), newskb->sk, newskb, RTN_UNSPEC) != 0)
 		goto free_nskb;
 
 	newip->ttl = ip4_dst_hoplimit(skb_dst(newskb));
diff --git a/extensions/xt_TARPIT.c b/extensions/xt_TARPIT.c
index 4926f2e..118c26c 100644
--- a/extensions/xt_TARPIT.c
+++ b/extensions/xt_TARPIT.c
@@ -265,7 +265,7 @@ static void tarpit_tcp4(struct net *net, struct sk_buff *oldskb,
 #endif
 		addr_type = RTN_LOCAL;
 
-	if (ip_route_me_harder(net, nskb, addr_type))
+	if (ip_route_me_harder(net, nskb->sk, nskb, addr_type))
 		goto free_nskb;
 	else
 		niph = ip_hdr(nskb);
@@ -398,8 +398,7 @@ static void tarpit_tcp6(struct net *net, struct sk_buff *oldskb,
 	              &ipv6_hdr(nskb)->daddr, sizeof(struct tcphdr),
 	              IPPROTO_TCP,
 	              csum_partial(tcph, sizeof(struct tcphdr), 0));
-
-	if (ip6_route_me_harder(net, nskb))
+	if (ip6_route_me_harder(net, nskb->sk, nskb))
 		goto free_nskb;
 
 	nskb->ip_summed = CHECKSUM_NONE;

```

upstream commit may be incorrect
@zx2c4  Please help review on this

I think it may use `oldskb->sk` instead of `newskb->sk` or `nskb->sk`
and use the `par->state->sk` instead of `newskb->sk`

zx2c4, 2020-11-19:
The rule of thumb with porting this is: if you can get access to `struct nf_hook_state *state` from the function, then use `state->sk`, instead of `skb->sk`.

jmarcet, 2020-11-22:
There is a new upstream release, 3.13, which builds fine on 5.4.79

neheb, 2020-11-23:
Version bumped.

closed 2020-11-23 by neheb

adschm, 2020-11-23:
Thank you for taking care of this.

adschm, 2020-11-24:
Due to https://github.com/openwrt/openwrt/commit/c9c7b4b3945c01c2aadf3ef5d9a77c8200db80f1, we will probably have to backport a fix for this package to 19.07 ...

db260179, 2020-11-24:
> Due to [openwrt/openwrt@c9c7b4b](https://github.com/openwrt/openwrt/commit/c9c7b4b3945c01c2aadf3ef5d9a77c8200db80f1), we will probably have to backport a fix for this package to 19.07 ...

Yes, can confirm this breaks the V19.07 branch build.

zizomizo84gh, 2021-04-30:
Hi Everyone,

I am encountering the same problem but I am not a developer guy like you to understand all the terms here (like upstream commit ). I do not understand even the diff outcome above and how can I get a benefit from it. 

Should I follow the diff outcome and substitute in the scripts?
May someone help me please to build the rtpengine?

I am working on:
Linux T450 5.8.0-50-generic #56~20.04.1-Ubuntu SMP Mon Apr 12 21:46:35 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

uname -r => 5.8.0-50-generic)

thanks in advance and best regards