openwrt/packages #13973: gst1-plugins-base: fails to build when no module is selected

luizluca, 2020-11-22:
Maintainer: @flyn-org @thess 
Environment: master for all targets

Description:

When gst1-plugins-base is built only for its libraries (no module is selected), the build will fail with:

```
cp: cannot stat './usr/lib/gstreamer-1.0/libgst*.so': No such file or directory
```

I know that it might be unusual but sometimes I just need to build a module that depends on no other module.

As a suggestion, it would work better with something like:

```
--- a/multimedia/gst1-plugins-base/Makefile
+++ b/multimedia/gst1-plugins-base/Makefile
@@ -183,9 +183,10 @@ define Build/InstallDev
                $(1)/usr/lib/ \
        )
        $(INSTALL_DIR) $(1)/usr/lib/gstreamer-$(GST_VERSION)
-       ( cd $(PKG_INSTALL_DIR); $(CP) \
-               ./usr/lib/gstreamer-$(GST_VERSION)/libgst*.so \
-               $(1)/usr/lib/gstreamer-$(GST_VERSION)/ \
+       ( cd $(PKG_INSTALL_DIR); $(FIND) \
+               ./usr/lib/gstreamer-$(GST_VERSION)/ -name 'libgst*.so' -print0 | \
+               $(XARGS) --null --no-run-if-empty $(CP) \
+               --target-directory=$(1)/usr/lib/gstreamer-$(GST_VERSION)/ \
        )
        $(INSTALL_DIR) $(1)/usr/lib/pkgconfig
        ( cd $(PKG_INSTALL_DIR); $(CP) \

```

It's just a quick suggestion. It might be a better or simpler way to fix this. Also, it might be interesting to fix other similar gst situations where the glob expression might not expand into existing files.

flyn-org, 2020-11-23:
Would you kindly submit a pull request?

luizluca, 2020-11-23:
I didn't do it because I wasn't sure if I should do it for every similar cp call or if I only change that specific cp. Is it interesting to have the other cases pass without an error when no file exists?

PR #14025 (2020-11-28): gst1-plugins-base: fix build with no module
gst1-plugins-base might be required only for its libraries, not modules.
However, InstallDev tries to copy them unconditionally, failing when
no modules are selected/compiled.

Fixes #13973

Maintainer: @flyn-org @thess
Compile tested: arc_generic,armvirt_64,ath79_generic,powerpc,ramips_mt7620,x86_64,x86_generic
Run tested: not needed (Makefile fix)

Description:

It deals with missing or empty $(PKG_INSTALL_DIR)/usr/lib/gstreamer-$(GST_VERSION)

I left other $(CP) cases intact as I don't know if those source files could be missing without being an e [... cut]
commit 56eb4078 (2020-11-28): gst1-plugins-base: fix build with no module

gst1-plugins-base might be required only for its libraries, not modules.
However, InstallDev tries to copy them unconditionally, failing when
no modules are selected/compiled.

Fixes #13973
closed 2020-11-28 by hnyman