openwrt/packages #13998: docker-ce: dependency check to runc throws an error

feckert, 2020-11-24:
Maintainer: @G-M0N3Y-2503 
Environment: x86_64, APU3, OpenWrt master

Description:

Since last [runc](https://github.com/openwrt/packages/commit/3a06ce55959016653767a1d2c276594ad1eeb743#diff-fc53584dbfaf6af9bd0e18df872ee34679ddf988a9d970f15dbebc2186938021) update I get the following error.

```
# Verify dependencies are the vendored version
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../containerd/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.13/components/engine/hack/dockerfile/install/containerd.installer" ); if [ $VEN_VER != $DEP_VER ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=$VEN_VER' in '../containerd/Makefile', found 'PKG_SOURCE_VERSION:=$DEP_VER'"; exit 1; fi )
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../libnetwork/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.13/components/engine/hack/dockerfile/install/proxy.installer" ); if [ $VEN_VER != $DEP_VER ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=$VEN_VER' in '../libnetwork/Makefile', found 'PKG_SOURCE_VERSION:=$DEP_VER'"; exit 1; fi )
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../runc/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/home/feckert/workspace/openwrt/LDM-master-64/build/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.13/components/engine/hack/dockerfile/install/runc.installer" ); if [ $VEN_VER != $DEP_VER ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=$VEN_VER' in '../runc/Makefile', found 'PKG_SOURCE_VERSION:=$DEP_VER'"; exit 1; fi )
ERROR: Expected 'PKG_SOURCE_VERSION:=dc9208a3303feef5b3839f4323d9beb36df0a9dd' in '../runc/Makefile', found 'PKG_SOURCE_VERSION:=ff819c7e9184c13b7c2607fe6c30ae19403a7aff'
```

feckert, 2020-11-24:
I have just seen that this change has been rushed in prematurely.
https://github.com/openwrt/packages/pull/13874#issuecomment-731669788

ptpt52, 2020-11-24:
same issue found.

G-M0N3Y-2503, 2020-11-24:
> I have just seen that this change has been rushed in prematurely.
> [#13874 (comment)](https://github.com/openwrt/packages/pull/13874#issuecomment-731669788)

Hmmm, it was supposed to go in with docker 20.x and didn't get much of a chance to advise against it going in early.

As mentioned in https://github.com/openwrt/packages/commit/3a06ce55959016653767a1d2c276594ad1eeb743#commitcomment-44428985 a patch like that should workaround this issue until the new version of docker, but I also can't vouch for the compatibility with the vastly different version than expected.

G-M0N3Y-2503, 2020-11-30:
> official fix besides the no check version patch above ?

Correct, the plan is to fix this when v20.10.0 goes in for docker.

zlf179, 2020-11-30:
> > official fix besides the no check version patch above ?
> 
> Correct, the plan is to fix this when v20.10.0 goes in for docker.

thanks for reply.  i will wait for it then.

closed 2020-12-15 by dangowrt