openwrt/packages #14003: mwan3: error on ifup event

pgaufillet, 2020-11-24:
Maintainer: @feckert @aaronjg 
Environment: TP Link Archer C7 v2, OpenWRT r15024-bc64b9c32e

Description: when a interface goes up, mwan3 fails at updating the corresponding routing table:

```
Tue Nov 24 21:33:55 2020 user.notice mwan3-hotplug[3755]: Execute ifup event on interface wan (eth0.2)
Tue Nov 24 21:33:56 2020 user.warn mwan3-hotplug[3755]: failed to add default via 192.168.90.1 dev eth0.2 proto static src 192.168.90.3 to table 1
Tue Nov 24 21:33:56 2020 user.warn mwan3-hotplug[3755]: failed to add 192.168.90.0/24 dev eth0.2 proto kernel scope link src 192.168.90.3 to table 1
Tue Nov 24 21:33:58 2020 user.info mwan3track[1802]: Detect ifup event on interface wan ()
Tue Nov 24 21:33:59 2020 user.notice mwan3track[1802]: Interface wan (eth0.2) is online
Tue Nov 24 21:34:04 2020 user.notice mwan3-hotplug[4541]: Execute ifup event on interface vpn (tun0)
Tue Nov 24 21:34:05 2020 user.warn mwan3-hotplug[4541]: failed to add 10.52.3.0/24 via 10.52.3.13 dev tun0 to table 2
Tue Nov 24 21:34:05 2020 user.warn mwan3-hotplug[4541]: failed to add 10.52.3.13 dev tun0 proto kernel scope link src 10.52.3.14 to table 2
Tue Nov 24 21:34:05 2020 user.warn mwan3-hotplug[4541]: failed to add 192.168.10.0/24 via 10.52.3.13 dev tun0 to table 2
Tue Nov 24 21:34:05 2020 user.warn mwan3-hotplug[4541]: failed to add 192.168.20.0/24 via 10.52.3.13 dev tun0 to table 2
Tue Nov 24 21:34:05 2020 user.warn mwan3-hotplug[4541]: failed to add 192.168.60.0/24 via 10.52.3.13 dev tun0 to table 2

```

aaronjg, 2020-11-24:
With mwan3 disabled, what is the output of ip route list table 1? And with mwan3 enabled? Any other packages you are running that might interfere?

What version of mwan3 are you running?

pgaufillet, 2020-11-24:
This result is obtained with mwan 2.10.3 in a home brewed openwrt image built today. 
Everything was working smoothly with a very similar configuration of OpenWRT r13405-02f08056bc (July 2020) with mwan 2.8.5-1. I have not added any new package since then.

I cannot extract the routing tables right now, but I'll provide them ASAP.

pgaufillet, 2020-11-28:
ip route list table 1
default via 192.168.70.1 dev eth0.2 proto static src 192.168.70.3 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.70.0/24 dev eth0.2 proto kernel scope link src 192.168.70.3 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3

pgaufillet, 2020-11-28:
mwan3 disabled, the routing table 1 is void:
```
ip route list table 1
```

feckert, 2020-12-16:
Fixed with commit https://github.com/openwrt/packages/commit/21ac1b2c3ebf4735644366e22dccba482a92d558

pgaufillet, 2020-12-17:
Tested today: it behaves better, but it seems a few routing rules are still missing when the vpn goes up. There are less "failed to ad ..." errors on hotplug event, but there are still a few of them. For now I still have to revert to r13405.

aaronjg, 2020-12-17:
Thanks. Can you post the remaining "failed to add" errors?

And just wanted to check - are these issues only occurring when the router boots up, or do they also appear when a new interface comes online?

pgaufillet, 2020-12-17:
It seems to happen when tun0 comes up, not especially at boot time. I'll post the log ASAP.

aaronjg, 2020-12-17:
Thanks, would be good to see both the log and which routes are missing.

pgaufillet, 2021-01-17:
Hello!

Finally got some time to test the latest version. In short: it doesn't work yet. 
Here are the results once interfaces are up:
```
ip route list table 1
default via 192.168.70.1 dev eth0.2 proto static src 192.168.70.3 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.70.0/24 dev eth0.2 proto kernel scope link src 192.168.70.3 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```

Routing table 1 looks ok, and everything in this subnet works as expected.

```
ip ro list table 2
10.45.3.0/24 via 10.45.3.13 dev tun0 
10.45.3.1 via 10.45.3.13 dev tun0 
10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 
192.168.0.0/24 via 10.45.3.13 dev tun0 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.2.0/24 via 10.45.3.13 dev tun0 
192.168.50.0/24 via 10.45.3.13 dev tun0 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```
Here the default route is missing and Internet is unavailable.

```
logread | grep mwan
Sun Jan 17 11:10:26 2021 user.notice mwan3-hotplug[2272]: mwan3 hotplug on lan not called because interface disabled
Sun Jan 17 11:13:56 2021 user.notice mwan3-hotplug[3338]: mwan3 hotplug on dmz not called because interface disabled
Sun Jan 17 11:14:02 2021 user.notice mwan3-hotplug[3896]: mwan3 hotplug on loopback not called because interface disabled
Sun Jan 17 11:14:05 2021 user.notice mwan3-hotplug[4183]: Execute ifup event on interface wan (eth0.2)
Sun Jan 17 11:14:06 2021 user.warn mwan3-hotplug[4183]: failed to add default via 192.168.70.1 dev eth0.2 proto static src 192.168.70.3 to table 1
Sun Jan 17 11:14:06 2021 user.warn mwan3-hotplug[4183]: failed to add 192.168.70.0/24 dev eth0.2 proto kernel scope link src 192.168.70.3 to table 1
Sun Jan 17 11:14:07 2021 user.info mwan3track[1907]: Detect ifup event on interface wan ()
Sun Jan 17 11:14:07 2021 user.notice mwan3track[1907]: Interface wan (eth0.2) is online
Sun Jan 17 11:15:15 2021 user.notice mwan3-hotplug[5261]: Execute ifup event on interface vpn (tun0)
Sun Jan 17 11:15:15 2021 user.warn mwan3-hotplug[5261]: failed to add 10.45.3.0/24 via 10.45.3.13 dev tun0 to table 2
Sun Jan 17 11:15:15 2021 user.warn mwan3-hotplug[5261]: failed to add 10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 to table 2
Sun Jan 17 11:15:16 2021 user.warn mwan3-hotplug[5261]: failed to add 192.168.0.0/24 via 10.45.3.13 dev tun0 to table 2
Sun Jan 17 11:15:16 2021 user.warn mwan3-hotplug[5261]: failed to add 192.168.2.0/24 via 10.45.3.13 dev tun0 to table 2
Sun Jan 17 11:15:16 2021 user.warn mwan3-hotplug[5261]: failed to add 192.168.50.0/24 via 10.45.3.13 dev tun0 to table 2
Sun Jan 17 11:15:17 2021 user.info mwan3track[1908]: Detect ifup event on interface vpn ()
Sun Jan 17 11:15:17 2021 user.warn mwan3track[1908]: no src ipv4 address found from netifd for interface 'vpn' dev 'tun0' guessing 10.45.3.14
Sun Jan 17 11:15:17 2021 user.notice mwan3track[1908]: Interface vpn (tun0) is online
```

pgaufillet, 2021-01-17:
As a reference, with exactly the same configuration, here are the results with OpenWrt r13405 and mwan 2.8.5-1:
```
mwan3 status
Interface status:
 interface wan is online and tracking is active
 interface vpn is online and tracking is active
```

```
ip route list table 1
default via 192.168.70.1 dev eth0.2 
10.45.3.0/24 via 10.45.3.13 dev tun0 
10.45.3.1 via 10.45.3.13 dev tun0 
10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 
192.168.0.0/24 via 10.45.3.13 dev tun0 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.2.0/24 via 10.45.3.13 dev tun0 
192.168.50.0/24 via 10.45.3.13 dev tun0 
192.168.70.0/24 dev eth0.2 proto kernel scope link src 192.168.70.3 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```

```
ip route list table 2
default dev tun0 scope link 
10.45.3.0/24 via 10.45.3.13 dev tun0 
10.45.3.1 via 10.45.3.13 dev tun0 
10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 
192.168.0.0/24 via 10.45.3.13 dev tun0 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.2.0/24 via 10.45.3.13 dev tun0 
192.168.50.0/24 via 10.45.3.13 dev tun0 
192.168.70.0/24 dev eth0.2 proto kernel scope link src 192.168.70.3 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```

```
logread | grep mwan
Sun Jan 17 11:50:20 2021 user.notice mwan3[1595]: Using firewall mask 0x3F00
Sun Jan 17 11:50:20 2021 user.notice mwan3[1595]: Max interface count is 60
Sun Jan 17 11:52:32 2021 user.notice ucitrack: Setting up /etc/config/mwan3 reload trigger for non-procd /etc/init.d/mwan3
Sun Jan 17 11:52:35 2021 user.info mwan3rtmon[2222]: Detect rtchange event.
Sun Jan 17 11:52:42 2021 user.info mwan3rtmon[2222]: Detect rtchange event.
Sun Jan 17 11:52:46 2021 user.notice mwan3[3331]: Execute ifup event on interface wan (eth0.2)
Sun Jan 17 11:52:49 2021 user.notice mwan3[3331]: Starting tracker on interface wan (eth0.2)
Sun Jan 17 11:52:50 2021 user.info mwan3rtmon[2222]: Detect rtchange event.
Sun Jan 17 11:52:54 2021 user.notice mwan3[3862]: Execute ifup event on interface vpn (tun0)
Sun Jan 17 11:52:55 2021 user.notice mwan3[3862]: Starting tracker on interface vpn (tun0)
Sun Jan 17 11:52:56 2021 user.info mwan3rtmon[2222]: Detect rtchange event.
```

aaronjg, 2021-01-17:
Thanks for testing. What is the entire protocol for VPB? Can you try setting a unique metric in the two interfaces in the openwrt network config as per:

https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3

If that solves the issue, I’ll have a solution in mind on the mwan3 side...

pgaufillet, 2021-01-17:
I guess you mean VPN and not VPB?

The configuration of wan is:
```
config interface 'wan'
	option ifname 'eth0.2'
	option proto 'dhcp'
	option delegate '0'
```
and the one of vpn (managed by openvpn, IPv4 only):
```
config interface 'vpn'
	option proto 'none'
	option delegate '0'
	option ifname 'tun0'
```

Up to now, I have never set the gateway metric. While it can be done for the wan interface, this parameter doesn't appear in Luci for the vpn interface.

aaronjg, 2021-01-17:
Yes VPN. It was an autocorrect error. Could you try adding a route-metric option to the openvpn config and/or a metric to the WAN config? You can set it back later if you like, just want to confirm that this is the issue before working on the fix to the 2.10 version.

pgaufillet, 2021-01-18:
Ok. I am going to test it ASAP.

pgaufillet, 2021-01-30:
Still in r15523, with metric 90 for vpn (using route-metric on client side) and 100 for wan (using option metric):

```
mwan3 status
Interface status:
 interface wan is online 00h:08m:40s, uptime 00h:09m:03s and tracking is active
 interface vpn is error (16) and tracking is active
```

```
ip route list table 1
default via 192.168.70.1 dev eth0.2 proto static src 192.168.70.3 metric 100 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.70.0/24 dev eth0.2 proto static scope link metric 100 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```

```
ip route list table 2
10.45.3.0/24 via 10.45.3.13 dev tun0 metric 90 
10.45.3.1 via 10.45.3.13 dev tun0 metric 90 
10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 
192.168.0.0/24 via 10.45.3.13 dev tun0 metric 90 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.2.0/24 via 10.45.3.13 dev tun0 metric 90 
192.168.50.0/24 via 10.45.3.13 dev tun0 metric 90 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```

```
logread | grep mwan
Sat Jan 30 11:08:58 2021 user.notice mwan3-hotplug[1982]: mwan3 hotplug on lan not called because interface disabled
Sat Jan 30 11:09:34 2021 user.notice mwan3-hotplug[3094]: mwan3 hotplug on dmz not called because interface disabled
Sat Jan 30 11:09:41 2021 user.notice mwan3-hotplug[3676]: mwan3 hotplug on loopback not called because interface disabled
Sat Jan 30 11:09:46 2021 user.notice mwan3-hotplug[4078]: Execute ifup event on interface wan (eth0.2)
Sat Jan 30 11:09:46 2021 user.warn mwan3-hotplug[4078]: failed to add default via 192.168.70.1 dev eth0.2 proto static src 192.168.70.3 metric 100 to table 1
Sat Jan 30 11:09:46 2021 user.warn mwan3-hotplug[4078]: failed to add 192.168.70.0/24 dev eth0.2 proto static scope link metric 100 to table 1
Sat Jan 30 11:09:47 2021 user.info mwan3track[1617]: Detect ifup event on interface wan ()
Sat Jan 30 11:09:48 2021 user.notice mwan3track[1617]: Interface wan (eth0.2) is online
Sat Jan 30 11:10:53 2021 user.notice mwan3-hotplug[4856]: Execute ifup event on interface vpn (tun0)
Sat Jan 30 11:10:53 2021 user.warn mwan3-hotplug[4856]: failed to add 10.45.3.0/24 via 10.45.3.13 dev tun0 metric 90 to table 2
Sat Jan 30 11:10:53 2021 user.warn mwan3-hotplug[4856]: failed to add 10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 to table 2
Sat Jan 30 11:10:53 2021 user.warn mwan3-hotplug[4856]: failed to add 192.168.0.0/24 via 10.45.3.13 dev tun0 metric 90 to table 2
Sat Jan 30 11:10:53 2021 user.warn mwan3-hotplug[4856]: failed to add 192.168.2.0/24 via 10.45.3.13 dev tun0 metric 90 to table 2
Sat Jan 30 11:10:53 2021 user.warn mwan3-hotplug[4856]: failed to add 192.168.50.0/24 via 10.45.3.13 dev tun0 metric 90 to table 2
Sat Jan 30 11:10:54 2021 user.info mwan3track[1618]: Detect ifup event on interface vpn ()
Sat Jan 30 11:10:54 2021 user.warn mwan3track[1618]: no src ipv4 address found from netifd for interface 'vpn' dev 'tun0' guessing 10.45.3.14
Sat Jan 30 11:10:54 2021 user.notice mwan3track[1618]: Interface vpn (tun0) is online

aaronjg, 2021-01-30:
What’s the output of ‘ip route list’?

pgaufillet, 2021-01-31:
```
ip route list
default via 192.168.70.1 dev eth0.2 proto static src 192.168.70.3 metric 100 
10.45.3.0/24 via 10.45.3.13 dev tun0 metric 90 
10.45.3.1 via 10.45.3.13 dev tun0 metric 90 
10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 
192.168.0.0/24 via 10.45.3.13 dev tun0 metric 90 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.2.0/24 via 10.45.3.13 dev tun0 metric 90 
192.168.50.0/24 via 10.45.3.13 dev tun0 metric 90 
192.168.70.0/24 dev eth0.2 proto static scope link metric 100 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```

aaronjg, 2021-02-01:
It looks like there is no [default route](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#prepare_and_the_check_the_default_routing_table_for_the_wan_interfaces) for the VPN. Does the VPN interface [work without mwan](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#verify_outbound_traffic_on_each_wan_interface)?

pgaufillet, 2021-02-01:
Well, it works as expected, aka it routes the local packets to the remotes internal LANs. It is not designed as a gateway for general purpose, but only for the packets coming from a DMZ, which are recognized and routed thanks to mwan3.
And as you can see a `default dev tun0 scope link` appears on the table 2 with openwrt r13405/mwan 2.8.5-1, which is not the case with openwrt r15523/mwan 2.10.x.

aaronjg, 2021-02-01:
Ah I see, so you are using mwan3 more for policy based routing type application than to load balance/failover between two WANs/VPNs.

That kind of set up is fine, but it sounds like for your application, you actually don't want the default route there on the VPN. Aside from the ` interface vpn is error (16)` message, does this version of mwan3 handle the traffic according to your needs?

BTW - If your setup is just to route packets from a particular source over the VPN you could set up a lighter weight solution with a few ip rules, and avoid the overhead of mwan3 checking link up status and marking packets.

pgaufillet, 2021-02-01:
Exactly. This configuration works nicely with mwan3 2.8.x but stopped working with mwan3 2.10.x.

Your suggestion regarding ip rules is good, but requires more network skills than I have. Moreover performances have not been an issue since I have set up this configuration :-)

aaronjg, 2021-02-02:
Okay, so for your setup it sounds like `default` route should not be in the routing table, since you are only accessing `10.45.3.0/24`. 

So what exactly is the issue?  You described some issues early on, but it is not clear to me what remains since 21ac1b2.  Is any traffic being routed incorrectly, or are you just concerned that warnings are showing up in the error log, even though things otherwise function okay.

pgaufillet, 2021-02-02:
Indeed, the current version just doesn't work with this LAN->WAN / DMZ->VPN configuration. The traffic coming from 192.168.100.0/24 to the Internet is not routed anymore through 10.45.3.1. aka ping google.com worked with mwan3 2.8.x and not anymore with mwan3 2.10.x (not reachable). If you compare the routing tables, I see 2 differencies:

mwan3 2.8.x:
```
ip route list table 2
--> default dev tun0 scope link <--
10.45.3.0/24 via 10.45.3.13 dev tun0 
10.45.3.1 via 10.45.3.13 dev tun0 
10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 
192.168.0.0/24 via 10.45.3.13 dev tun0 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.2.0/24 via 10.45.3.13 dev tun0 
192.168.50.0/24 via 10.45.3.13 dev tun0 
-->192.168.70.0/24 dev eth0.2 proto kernel scope link src 192.168.70.3<--
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```
mwan3 2.10.x:
```
ip route list table 2
10.45.3.0/24 via 10.45.3.13 dev tun0 metric 90 
10.45.3.1 via 10.45.3.13 dev tun0 metric 90 
10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 
192.168.0.0/24 via 10.45.3.13 dev tun0 metric 90 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.2.0/24 via 10.45.3.13 dev tun0 metric 90 
192.168.50.0/24 via 10.45.3.13 dev tun0 metric 90 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```

I guess the issue is coming from the fact mwan3 2.10 is not setting anymore the `default dev tun0 scope link` directive?

aaronjg, 2021-02-02:
> Well, it works as expected, aka it routes the local packets to the remotes internal LANs. It is not designed as a gateway for general purpose, but only for the packets coming from a DMZ, which are recognized and routed thanks to mwan3.

> Indeed, the current version just doesn't work with this LAN->WAN / DMZ->VPN configuration.

Sorry, I'm still a bit confused, are you saying that traffic from the DMZ should be routable to the the internet eg DMZ->VPN->VPN's WAN? Or do you only want traffic from the DMZ to go to the broader internet via WAN?

Do you want traffic from LAN to be able to access internal VPN hosts? 

Just trying to understand the use case here.

pgaufillet, 2021-02-02:
Don't worry: I have not been that clear :-) 

I guess it will be easier with some context:
- The DMZ zone (`192.168.100.0/24`) hosts a mail server. Unfortunately, our ISP blocks communications on port 25.
- To solve this issue, a router has been set up on a third party cloud. This remote router (`10.45.3.1`) is connected to our local router (`192.168.1.2`/`192.168.100.1`) through a VPN.
- The local router splits the DMZ/LAN traffic thanks to mwan3.
- DMZ->Internet is directed to the remote router (through the VPN) that acts as gateway for this zone.
- LAN->Internet is managed locally and the local router acts as gateway for that zone.
- Apart from being the Internet gateway of the DMZ, the remote router aggregates several remote subnetworks of LAN through othe VPN links (`192.168.(0|2|50).0/24`).
- `192.168.70.x`correspond to the WAN interface of the local router.

See the attached schema.

aaronjg, 2021-02-02:
I see. You should be able to accomplish this more simply with something like:

```
ip route add default dev tun0 scope link table 1
```

and 
```
ip rule add from 192.168.100.0/24 lookup table 1
```
or 
```
ip rule add iif eth1.20 lookup table 1
```

Anyway, as far as mwan3 is concerned, I am still a bit confused - does this work [without mwan3](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#verify_outbound_traffic_on_each_wan_interface), as per the guide? IE can you ping the internet via the tun0 device with mwan3 turned off?

I understand it used to work in the old version of mwan3, but I want to make sure I understand exactly what the issue is before attempting a compatibility fix.

pgaufillet, 2021-02-03:
Thanks for your ip hints. I definitely need to have a look to this kind of rules.

I have double checked yesterday the status of wan and tun0 interfaces without mwan3. A default route was missing on the VPN side. After correction, the default routing table looks like:
```
ip route
default via 192.168.70.1 dev eth0.2 proto static src 192.168.70.3 metric 100 
-->default via 10.45.3.13 dev tun0 metric 110 <--
10.45.3.0/24 via 10.45.3.13 dev tun0 
10.45.3.1 via 10.45.3.13 dev tun0 
10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 
192.168.0.0/24 via 10.45.3.13 dev tun0 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.2.0/24 via 10.45.3.13 dev tun0 
192.168.50.0/24 via 10.45.3.13 dev tun0 
192.168.70.0/24 dev eth0.2 proto static scope link metric 100 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```
With this change, it is possible now without mwan3 to ping google.com through eth0.2 and tun0. And it fixes the routing issue in DMZ when mwan3 is active! Obviously my set up was working by chance, relying on a unexpected behavior of mwan3 <=2.8.x. 

The DMZ route is now:
```
ip route list table 2
-->default via 10.45.3.13 dev tun0 metric 110 <--
10.45.3.0/24 via 10.45.3.13 dev tun0 
10.45.3.1 via 10.45.3.13 dev tun0 
10.45.3.13 dev tun0 proto kernel scope link src 10.45.3.14 
192.168.0.0/24 via 10.45.3.13 dev tun0 
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.3 
192.168.2.0/24 via 10.45.3.13 dev tun0 
192.168.50.0/24 via 10.45.3.13 dev tun0 
192.168.100.0/24 dev eth1.20 proto kernel scope link src 192.168.100.3 
```
Thank you for your patience and sorry for the time you lost.  :-/

closed 2021-02-03 by pgaufillet

aaronjg, 2021-02-03:
Great! Glad we got to the bottom of this. Not wasted time, it's an important compatibility issue with the new version. Now the question is whether we should change the behavior in 2.10.x for compatibility.

@feckert: In 2.10.x, the route tables are all copied from the main routing table, so this means if a default route is not present in the main table, it won't be present in the interfaces specific table.  The guide explicitly asks users to check that a default route is present so a "correct" mwan3 config won't have any issue with the upgrade. However, there are potentially more configs such as @pgaufillet that may be working in the older version that could break.

Should we add a default route to the interface tables even if it is not in the main table? The potential downside is that if a user has failover between two VPNs that should not have a default route, I'm not sure we really want to be adding one. We could also add a default route with a config option turn this off.

My preference would be to add missing default routes by default, with a config option available to disable the behavior. @feckert What's your take?

pgaufillet, 2021-02-03:
My 2 cents: 

The new behavior looks better to me once understood, for the reasons you exposed.
It is also possible to plan a transitional stage, keeping the behavior of 2.8 by default for e.g. one major OpenWRT version.
It would be good anyway to log a warning if a default route is added.

feckert, 2021-02-04:
If no default rule is set in the main routing table, none should be created in the interfaces table. This is of course different from the previous mwan3 versions. But I am in favor that we discard wrong and therefore old things. The backwards compatibility is of course a problem. Therefore a warning in the mwan3 would not be bad, that the behavior has changed now. In the LuCI we also give a warning (not distinguishable for IPv4 and IPv6 needs be fixed). Also in the [wiki](https://openwrt.org/docs/guide-user/network/wan/multiwan/mwan3#prepare_and_the_check_the_default_routing_table_for_the_wan_interfaces) we have a note that a default rule for the interface should be set with a different metric.