openwrt/packages #14040: gpsd build broken

borkra, 2020-11-27:
Commit https://github.com/openwrt/openwrt/commit/e52d0487e88c3c8c57e1310d1a02b18eae0d142e broke gpsd build
Verified that reverting commit fixes the problem.

scons: Building targets ...
arm-openwrt-linux-muslgnueabi-gcc -o cgps -Wl,-z,now -Wl,-z,relro -pthread cgps.o -L. -L/home/boris/devel/openwrt/staging_dir/toolchain-arm_cortex-a9+neon_gcc-9.3.0_musl_eabi/usr/lib -L/home/boris/devel/openwrt/staging_dir/toolchain-arm_cortex-a9+neon_gcc-9.3.0_musl_eabi/lib -L/home/boris/devel/openwrt/staging_dir/hostpkg/lib -L/home/boris/devel/openwrt/staging_dir/host/lib -lgps_static -lm -lrt -lncurses
/home/boris/devel/openwrt/staging_dir/toolchain-arm_cortex-a9+neon_gcc-9.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/9.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: /home/boris/devel/openwrt/staging_dir/hostpkg/lib/libncurses.a: error adding symbols: file format not recognized
collect2: error: ld returned 1 exit status
scons: *** [cgps] Error 1

borkra, 2020-11-27:
Adding, sysroot=$(STAGING) into SCONS_CONF in gpsd makefile fixes it too.

neheb, 2020-11-27:
ping @ynezz

ynezz, 2020-11-27:
https://github.com/openwrt/openwrt/commit/e52d0487e88c3c8c57e1310d1a02b18eae0d142e#commitcomment-44602909

neheb, 2020-11-29:
Is this still an issue?

borkra, 2020-11-29:
Yes, it is still an issue, there were no changes to gpsd

borkra, 2020-11-29:
One can apply this patch to fix this issue:

```
diff --git a/utils/gpsd/Makefile b/utils/gpsd/Makefile
index 53428613c..3177d809c 100644
--- a/utils/gpsd/Makefile
+++ b/utils/gpsd/Makefile
@@ -104,6 +104,7 @@ SCONS_OPTIONS += \
        implicit_link=no \
        chrpath=no \
        manbuild=no \
+       sysroot="$(STAGING_DIR)" \
        target="$(TARGET_CROSS:-=)"
 
 define Build/InstallDev
```

neheb, 2020-11-30:
Hmm I can't replicate the issue. Maybe https://github.com/openwrt/packages/pull/14081 fixes it.

borkra, 2020-11-30:
#14081 is about libpam, it does not fix it, I am not even building libpam. There are no user accounts, or need in Pluggable Authentication Modules.

The patch I provided came from buildroot, that is how they handle gpsd.
https://github.com/buildroot/buildroot/blob/master/package/gpsd/gpsd.mk

neheb, 2020-11-30:
Right. My confusion is why it's not failing on my end.

borkra, 2020-11-30:
Maybe the problem is Ubuntu 20.10 special, that is what my host is.

neheb, 2020-11-30:
Doubt it. The buildbots which use Debian 10 also fail: https://downloads.openwrt.org/snapshots/faillogs/arc_arc700/packages/gpsd/compile.txt

My hosts are Fedora 32 and CentOS 7.

borkra, 2020-11-30:
Anyway setting sysroot fixes it.

neheb, 2020-11-30:
I merged a bunch of fixes for several packages. If the buildbots still complain after a day, I'll look into this.

borkra, 2020-11-30:
I looked into the logs a bit more. It looks like gpsd fails only on 'arc' and 'arm32' platforms. On all other platforms it compiles just fine. That would explain why you could not reproduce it.

pprindeville, 2020-12-01:
This is clearly wrong on a target-build: `-L/home/boris/devel/openwrt/staging_dir/hostpkg/lib`

neheb, 2020-12-16:
This is fixed.

closed 2020-12-16 by neheb