openwrt/packages #14075: Package not deleted on adblock and banip

Ansuel, 2020-11-29:
@dibdot I know i should just make a fix and propose a pr but I notice (and this is for a long time now) that for some reason, the downloaded zip for your package are not deleted on extraction. This happen both on adblock and banip. Is this intended?

dibdot, 2020-11-29:
Do you mean a Makefile tweak during opkg package installation?

Ansuel, 2020-11-29:
Think I explained this bad...
The zip downloaded by the package (the block list for example) are stored in /tmp but never deleted when extracted and used. So we have these zip file wasting useful ram. Shouldn't they be removed on script termination?

I notice that there is actually a clean function in the script but in some case, the function is not called.

If you want I can provide config and a screen of the package. But it's mainly the zip the script download for the blocklist and I'm using a very basic config with a few list selected and using aria2c as a download daemon

dibdot, 2020-11-29:
I can't reproduce that. The used temp folder is deleted after adblock processing.

Ansuel, 2020-11-29:
I added my config... Do you have some way to backtrack this. I'm using master I can't be the only one experiencing this.

```

config adblock 'global'
	option adb_enabled '1'
	option adb_debug '0'
	option adb_forcedns '0'
	option adb_safesearch '0'
	option adb_dnsfilereset '0'
	option adb_mail '0'
	option adb_report '0'
	option adb_backup '1'
	option adb_maxqueue '4'
	list adb_sources 'adaway'
	list adb_sources 'adguard'
	list adb_sources 'disconnect'
	list adb_sources 'yoyo'
	option adb_dns 'unbound'
	option adb_fetchutil 'aria2c'
	option adb_trigger 'wan'

```

[image]

dibdot, 2020-11-29:
Ah OK, you mean the generated gz archives ... that's the pre-generated/sorted backup files during processing, controlled by this flag:

[image]

Ansuel, 2020-11-29:
Oh didn't notice that. So it's intended and enabled by default. Can you consider placing them in a dedicated directory in a future version?

dibdot, 2020-11-29:
Well, /tmp is the default (and available on every openwrt device) ... and you can change that in LuCI right under the above listed option. Of course, I could put them in a "better" subdir - noted! ;-)

Ansuel, 2020-11-29:
Yep i mean something like /tmp/adblock_backup :D 
Looks a stupid question but something like me can open another issue thinking it's a bug that there are random zip in the /tmp folder.

closed 2020-11-29 by Ansuel