openwrt/packages #14076: Adblock (v4.0.7) has unknown parameters for `sort`

madelaney, 2020-11-29:
With AdBlock (v4.0.7), the provided `adblock.sh` appears to have unknown parameters for `sort` in OpenWRT (`19.07.3 r11063-85e04e9f46`).

This results in a blocked domain count of zero regardless of how many sources are configured.

If I comment out [adblock.sh#416](https://github.com/openwrt/packages/blob/4fad6e2dc1c85beb591219cc2f129faeb155c42c/net/adblock/files/adblock.sh#L416), then everything seems to work (meaning that the black listed domain count is not zero).

```console
opkg info adblock
Package: adblock
Version: 4.0.7-2
Depends: libc, jshn, jsonfilter, coreutils, coreutils-sort, ca-bundle, opkg
Status: install user installed
Section: net
Architecture: all
Size: 18907
Filename: adblock_4.0.7-2_all.ipk
Conffiles:
 /etc/config/adblock f29522fde8a79baadfc32621874cd9c64feec6a082fd1222336ef9c826316625
 /etc/adblock/adblock.whitelist e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855
 /etc/adblock/adblock.blacklist e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855
Description: Powerful adblock script to block ad/abuse domains via dnsmasq, unbound, named or kresd.
 The script supports many domain blacklist sites plus manual black- and whitelist overrides.
 Please see https://github.com/openwrt/packages/blob/master/net/adblock/files/README.md for further information.
Installed-Time: 1606663051
```

madelaney, 2020-11-29:
I have not overwriten the default sort.
```console
root@OpenWrt:/tmp# ls -lart /usr/bin/sort
lrwxrwxrwx    1 root     root            17 Nov  5 23:58 /usr/bin/sort -> ../../bin/busybox
```

dibdot, 2020-11-29:
adblock requires coreutils-sort (it's a package dependency), busybox sort doesn't support megre sorts.

closed 2020-11-29 by dibdot

madelaney, 2020-12-01:
@dibdot `coreutils-sort` is installed but adblock is picking up the default version of `sort`.

dibdot, 2020-12-01:
Then you've borked your system ... most probably you've updated a core component like busybox via opkg and that has overwritten the symlink to gnu-sort. Just re-install coreutils-sort.