openwrt/packages #14085: ddns: cloudflare: API Tokens can't be used for authentication

erozen, 2020-11-30:
Maintainer: @feckert 
Environment: HEAD, all archs

Description:
Current iteration of the script doesn't allow for the use of API tokens for authentication, which is the preferred method (https://api.cloudflare.com/#getting-started-requests).

The reason they don't work is that the value must be passed under different header name:
```Authorization: Bearer <value>```

Instead of:
```X-Auth-Key: <value>```

It's enough just to include the bearer token, but perhaps the best approach would be to allow a null account id, and assume the password is a token if so?

SquallATF, 2020-12-02:
https://github.com/openwrt/packages/blob/7a41ef82835060b6ac24b7eaf00ff21e761deda6/net/ddns-scripts/files/usr/lib/ddns/update_cloudflare_com_v4.sh#L128-L131
username set to `Bearer` then can use api token

erozen, 2020-12-02:
Looks good.  Where is the documentation located?  I will update that.

feckert, 2020-12-02:
@erozen I am not happy with the whole script handling but for now I would document it [here](https://openwrt.org/docs/guide-user/base-system/ddns?s[]=ddns&s[]=scripts) That would be great if you would take care of it.

closed 2020-12-02 by feckert