openwrt/packages #14088: strongswan: pivot from ipsec- to swanctl-based configs from UCI

pprindeville, 2020-12-01:
Maintainer: @stintel 
Environment: x86_64, generic, HEAD (b0cb3052)

Description:

Generate `swanctl.conf`-based config files and use `swanctl` to manage tunnel state.

pprindeville, 2020-12-01:
@dedeckeh Don't know if you already have some work done on this or not; I'll assume not unless I hear otherwise, so will be starting from scratch...

dedeckeh, 2020-12-01:
@pprindeville No; I don't have the time atm to work on this due to my paid job and other stuff

pprindeville, 2020-12-03:
> No; I don't have the time atm to work on this due to my paid job and other stuff

@dedeckeh Would you have time to review it or test it?  Or is that not possible either?

dedeckeh, 2020-12-04:
@pprindeville I won't be able to test it as i don't have access anymore to devices in IPSec test setups

pprindeville, 2020-12-18:
I have a couple of KVM servers that have room for a dozen or so simultaneous VM's if you don't mind using x86 or amd64 images...

pprindeville, 2023-09-14:
Fixed via PR #14708.

closed 2023-09-14 by pprindeville