openwrt/packages #14113: Darkstat multiple local networks

simm42, 2020-12-02:
Maintainer: @padre-lacroix (find it by checking history of the package Makefile)
Environment: (19.7, OpenWrt version)

Description:

I recently started using darkstat on my openwrt router and noted that you are the listed maintainer (assuming I have the right Jean-Michel Lacroix)

I wanted to use multiple local networks for the host diplay and graphing to include both the ipv4 and ipv6 traffic along with having my internal and DMZ traffic in the same graph.

Here is the patch if you want to include it in the maintained package source - I have also sent this to the package developer.

```

diff --git a/acct.c b/acct.c
index 540bf35..35460b8 100644
--- a/acct.c
+++ b/acct.c
@@ -37,8 +37,9 @@

 uint64_t acct_total_packets = 0, acct_total_bytes = 0;

+#define LOCAL_NET_MAX 10
 static int using_localnet4 = 0, using_localnet6 = 0;
-static struct addr localnet4, localmask4, localnet6, localmask6;
+static struct addr localnet4[LOCAL_NET_MAX], localmask4[LOCAL_NET_MAX], localnet6[LOCAL_NET_MAX], localmask6[LOCAL_NET_MAX];

 /* Parse the net/mask specification into two IPs or die trying. */
 void
@@ -120,13 +121,19 @@ acct_init_localnet(const char *spec)
    /* Register the correct netmask and calculate the correct net.  */
    addr_mask(&localnet, &localmask);
    if (localnet.family == IPv6) {
-      using_localnet6 = 1;
-      localnet6 = localnet;
-      localmask6 = localmask;
+      if(using_localnet6 >= LOCAL_NET_MAX){
+       errx(1, "Exceeded maximum IPv6 local networks");
+      }
+      localnet6[using_localnet6] = localnet;
+      localmask6[using_localnet6] = localmask;
+      using_localnet6++;
    } else {
-      using_localnet4 = 1;
-      localnet4 = localnet;
-      localmask4 = localmask;
+      if(using_localnet4 >= LOCAL_NET_MAX){
+       errx(1, "Exceeded maximum IPv4 local networks");
+      }
+      localnet4[using_localnet4] = localnet;
+      localmask4[using_localnet4] = localmask;
+      using_localnet4++;
    }

    verbosef("local network address: %s", addr_to_str(&localnet));
@@ -138,11 +145,15 @@ static int addr_is_local(const struct addr * const a,
    if (is_localip(a, local_ips))
       return 1;
    if (a->family == IPv4 && using_localnet4) {
-      if (addr_inside(a, &localnet4, &localmask4))
-         return 1;
+      for (int i=0; i < using_localnet4; i++){
+        if (addr_inside(a, &localnet4[i], &localmask4[i]))
+           return 1;
+      }
    } else if (a->family == IPv6 && using_localnet6) {
-      if (addr_inside(a, &localnet6, &localmask6))
-         return 1;
+      for (int i=0; i < using_localnet6; i++){
+        if (addr_inside(a, &localnet6[i], &localmask6[i]))
+           return 1;
+      }
    }
    return 0;
 }
diff --git a/darkstat.c b/darkstat.c
index e3282ec..ed0e31b 100644
--- a/darkstat.c
+++ b/darkstat.c
@@ -193,7 +193,7 @@ static struct cmdline_arg cmdline_args[] = {
    {"-r",             "capfile",         cb_capfile,      0},
    {"-p",             "port",            cb_port,         0},
    {"-b",             "bindaddr",        cb_bindaddr,    -1},
-   {"-l",             "network/netmask", cb_local,        0},
+   {"-l",             "network/netmask", cb_local,       -1},
    {"--base",         "path",            cb_base,         0},
    {"--local-only",   NULL,              cb_local_only,   0},
    {"--snaplen",      "bytes",           cb_snaplen,      0},

```

padre-lacroix, 2020-12-02:
Yes, indeed you have the right Jean-Michel Lacroix and I am the current maintainer for the package Darkstat.
I see by your e-mail that you have an extensive patch.  I would like to know a few things about this patch:
- Did you test it on your OpenWRT router or you tested it on anothert kind of paltform (ex: Debian, Suse, FreeBSD, OpenBSD, etc.)?
- You mention that you sent it to the package developer (https://unix4lyfe.org/darkstat/ I assume): did you get any feedback from the developper (although it may be too early)?

If the maintainer (https://unix4lyfe.org/darkstat) is updating darkstat with your patch, I will most probably update the OpenWRT package with the new darkstat release.  If the maintainer does not update darkstat, then it is more complicated for me to put it in OpenWrt at the current moment as:
- I do not have IP V6 in my network (I do not need it) and cannot fully test your patch
- I do not have multiple local network (although that can be arranged)
- I do not have currently the time to work on this as I have other obligations

But, if you want to become the maintained of darkstat, feel free to do so: 5 years ago I took the maintenance of the package to prevent it for dissapearing from OpenWRT as there was no maintainer at the time and it would have been removed from master and put in the obsolete packages.  If the maintainer of darkstat updates it, I will update the OpenWRT package, but I will not fork a branch of the current darkstat to implement additional features like the one that you propose.  But if you want to do it and be the maintainer of darkstat in OpenWRT, I will encourage you to do so. 

Let me know what you would like to do

simm42, 2020-12-03:
I got a response from the upstream developer (yes https://unix4lyfe.org/darkstat/)

The response: "Nice self-contained patch. If I find some time to work on darkstat again, I'll merge it."

So yes it could end up in the package, might not.

I've tested it on the raspberry pi I built it on, I need to setup an openwrt build system to get it onto my router. I'm going to try to build it as a quilt patch through the openwrt build system and if I get that working and tested I'll make a pull request with the patch there. Given the base package hasn't been updated in 5 years I'm not expecting the fix to come quickly....

simm42, 2020-12-04:
https://github.com/openwrt/packages/pull/14135 raised to add patch for this feature

Passes all build system tests and build artefact darkstat_3.0.719-5_mips_24kc.ipk installed and tested on openwrt 19.07-4 build on lantiq/xrx200/bt_homehub_v5a install