openwrt/packages #14114: [lighttpd] -mod-openssl missing from package feed

n8v8R, 2020-12-02:
Maintainer: @flyn-org 
Environment: mvebu (Turris Omnia / armv7l / Cortex-A9)
lighttpd: 1.4.56
_____
from the make file https://github.com/openwrt/packages/blob/master/net/lighttpd/Makefile#L74 it would be expected that lighttpd-mod-openssl is available through the package feed but it is however not, probably not being compiled

https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/

[image]

gstrauss, 2020-12-02:
The packages are present for https://downloads.openwrt.org/snapshots/packages/x86_64/packages/ but not https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/

Why would the platform make a difference?

gstrauss, 2020-12-02:
Also present in https://downloads.openwrt.org/snapshots/packages/mips_mips32/packages/, but not present in a couple other 32-bit arches I spot-checked.

n8v8R, 2020-12-02:
Wow, there are plenty of other _**lighttpd***_ packages missing for mvebu in comparison to other platforms...

gstrauss, 2020-12-02:
There are many missing lighttpd packages on a few architectures (more than one), and there are also numerous architectures where all the packages are present (more than one), so this is not a one-off broken architecture in either direction.

Are you the right person to look into this further, or should we try to involve other openwrt maintainers?

I am going to try to go spelunking in the logs, but this is not my area of expertise.

gstrauss, 2020-12-02:
In my (personal) test build of i386_pentium-mmx, I see lighttpd-mod-openssl_1.4.56-2_i386_pentium-mmx.ipk in my bin/ dir, so I know that it can build the package on i386_pentium-mmx, even though I do not see that package in https://downloads.openwrt.org/snapshots/packages/i386_pentium-mmx/packages/

I do not understand why some architectures have all the lighttpd packages, and others only have some of the lighttpd packages.  The openwrt lighttpd Makefile is architecture-agnostic.

** However, I see lighttpd_1.4.56-1_i386_pentium-mmx.ipk (1.4.56-1) instead of 1.4.56-2 on the platforms that are missing packages. ** So maybe there are some architectures that have not yet picked up lighttpd 1.4.56-2 from #14098 which was merged 15 hours ago (as I write this note)

flyn-org, 2020-12-02:
This might be related: I am starting to see some build failures related to liblua.
```
rstrip.sh: /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/lighttpd-1.4.56/ipkg-arm_cortex-a9_vfpv3-d16/lighttpd-mod-cgi/usr/lib/lighttpd/mod_cgi.so: shared object
Packaged contents of /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/lighttpd-1.4.56/ipkg-arm_cortex-a9_vfpv3-d16/lighttpd-mod-cgi into /builder/shared-workdir/build/sdk/bin/packages/arm_cortex-a9_vfpv3-d16/packages/lighttpd-mod-cgi_1.4.56-1_arm_cortex-a9_vfpv3-d16.ipk
Package lighttpd-mod-cml is missing dependencies for the following libraries:
liblua5.3.so.0.0.0
Makefile:198: recipe for target '/builder/shared-workdir/build/sdk/bin/packages/arm_cortex-a9_vfpv3-d16/packages/lighttpd-mod-cml_1.4.56-1_arm_cortex-a9_vfpv3-d16.ipk' failed
make[3]: *** [/builder/shared-workdir/build/sdk/bin/packages/arm_cortex-a9_vfpv3-d16/packages/lighttpd-mod-cml_1.4.56-1_arm_cortex-a9_vfpv3-d16.ipk] Error 1
time: package/feeds/packages/lighttpd/compile#48.31#5.54#37.73
```
This was in https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9_vfpv3-d16/packages/lighttpd/compile.txt. That said, it looks like the current lighttpd Makefile pulls in liblua as a dependency when the cml plugin is active. Does it need to pull in liblua53 instead?

It might be a good idea to let the build server settle for a bit.

gstrauss, 2020-12-02:
The commit in #14098 was to address the lua dependency issue in the lighttpd-1.4.56-1 package.

n8v8R, 2020-12-02:
If not mistaken it will take another 12 hrs from now till the build bot for mvebu/cortexa9 gets there. http://builds.openwrt.org/master/images/builders/mvebu%2Fcortexa9?numbuilds=25

n8v8R, 2020-12-03:
appears that the upgrade from 1.4.56**-1** to 1.4.56**-2** sorted the issue

closed 2020-12-03 by n8v8R