openwrt/packages #14118: Transmission error

9000000, 2020-12-02:
```
Model
Xiaomi Mi Router 3 Pro
Architecture
MediaTek MT7621 ver:1 eco:3
Firmware Version
OpenWrt SNAPSHOT r15123-36db12b883 / LuCI Master git-20.334.32813-3b5eb1a
Kernel Version
5.4.80
```

```
daemon.warn procd: Seccomp support for transmission::instance1 not available
Thu Dec  3 06:44:00 2020 daemon.err transmission-daemon[12735]: [2020-12-03 06:44:00.161] Couldn't save file "/mnt/sda1/transmission/settings.json": File exists (variant.c:1226)
Thu Dec  3 06:44:12 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Thu Dec  3 06:44:14 2020 daemon.err transmission-daemon[12739]: [2020-12-03 06:44:14.133] Couldn't save file "/mnt/sda1/transmission/settings.json": File exists (variant.c:1226)
```

neheb, 2020-12-04:
I assume similar to https://github.com/openwrt/packages/issues/13243

I have no idea why this happens.

Gingernut1978, 2020-12-05:
Also getting a similar error when Transmission just crashes after loading a torrent.

```
x86_64
Model: Micro-Star International Co., Ltd. Cubi N 8GL (MS-B171)
Architecture: Intel(R) Celeron(R) N4000 CPU @ 1.10GHz
Firmware Version:OpenWrt SNAPSHOT r15132-9c13513ce1 / LuCI Master git-20.339.47288-086f323
Kernel Version: 5.4.80
```

```
Sat Dec  5 17:35:44 2020 kern.info kernel: [27752.826230] traps: transmission-da[5705] general protection fault ip:7f68e5032deb sp:7f68e4f49320 error:0 in libwolfssl.so.24.2.0[7f68e4fbe000+7a000]
Sat Dec  5 17:35:49 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:35:58 2020 kern.info kernel: [27766.863157] traps: transmission-da[6901] general protection fault ip:7fca9a70cdeb sp:7fca9a623320 error:0 in libwolfssl.so.24.2.0[7fca9a698000+7a000]
Sat Dec  5 17:36:03 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:36:12 2020 kern.info kernel: [27780.895850] traps: transmission-da[6912] general protection fault ip:7f369c520deb sp:7f369c437320 error:0 in libwolfssl.so.24.2.0[7f369c4ac000+7a000]
Sat Dec  5 17:36:17 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:36:26 2020 kern.info kernel: [27794.925846] traps: transmission-da[6946] general protection fault ip:7f7498eb6deb sp:7f7498dcd320 error:0 in libwolfssl.so.24.2.0[7f7498e42000+7a000]
Sat Dec  5 17:36:31 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:36:41 2020 kern.info kernel: [27809.456663] traps: transmission-da[6951] general protection fault ip:7f9f65a56deb sp:7f9f6596d320 error:0 in libwolfssl.so.24.2.0[7f9f659e2000+7a000]
Sat Dec  5 17:36:46 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:36:55 2020 kern.info kernel: [27823.491442] traps: transmission-da[6960] general protection fault ip:7f405b383deb sp:7f405b29a320 error:0 in libwolfssl.so.24.2.0[7f405b30f000+7a000]
Sat Dec  5 17:37:00 2020 daemon.warn procd: Seccomp support for transmission::instance1 not available
Sat Dec  5 17:37:09 2020 daemon.info procd: Instance transmission::instance1 s in a crash loop 6 crashes, 9 seconds since last crash
Sat Dec  5 17:37:09 2020 kern.info kernel: [27837.522517] traps: transmission-da[6965] general protection fault ip:7f02f99c6deb sp:7f02f98dd320 error:0 in libwolfssl.so.24.2.0[7f02f9952000+7a000]
```

9000000, 2020-12-09:
```
22082.496880] do_page_fault(): sending SIGSEGV to transmission-da for invalid write access to 0221d518
[22082.506021] epc = 77cc2814 in libwolfssl.so.24.2.0[77c16000+df000]
[22082.512233] ra  = 00413167 in transmission-daemon[400000+43000]
```

closed 2025-11-14 by GeorgeSapkin