openwrt/packages #14119: vpnc: build failure: svn SSL certificate verification failed

amezin, 2020-12-03:
Maintainer: @danielg4
Environment: ipq806x, OpenWRT git master

Description:

When trying to build OpenWRT, I get the following error:

`logs/package/feeds/packages/vpnc/compile.txt`:
```
Checking out files from the svn repository...
Redirecting to URL 'https://svn.unix-ag.uni-kl.de/vpnc/trunk':
svn: E170013: Unable to connect to a repository at URL 'https://svn.unix-ag.uni-kl.de/vpnc/trunk'
svn: E230001: Server SSL certificate verification failed: issuer is not trusted
make[3]: *** [Makefile:95: /home/amezin/openwrt/dl/vpnc-0.5.3.r550.tar.gz] Error 1
time: package/feeds/packages/vpnc/compile#0.15#0.09#1.70
```

I've got around the issue by manually performing `svn co https://svn.unix-ag.uni-kl.de/vpnc/trunk` and permanently accepting the certificate

dhewg, 2020-12-03:
Same here...
For god's sake, get rid of that svn crap

danielg4, 2020-12-03:
There are two possible courses of action here:
1) @dwmw2 could integrate vpnc's functionality into the upstream openconnect source, which he has at times said he intended to do, or:
2) I could repoint the package to another upstream, as suggested in streambinder/vpnc#14.

Option 1 is probably not likely to happen soon, if at all. Option 2 would have the following hiccups:
1) I can no longer test the resulting builds myself, because the provider I previously tested with switched to openvpn.
2) Once an update is tested (by you?), I'd still have to submit it as a GitHub PR and wait for somebody to approve it, because I do not have write access to the repo myself.

On the other hand, if anybody who has actual write access to the packages repo is interested in taking over the package maintainership of vpnc, I would have no objection. However, I'm a bit unclear on which people currently have such access after some recent changes.

Sorry.

neheb, 2020-12-04:
Just make a PR. I'll merge.

dhewg, 2020-12-04:
Option #2 sounds like a perfect match for openwrt.
I was a little shocked how old r550 of the original svn repo is :)

PR #14698 (2021-02-09): vpnc: update to maintained github fork
The SVN-based version has not changed in years. Many distros use this
fork as evident here: https://github.com/streambinder/vpnc/issues/14

Compile tested against GnuTLS and OpenSSL on ramips target.

Fixes #14119.

Maintainer: @danielg4 
Compile tested: ramips, mt7621, master
commit 9cbd369a (2021-02-09): vpnc: update to maintained github fork

The SVN-based version has not changed in years. Many distros use this
fork as evident here: https://github.com/streambinder/vpnc/issues/14

Compile tested against GnuTLS and OpenSSL on ramips target.

Fixes #14119.
closed 2021-02-09 by neheb