openwrt/packages #14120: gnutls: build failure: tries to link to host libnettle.so

amezin, 2020-12-03:
Maintainer: @nmav 
Environment: ipq806x, OpenWRT git master

Description:

When building gnutls I get the following error:

```
libtool: install: /usr/bin/install -c .libs/libgnutls.lai /home/amezin/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gnutls-3.6.15/ipkg-install/usr/lib/libgnutls.la
libtool: warning: relinking 'libgnutlsxx.la'
libtool: install: (cd /home/amezin/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gnutls-3.6.15/lib; /bin/sh "/home/amezin/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gnutls-3.6.15/libtool"  --silent --tag CXX --mode=relink arm-openwrt-linux-muslgnueabi-g++ -I./includes -I./includes -O2 -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -fmacro-prefix-map=/home/amezin/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gnutls-3.6.15=gnutls-3.6.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -no-undefined -version-info 29:0:1 -L/home/amezin/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-10.2.0_musl_eabi/usr/lib -L/home/amezin/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-10.2.0_musl_eabi/lib -znow -zrelro -o libgnutlsxx.la -rpath /usr/lib libgnutlsxx_la-gnutlsxx.lo libgnutls.la -inst-prefix-dir /home/amezin/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/gnutls-3.6.15/ipkg-install)
/usr/lib/libnettle.so: file not recognized: file format not recognized

collect2: error: ld returned 1 exit status
libtool:   error: error: relink 'libgnutlsxx.la' with the above command before installing it
```

Full log: [compile.txt](https://github.com/openwrt/packages/files/5633014/compile.txt)

NoTengoBattery, 2020-12-05:
I am experiencing the same problem. The thing is that it fails for `libnettle.so` and for `libkmod.so` when building for the IPQ40xx target.

 The other weird thing is that using the same configuration (with the respective differences) for my MT7621 build, it builds just fine. It's the same host environment, but every build is done in a separated source tree. Therefore it looks like this does not affect the 74kc target.

I already tried doing a `make distclean` and it changed nothing.

neheb, 2020-12-06:
More fallout from https://github.com/openwrt/openwrt/commit/e52d0487e88c3c8c57e1310d1a02b18eae0d142e

Strange that GnuTLS picks up the host library.

Try removing nettle from the host with "apt remove nettle" or whatever.

NoTengoBattery, 2020-12-06:
Manjaro here. I can rename `libnettle` temporarily from the host and it will fail with the same error for `libkmod`.

And as I said before, running in the same host but in a different folder, for a different target, does not show the same problem.

The folders are siblings and one is called `mt7621` and the other is called `ipq40xx`.

neheb, 2020-12-06:
I'd want to see a config.log file.

I don't see anything in mine that would cause failure.

NoTengoBattery, 2020-12-06:
@neheb I am ready to test the suggested change now. In the meantime, and before I clean the `gnutls` package, take a look at the `config.log`.
[config.log](https://github.com/openwrt/packages/files/5648173/config.log)

neheb, 2020-12-06:
Biggest different is the paths in the beginning.

neheb, 2020-12-10:
Is this still an issue?

NoTengoBattery, 2020-12-12:
> Is this still an issue?

Yes, if I do not patch the files, building the packages will always fail.

neheb, 2020-12-15:
Can you post a .config file? I seriously cannot reproduce this.

neheb, 2020-12-17:
Uncomment the PKG_FIXUP line in the Makefile.

neheb, 2020-12-17:
I give up. I hate this packages.

ping @nmav

nmav, 2020-12-17:
I'm not sure what's the problem, or if there is a problem at all. Let me follow up when there is a way to reproduce it (a .config file and the branches it applies to).

NoTengoBattery, 2020-12-17:
There is a problem and even if the OP didn't provided more information, I isolated the problem to specific packages and I know it happens only on Arch-like distros (because I have Debian-like distros to test and the problem is not present).

I provided both build logs, one before applying the suggested change and one after. I actually don't have time during the week, so I can't efficient test things till the weekend.

If you want to see my environment (a snapshot of the whole source code including the configuration): https://github.com/NoTengoBattery/openwrt/blob/linksys-ea6350v3-mastertrack

neheb, 2020-12-17:
The problem is there is an rpath parameter that links to /use/lib. This is strange as --disable-rpath is passed.

neheb, 2020-12-17:
Note that with a different packages I was able to fix the issue by running autoreconf. With this one however, it errors in some strange way. There's are a bunch of broken libc headers included in gnutls for some reason.

nmav, 2020-12-18:
I wonder what breaks with autoreconf/trying to reproduce now. Do you have the errors handy? About the rpath, would a fixup on libtool only address it? Something like:
```
diff --git a/libs/gnutls/Makefile b/libs/gnutls/Makefile
index 6dba3b10a..0d3c324a0 100644
--- a/libs/gnutls/Makefile
+++ b/libs/gnutls/Makefile
@@ -9,13 +9,13 @@ include $(TOPDIR)/rules.mk
 
 PKG_NAME:=gnutls
 PKG_VERSION:=3.6.15
-PKG_RELEASE:=1
+PKG_RELEASE:=2
 PKG_USE_MIPS16:=0
 
 PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
 PKG_SOURCE_URL:=https://www.gnupg.org/ftp/gcrypt/gnutls/v3.6
 PKG_HASH:=0ea8c3283de8d8335d7ae338ef27c53a916f15f382753b174c18b45ffd481558
-#PKG_FIXUP:=autoreconf gettext-version
+PKG_FIXUP:=libtool
 PKG_MAINTAINER:=Nikos Mavrogiannopoulos <nmav@gnutls.org>
 PKG_LICENSE:=LGPL-2.1-or-later
 PKG_CPE_ID:=cpe:/a:gnu:gnutls
```

neheb, 2020-12-18:
```
OpenWrt-libtool: compile:  powerpc-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I.. -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-10.2.0_musl/usr/include -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-10.2.0_musl/include -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/mangix/devstuff/openwrt/build_dir/target-powerpc_464fp_musl/gnutls-3.6.15=gnutls-3.6.15 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -MT bitrotate.lo -MD -MP -MF .deps/bitrotate.Tpo -c bitrotate.c  -fPIC -DPIC -o .libs/bitrotate.o
In file included from bitrotate.h:23,
                 from bitrotate.c:3:
./stdint.h:89:5: error: #if with no expression
   89 | #if

```

nmav, 2020-12-18:
> ```
> OpenWrt-libtool: compile:  powerpc-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I.. -I/home/mangix/devstuff/openwrt/staging_dir/toolchain-powerpc_464fp_gcc-10.2.0_musl/usr/include [...]

That's gnulib that tries to reimplement everything and often fails with this ugly errors :(
I do not see this error on mvebu, is it only after autoreconf, and powerpc?

neheb, 2020-12-18:
After autoreconf (which libtool also calls).

It's with every platform, not just powerpc.

nmav, 2020-12-18:
I cannot reproduce it. Let me try the CI with https://github.com/openwrt/packages/pull/14256

nmav, 2020-12-18:
I wonder whether there are some cross compilation variables that are set differently depending on what exists in the system as you say.

nmav, 2020-12-21:
@neheb the cause of the problem you see (seems unrelated to the report) is due to wint_t.m4. It may be that the one used after autoreconf is problematic (that can be the system one, or the shipped by gnutls - autotools is such a joy).

Try setting these variables to configure:
```
ac_cv_type_wint_t=yes
gl_cv_type_wint_t_too_small=no
gt_cv_c_wint_t=yes
```

neheb, 2020-12-22:
Reminds me of @jow- 's fix for lldpd: https://github.com/lldpd/lldpd/pull/423

I assume something similar can be applied here?

And yes, it seems the system wint_t.m4 is overriding the one in GnuTLS.

neheb, 2020-12-22:
@NoTengoBattery it seems arch/manjaro is a special kind of hell in terms of compiling stuff. All sorts of packages break after https://github.com/openwrt/openwrt/commit/e52d0487e88c3c8c57e1310d1a02b18eae0d142e whereas they compile fine on other platforms.

cotequeiroz, 2021-08-18:
I'm not sure the issue here is still happening, but at least the [problempointed out by @neheb](https://github.com/openwrt/packages/issues/14120#issuecomment-747798961)  is still happening. 

This is another one in my "weird stuff that was driving me crazy" series.  gnutls is running `aclocal` with the `--install`, which, after the above patch was applied, copies old `wint_t.m4` file from `staging_dir/host/share/aclocal` to `staging_dir/usr/share/aclocal`, causing trouble with other packages.  This will fail, for example:

```
make package/{gnutls,fuse-overlayfs}/compile
```
```
In file included from bitrotate.h:23,
                 from bitrotate.c:3:
./stdint.h:89:5: error: #if with no expression
 #if
     ^
In file included from bitrotate.h:23,
                 from hash.c:29:
./stdint.h:89:5: error: #if with no expression
 #if
     ^
make[7]: *** [Makefile:498: bitrotate.o] Error 1
```
This is happening because of the `wint_t.m4` patch, and because of the strange way the autotools macros are being used in by `fuse-overlayfs`: it just blindly counts on the macros being defined, in this specific case `GNULIB_OVERRIDES_WINT_T`.  If we use the most recent version of `wint_t.m4`, we will have a problem again because [the macro was renamed to `GNULIBHEADERS_OVERRIDE_WINT_T`](https://git.savannah.gnu.org/cgit/gnulib.git/commit/m4/wint_t.m4?id=db16856ae761bc213942c17f6fc2b7e9655014b4).

While we can go around the problem by using a newer (but not the newest) version of `wint_t.m4` in `tools/missing_macros/src/m4`, we will eventually run into trouble with some package that needs a specific and different version of it, since the macros in the staging dir are used before the ones in the package's own source dir.  `fuse-overlayfs` has a working version of `wint_t.m4` in its source, but the one installed in the staging_dir is preferred.

I don't quite see how the `wint_t.m4` patch is affecting where `/usr/lib/libnettle` is used, but I can't reproduce the issue here either.

Manually removing the file after it was copied will create problems with parallel jobs.

Another way around it is to disable the `--install` flag passed to `aclocal`, but I don't know if it will bring back the linking issue:
```patch
--- a/Makefile.am
+++ b/Makefile.am
@@ -57,7 +57,7 @@ if ENABLE_DOC
 SUBDIRS += doc
 endif
 
-ACLOCAL_AMFLAGS = -I m4 -I src/libopts/m4 -I src/gl/m4 -I lib/unistring/m4 --install
+ACLOCAL_AMFLAGS = -I m4 -I src/libopts/m4 -I src/gl/m4 -I lib/unistring/m4
 
 EXTRA_DIST = cfg.mk maint.mk CONTRIBUTING.md README.md LICENSE AUTHORS NEWS \
 	ChangeLog THANKS INSTALL.md RELEASES.md
```